subsurface/subsurface #3221: Programme crash v 5.0.0

SousaMartins, 2021-03-28:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

SousaMartins, 2021-03-28:
The programme crash when trying to load a local log from the file system
Windows 10
5.0.0

dirkhh, 2021-03-28:
I can reproduce this here locally (oddly, my test sequence on Windows doesn't include loading a local file - I simply test with the cloud for some reason...). 
But I cannot reproduce it when running inside a debugger, so I can't seem to create a stack trace that would tell me where things crash - which makes fixing this bug a lot harder.
I'll try a few more things, but I at least wanted to acknowledge the bug.

mturkia, 2021-03-29:
You can register a debugger in Windows to grab crashes. So it will allow
you to inspect the crash after it occurs. IIRC it is called Just-In-Time
Debugger

hth

dirkhh, 2021-03-29:
I managed to do that - and the backtrace is completely value free. Tells me nothing about where the code actually crashed.
Oh how I hate Windows.

dirkhh, 2021-03-29:
A debug build doesn't crash.

SousaMartins, 2021-03-29:
Are you using visual studio?
It allows you to debug line by line.

PeterZaal, 2021-03-29:
All I can see from the Windows error report and dump file it's an Access Violation on a read, but that is probably something you have already figured out.

dirkhh, 2021-03-29:
Indeed it's an access violation - so I'm trying to figure out what causes it with Valgrind under Linux. 
And no, I'm not using VisualStudio.
The binaries are cross-compiled using MXE under Linux. I use gdb from MSYS2 to debug things on Windows. Which allows line by line debugging - but annoyingly the code doesn't crash when running under gdb...