subsurface/subsurface #3224: Shearwater

SousaMartins, 2021-03-31:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The Shearwater Perdix connects with the surface and starts importing the data, but few moments after launch an error.
"Dive data import Error"

### Operating system:

Windows 10 Pro 
version 20H2
OS Build 19042.868
Windows Feature Experience Pack 120.2212.551.0

### Subsurface version:

Subsurface 5.0.1
Official Release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

ryanm101, 2021-07-28:
v84 is out now might be worth updating the firmware

SousaMartins, 2021-08-02:
I will

 

Thanks

 

Sousa Martins

dirkhh, 2021-08-19:
many, many instances of this issue - in all cases people were able to get it to work eventually.

closed 2021-08-19 by dirkhh