subsurface/subsurface #3227: Location permission possibly impacting Background Location Service on Android

pabdakine, 2021-04-05:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After enabling the Location Service, I get a single fix but nothing later when the phone is asleep, despite having substantially changed location. When inspecting the Location permission in the OS/app settings, I have only "Deny" and "Allow only while using the app". An email exchange with Dirk indicated there should be a third, "Allow all the time". (I do see this on some of my other apps, so I don't think it's an OS issue.)

### Operating system:

Android 10 (English) / Oxygen OS 10.3.8
OnePlus 7 GM1900

### Subsurface version:

Subsurface-mobile 3.2.0 (5.0.0), installed from Google Play
Same behavior on the immediately prior version (tested 22 March, 2021).

### Steps to reproduce:

1. Enable Location Service
2. Set the device to sleep
3. Change location substantially
4. Wake the device and inspect the list of GPS fixes

### Current behavior:

Single location fix after enabling the Location Service, then no additional fixes.

### Expected behavior:

Various GPS fixes in the list as the device was moved.

### Additional information:

### Mentions:

dirkhh, 2021-04-05:
It appears that some Android flavors refuse to give the app background location permission.
I'm investigating the issue.

dirkhh, 2021-08-19:
with the latest version on a Google Pixel 5 I was able to run the service in the background.
But then that's just an example of "hey, here it works". This doesn't mean it will now work for your device.

pabdakine, 2021-08-19:
Thank you for looking into it further. For me, version 3.2.2 doesn't present the "Allow All The Time" location permission as an option. Perhaps it's something specific to my OS (now Oxygen OS 11.0.2.1 / Android 11). OTOH, I checked some weather apps, which do present that option. Puzzling.

Again, thank you for all your work on Subsurface. Cheers!

Paul

flindeberg, 2021-09-08:
I just skimmed the manifest and it seems to assemble some parts dynamically during the build process. So I don't actually know if these permissions are missing in the resulting manifest. Anyhow the following permission has to be in the manifest for Android 10+ for background location to work _at all_:

```
  <!-- Required only when requesting background location access on
       Android 10 (API level 29) and higher. -->
  <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
```
Could be of interest as well:
```
  <!-- To request foreground location access, declare one of these permissions. -->
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

See https://developer.android.com/training/location/permissions

dirkhh, 2021-09-08:
Those permissions should be added in the build/assembly process for the app.
Here is the effective manifest of the latest app bundle:
```
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:compileSdkVersion="29" android:compileSdkVersionCodename="10" android:installLocation="auto" android:versionCode="3122" android:versionName="3.3.0 (5.0.3)" package="org.subsurfacedivelog.mobile" platformBuildVersionCode="3122" platformBuildVersionName="3.3.0 (5.0.3)">
    <uses-sdk android:minSdkVersion="21" android:targetSdkVersion="29"/>
    <supports-screens android:anyDensity="true" android:largeScreens="true" android:normalScreens="true" android:smallScreens="true"/>
    <uses-permission android:name="android.permission.WAKE_LOCK"/>
    <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE"/>
    <uses-feature android:name="android.hardware.location.gps" android:required="false"/>
    <application android:extractNativeLibs="true" android:hardwareAccelerated="true" android:icon="@drawable/subsurface_mobile_icon" android:label="@string/app_name" android:name="org.qtproject.qt5.android.bindings.QtApplication">
        <activity android:configChanges="orientation|uiMode|screenLayout|screenSize|smallestScreenSize|layoutDirection|locale|fontScale|keyboard|keyboardHidden|navigation|mcc|mnc|density" android:label="@string/app_name" android:launchMode="singleTop" android:name="org.subsurfacedivelog.mobile.SubsurfaceMobileActivity" android:screenOrientation="unspecified" android:theme="@style/AppTheme" android:windowSoftInputMode="adjustResize">
            <intent-filter>
                <action android:name="android.intent.action.MAIN"/>
                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>
            <meta-data android:name="android.app.lib_name" android:value="Subsurface-mobile"/>
            <meta-data android:name="android.app.qt_sources_resource_id" android:resource="@array/qt_sources"/>
            <meta-data android:name="android.app.repository" android:value="default"/>
            <meta-data android:name="android.app.qt_libs_resource_id" android:resource="@array/qt_libs"/>
            <meta-data android:name="android.app.bundled_libs_resource_id" android:resource="@array/bundled_libs"/>
            <meta-data android:name="android.app.bundle_local_qt_libs" android:value="1"/>
            <meta-data android:name="android.app.use_local_qt_libs" android:value="1"/>
            <meta-data android:name="android.app.libs_prefix" android:value="/data/local/tmp/qt/"/>
            <meta-data android:name="android.app.load_local_libs_resource_id" android:resource="@array/load_local_libs"/>
            <meta-data android:name="android.app.load_local_jars" android:value="jar/QtAndroid.jar:jar/QtAndroidExtras.jar:jar/QtAndroidBluetooth.jar:jar/QtAndroidBearer.jar:jar/QtPositioning.jar"/>
            <meta-data android:name="android.app.static_init_classes" android:value="org.qtproject.qt5.android.bluetooth.QtBluetoothBroadcastReceiver:org.qtproject.qt5.android.positioning.QtPositioning"/>
            <meta-data android:name="android.app.splash_screen_drawable" android:resource="@drawable/splash"/>
            <intent-filter>
                <action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"/>
            </intent-filter>
            <meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" android:resource="@xml/device_filter"/>
            <meta-data android:name="android.max_aspect" android:value="3"/>
        </activity>
    </application>
</manifest>
```

flindeberg, 2021-09-08:
> Here is the effective manifest of the latest app bundle:

There we have it, `<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />` is missing. This will disable the "Allow all the time"-option of Android-distributions strictly following AOSP-guidelines.

dirkhh, 2021-09-08:
Which is so funny, since on my Google Pixel 5 with latest Android this works just fine.
But sure, I'll see if I can convince the Android build tools to add that permission as well.

flindeberg, 2021-09-09:
Did it work? I've fixed the background location issue in other projects by adding the above permission in the manifest.

dirkhh, 2021-09-09:
Well, that was easy. The app update was rejected as non-compliant with app store policies

flindeberg, 2021-09-09:
Interesting, had no idea that Google changed their policy visa vi background location: https://support.google.com/googleplay/android-developer/answer/9799150

It seems apps requesting background location must be manually verified nowadays. I.e. probably more work than its worth to get it working.

dirkhh, 2021-09-09:
Ironically I went through all this, I added the warnings and tried to follow all their rules. And got a rejection notice that basically states "you're doing it wrong". It's mildly challenging to come up with the motivation to fight for this feature... I understand and appreciate that Google (and Apple) want to prevent abuse of location tracking. But we aren't abusing it...

dirkhh, 2021-09-13:
OK, closing this - I ended up doing kinda the opposite.
All GPS support has been removed from the mobile apps as the app is not able to comply with the Google play location permission rules.

closed 2021-09-13 by dirkhh