subsurface/subsurface #3236: small translation issue in french ios app

Ospeleo, 2021-04-30:
Hi,

in iOS app, using french translation
touching the hamburger menu (bottom left)
there is the switch option "Désactiver la synchro auto avec le cloud" ( when the synchronization is manual)
and "Activer la synchro automatic" (when synchronization is automatic)

due to the conjugation of the verb, the phrasing in french means the contrary of what it actially does.

the correct phrases should be:
"Synchro manuelle avec le cloud" (or " Synchro auto désactivée avec le cloud")
and "Synchro auto activée" or "Synchro auto activée avec le cloud "

and maybe, because the phrasing is very similar, "Synchro manuelle avec le cloud" (the icon above) could be "Synchroniser maintenant avec le cloud"
[image]
[image]

schtibal, 2021-05-30:
I don't believe that it's an incorrect language translation - I believe that the French translation implies the same as what the English does, it's just that both the text and the icon show what will happen when you select it, not what it currently is.

Ie,
When cloud sync is currently disabled, the cloud with a tick is displayed:
[image]
And the text is:
English: Disable auto cloud sync
French: Désactiver la synchro auto ...

When cloud sync is currently enabled, the cloud with a cross through it is displayed:
[image]
And the text is:
English: Enable auto cloud sync
French: Activer la synchro ...

Some could argue that it is counter intuitive that the icon shows a broken cloud when auto sync is disabled, and a tick when it is enabled.
Others could argue that pressing the broken cloud disables auto sync, and pressing the cloud with a tick enables it...

mikeller, 2026-10-18:
Hi @Ospeleo,

Thanks for flagging this and to schtibal for the clearer suggested wording. Unfortunately we can't action this as a direct code/PR change, Subsurface's translations (including the French strings for the mobile hamburger menu) are managed through Transifex, not edited directly in the repository; see CONTRIBUTING.md's translation section. A direct pull request changing the .ts/translation files wouldn't be the right path and would likely get overwritten by the next Transifex sync anyway.

If you'd like to fix this yourself, the way to do it is to join the Subsurface project on Transifex and update the French translation there with schtibal's suggested wording (or your own improvement). Closing this issue since it's not actionable as a direct code change, but very welcome to be picked up via Transifex. Please reopen if you'd like help finding the specific string(s) to update there.

closed 2026-10-18 by mikeller