subsurface/subsurface #3244: Subsurface for Fedora 34

neufeind, 2021-05-08:
Could you maybe enable a repo for Fedora 34 on OBS? It's available until Fedora 33 (and for devel).
Also I found that we had an old issue at #1967 regarding packaging of Qt private-headers. Is that resolved for newer Fedora-releases maybe?

dirkhh, 2021-05-08:
I can't find an F34 repo on OBS - not sure how to enable something I can't see...

neufeind, 2021-05-10:
Hmm, I expected it should be an available option.
https://build.opensuse.org/ says 4 days ago today "Fedora 34 and Ubuntu 2021.4 are available since a few days". So that should probably be available since a week? If not, maybe you could get in touch with Andrian (author of that news there) as to how to enable F34?

PS: Interesting to see their wiki seems to be outdated as well, since it only lists Fedora until F32. https://en.opensuse.org/openSUSE:Build_Service_supported_build_targets

neufeind, 2021-05-11:
Installed from the repo, works for me.