subsurface/subsurface #3253: Subsurface closes when starting plan function

SNAnalyst, 2021-05-23:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface terminates itself when I click "Log > Plan Dive" or when using Ctrl-L.
This happens on multiple machines and every time I try this.

### Operating system:
I tried this on two laptops, each running Windows 10 Enterprise H0H2 x64, fully up-to-date.
I also tried it on a desktop, running Windows 10 Pro 21H1 x64.

### Subsurface version:
I use the latest Subsurface version (v5.0.1) and tried it both with the 64bit and 32bit versions.
The software was downloaded from https://subsurface-divelog.org/.

### Steps to reproduce:
1. start Subsurface
2. click Ctrl-L

or
1. start Subsurface
2. Log > Plan dive

Then, after about 1 second, Subsurface terminates itself.

### Current behavior:
Subsurface terminates itself when starting the dive planner functionality.

### Expected behavior:
The expected behavior would be for Subsurface to continue working and start the planner.

atdotde, 2021-05-24:
Hmm. Of course that should not happen. Could you please try between steps 1 and 2 to press Ctrl-N to open an empty log file and if that lets you open the planner it implies that something in your log causes the crash. In that case, I would appreciate if you could send me your log (helling@atdotde.de) as .ssrf so I can track down the problem. In "Export" you can choose to anonymise your file which strips out all textual data.

SNAnalyst, 2021-05-24:
Interesting. I did what you asked and subsurface still closed itself.
I could not send you my log, since this is a fresh install where the log is still empty. So, I decided to experiment and add a dive to the log. After saving that dive to the log, the planner works.
However, as I hit Ctrl-N again to create a new log, subsurface closes again when I start the planner. 

Hence, it seems that having an empty log seems to cause/trigger the issue.

schtibal, 2021-06-03:
I just tried to recreate your problem, but I couldn't even get that far. I opened up Subsurface (Windows 10), and it opened my default dive log from cloud. I then hit CTRL-N to create a new logbook, and Subsurface closed. Not sure if the two are related.

dirkhh, 2021-06-03:
> I just tried to recreate your problem, but I couldn't even get that far. I opened up Subsurface (Windows 10), and it opened my default dive log from cloud. I then hit CTRL-N to create a new logbook, and Subsurface closed. Not sure if the two are related.
> I'll need to do a little more digging.

Are you on v5.0.0 or v5.0.1
What you are describing looks like the known bug in 5.0.0 that caused us to cut a 5.0.1 right away.

schtibal, 2021-06-03:
You are correct. I thought I had upgrade to 5.0.1 but looks like I must've got distracted after downloading the install package :)

So after upgrading from 5.0.0 to 5.0.1 I can re-produce the problem.

1. Ctrl-N for a new logbook
2. Ctrl-L to plan a dive
3. Subsurface closes

Nothing is added to `subsurface_err.log` nor to `subsurface_out.log`

The crash does not occur on my Linux version (built from current master branch)

PeterZaal, 2021-06-04:
Can reproduce with Ctrl-N, Ctrl-L.
Windows Eventlog shows an Application Error with Exception code 0xc0000005 which is Access Violation. Dumpfile shows 'Attempt to read from address 0000000000000058'.

dirkhh, 2021-06-04:
Yes, that bug in 5.0.1 is known. You have to have at least one dive in the dive log before you plan, otherwise the Windows version of Subsurface will crash :(
@atdotde - have you had a chance to fix that ? Since I may have to cut a 5.0.2 anyway to deal with the incorrect user manual that is bundled, we might as well make sure that this one is addressed.

dirkhh, 2021-06-04:
(oh - and remember... this only crashes on Windows, so you need to use a memory debugger to see the illegal memory access on Mac or Linux)

dirkhh, 2021-06-05:
Can't create useful memory debug output on either Linux or macOS. Not with current development version, not with 5.0.1.
Which makes figuring this out a bit harder - I hate Windows-only bugs

schtibal, 2021-06-06:
So I've never used DebugDiag before, and have only just started to dip my fingers into Subsurface, so am running a little bit blind.  Here's the dump that I've generated from the crash. 

It notes that it couldn't find debug symbols for subsurface.exe so the attached is of little to no use. Is there are debug build available to use that may be able to add to the diag output?

[subsurface_crash_202106061109.pdf](https://github.com/subsurface/subsurface/files/6603461/subsurface_crash_202106061109.pdf)

atdotde, 2021-06-08:
I now spend an hour with a debugger stepping through starting the planner from an empty log. Nothing is obviously wrong. But I don't have valgrind, maybe somebody experienced with it can run with with just starting subsurface on an empty log (or pressing Ctrl-N at start) and then pressing Ctrl-L?

What is not clear is where displayed_dive is being set. But my money would be on dive.c free_dive_structures() being called on it has the highest chance of being the problem. Lacking a proper stack dump of the crash on windows, maybe one could produce a windows executable that prints at the start and and end of this function to check this guess.

bstoeger, 2021-06-12:
@atdotde: can you actually reproduce the crash on Windows? Are we sure that it is in current master?

I tried valgrind, but there is nothing of note (on Linux with current master). Sadly, I don't have access to Windows. Can someone (Dirk?) produce a backtrace with a debug build? That would be mostly useful.

> What is not clear is where displayed_dive is being set.

`displayed_dive` is a global structure, therefore it should be initialized to 0 in the BSS segment. Initialization shouldn't be the problem here. Maybe there is some stale data in `displayed_dive` though?

In current master the use of `displayed_dive` was reduced considerably, so maybe it is really worth checking that out.

atdotde, 2021-06-12:
Unfortunately (or not), I don't have access to a computer running Windows.

bstoeger, 2021-06-12:
Next week I can try to run it on one of our measurement computers (to the joy of everyone). Does CI produce a debug build that gives usable backtraces?

dirkhh, 2021-06-12:
I went through this exercise with the last bug we had that was Windows only.
By default we don't create debug builds - it's a pain in the rear to get a working debug build and release build out of the same setup (at least for me) - and then the debug build didn't crash and it was all for naught.
So yes, I could try to get you a debug build of the latest developer version, but it's painful - and I'm not sure it will be helpful.

I guess the moral of the story is that I need to automate that whole thing so I have a reliable debug build that I can easily reproduce every time, right?

dirkhh, 2021-06-12:
Before trying to recreate that debug build, I tried to recreate the bug with master and I can't in a Windows VM.
So maybe this is already fixed?
I'm really confused.
Which reminds me that we really need to do a 5.0.2 release with the correct user manual bundled.
@schtibal @PeterZaal have either of you tried the latest test binary over here?
https://subsurface-divelog.org/downloads/test/subsurface-5.0.1-93-g9eb10af68fbf.exe

PeterZaal, 2021-06-13:
As said I can reproduces the issue with the official 5.0.1 with Ctrl-N, Ctrl-L on my Windows 10 Home version 20H2.
I just tried the subsurface-5.0.1-93-g9eb10af68fbf.exe from /downloads/test and this does **not** crash, so it seems already fixed.
I am so sorry I didn't try this before.

atdotde, 2021-06-13:
@SNAnalyst can you confirm what @PeterZaal says? Then we could close this issue.

SNAnalyst, 2021-06-14:
I just tried it and that version indeed does not crash. This new binary seems to solve it.

atdotde, 2021-06-14:
In that case, I will close this issue even though we do not fully understand it. One the problem resurfaces, we can of course open it again.

closed 2021-06-14 by atdotde