subsurface/subsurface #3254: Gas change is not downloaded from Suunto Eon Steel FW 2.5

KKeXX, 2021-05-25:
### Describe the issue:

- [X] Bug

### Issue long description:

Since I've upgraded the Firmware from my Suunto Eon Steel from 1.6.5 to 2.5 a gas change is not transfered to subsurface (nor subsurface-mobile)

### Operating system:
Ubuntu 20.04.2 LTS 5.4.0-73-generic # 82-Ubuntu SMP Wed Apr 14 17:39:42 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
via USB

OxygenOS 10.3.10, Android 10 on Oneplus 6 4.9.179-perf+ '# 1 Tue Apr 20 23:12:55 CEST 2021
via Bluetooth

### Subsurface version:
Subsurface 5.0.1 from ppa:subsurface/subsurface PPA 
Subsurface-mobile 3.2.1(5.0.1) from playstore

### Steps to reproduce:
Import new dive from computer

### Current behavior:
Change of gas is not detected. Multiple gases are used in parallel for the whole dive

### Expected behavior:
Change of gas should be detected. Is detected by other App (Divemate).

### Additional information:
Here is a log from libdivecomputer

```
Subsurface: v5.0.1, built with libdivecomputer v0.7.0-devel-Subsurface-NG (81f95d388019d1283f5e0b4dbed108b4ab10946a)
[21.245907] INFO: Open: vid=1493, pid=0030
[21.245913] INFO: Open: interface=0, endpoints=82,02
[21.246640] INFO: Timeout: value=5000
[21.247701] INFO: Write: size=64, data=3F1000000100000000000400000002002A007AA09C30000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[21.247707] DEBUG: cmd: size=16, data=00000100000000000400000002002A00
[21.266706] INFO: Read: size=64, data=3F3C000202007A2F000030000000475552550000000000000000000000003136303331303730333000000000000002056F054707000002056F05000400004041
[21.266716] DEBUG: rcv: size=60, data=000202007A2F000030000000475552550000000000000000000000003136303331303730333000000000000002056F054707000002056F0500040000
```

A drump from libdivecomputer results in an Error

```
Starting download from  "USBHID"
downloading only new dives
INFO: dev_info: Connecting to USB HID device

ERROR: didn't find HID device

Finishing download thread: "Fehler beim Öffnen von  Suunto (EON Steel)"
```
Someone seen this behaviour?

KKeXX, 2021-06-06:
Nobody else discovered this issue?

dirkhh, 2021-06-06:
I think very few people are diving a lot these days... maybe it's just about to start again?
Anyway, I haven't had a chance to create a dive with a gas change since updating to FW 2.5, so I have no way to test / reproduce this.

chif142, 2021-07-24:
I have the same issue with my eon steel black. On windows and on android as well.  Its possible to add the gas changes manually though. My firmware number is 2.5.168.

commit af734c81 (2021-08-19): update libdivecomputer

Add support for new gas change data from EON Steel with FW 2.5

Fixes #3254
closed 2021-08-19 by dirkhh