subsurface/subsurface #3255: Drop down menu is shifted on Fedora 34 on multiple monitors

drakche, 2021-05-26:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When installing from the package and If I have multiple monitors connected I get  the problem that the menu is offset
[image]
[image]
[image]

This doesn't not occur when I disconnect the rest of the monitors

### Operating system:

Fedora 34, Gnome, English

Dell XPS15 9500

### Subsurface version:

Subsurface 5.0.1

Yes

Official release from the OBS

### Steps to reproduce:

Have multiple monitors connected in different layout than horizontal.
[image]

### Current behavior:

When opining the menu it's offset. And not displayed under the toolbar item.

### Expected behavior:

For the menu to be displayed under the toolbar item.

### Additional information:

### Mentions:

dirkhh, 2021-05-26:
I have no idea what would cause that, but my obvious guess would be a Qt/Gnome incompatibility.
I don't have access to such an elaborate monitor setup, so I cannot reproduce this... so... 🤷🏼‍♂️

drakche, 2021-05-27:
This started after the update to Fedora 34.
The display engine is Wayland.

Also, one thing I did notice is that when running the app though appImage it works fine.

dirkhh, 2021-05-27:
Which only confirms my statement that this is a Qt/Gnome incompatibility.
Because the AppImage is linked against (and includes) a much older Qt version than is native on Fedora 34.
So my suggestion would be to file a bug against Fedora since this is a problem in the libraries they ship, not in Subsurface.

closed 2021-05-27 by dirkhh