subsurface/subsurface #3260: git clone problem on Windows (file name with a colon)

embeddedmz, 2021-06-03:
### Describe the issue:

- [ X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I couldn't clone the repo on Windows because there's a file name with a colon ":"

```
$ git clone https://github.com/subsurface/subsurface.git
Cloning into 'subsurface'...
remote: Enumerating objects: 102550, done.
remote: Counting objects: 100% (756/756), done.
remote: Compressing objects: 100% (302/302), done.
remote: Total 102550 (delta 488), reused 606 (delta 453), pack-reused 101794
Receiving objects: 100% (102550/102550), 128.27 MiB | 6.81 MiB/s, done.
Resolving deltas: 100% (77182/77182), done.
**error: invalid path 'Documentation/images/Filter:LocTemp.png'**
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/
```

### Operating system:
Windows 10

dirkhh, 2021-06-03:
I agree, definitely a bug in Windows - just not sure why you are filing this here instead of with Microsoft.
Joking aside, you can't build Subsurface on Windows, anyway, so this isn't really a major issue as far as I am concerned.

That said, it's easy enough to fix.

dirkhh, 2021-06-03:
should be fixed by 49130150cdd5

closed 2021-06-03 by dirkhh