subsurface/subsurface #3267: unable to open cloud storage "Moved cache data to <...>. Please try the operation again."

IkDuik, 2021-06-16:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
when opening the software it tries to connect to the cloud storage and want to download my list of dives. 
but it fails with the error: "Moved cache data to C:\Users\<username>\AppData\Roaming\Subsurface/cloudstorage/<reference>. Please try the operation again."
tried to close and open the program: same result
did a save from my other device to the cloud, same result

### Operating system:
running windows 10, NL version

### Subsurface version:
using latest version of the subsurface application (5.0.1)
on my phone I also use the latest version available

### Steps to reproduce:
open subsurface
file > Open cloud storage

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2021-06-16:
Windows falling over itself.
Simply quit Subsurface and remove that cloudstorage directory (using whatever tool Windows uses for things like that... file browser?)
After that things should work.

IkDuik, 2021-06-18:
Hi Dirk,

I can indeed confirm that this solved the problem

thanks,
stefan

cache data to <...>. Please try the operation again." (#3267)

dirkhh, 2021-06-18:
Thanks for reporting back - I'll close this issue

closed 2021-06-18 by dirkhh