subsurface/subsurface #3274: Build fails on macOS Big Sur

Mourp1979, 2021-07-07:
### Describe the issue:
- [x] Bug
- [x] Change request
- [x] New feature request
- [x] Discussion request

### Issue long description:
Build fails on macOS Big Sur. There seems to be a bug in the build script or the install guid is insufficient.
I followed the INSTALL guide at https://github.com/subsurface/subsurface/blob/master/INSTALL#L273
I used Homebrew to install the dependencies.

Executing `scripts/build.sh` leads to:

```
Project ERROR: Unknown module(s) in QT: location positioning
```

Using `scripts/build.sh -build-deps -desktop` ends in:

```
Undefined symbols for architecture x86_64:
  "_charp2str", referenced from:
      __wrap_read_data in ftdi1PYTHON_wrap.c.o
      __wrap_read_pins in ftdi1PYTHON_wrap.c.o
      __wrap_get_latency_timer in ftdi1PYTHON_wrap.c.o
      __wrap_get_eeprom_buf in ftdi1PYTHON_wrap.c.o
  "_str2charp_size", referenced from:
      __wrap_write_data in ftdi1PYTHON_wrap.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ 63%] Built target bitbang_ft2232
make[2]: *** [python/_ftdi1.so] Error 1
make[1]: *** [python/CMakeFiles/_ftdi1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs….
[ 68%] Built target bitbang2
[ 73%] Built target bitbang
Consolidate compiler generated dependencies of target serial_test
[ 78%] Built target serial_test
make: *** [all] Error 2
```

### Operating system:
macOS Big Sur 11.4 (English)

### Subsurface version:
Tried to compile latest `master` (7944978)

### Steps to reproduce:
Install macOS Big Sur 11.4 (English), follow steps in https://github.com/subsurface/subsurface/blob/master/INSTALL#L273

### Current behavior:
`Project ERROR: Unknown module(s) in QT: location positioning` and `ld: symbol(s) not found for architecture x86_64`

### Expected behavior:
Project compiling

atdotde, 2021-07-07:
Could you provide a bit more context above? Building on Big Sur does work for me.

Mourp1979, 2021-07-07:
> Could you provide a bit more context above? Building on Big Sur does work for me.

I added more details in the issue description. I am using SDK 11.3, is this a problem?

Mourp1979, 2021-07-07:
After I used Qt 5 instead of Qt 6 it worked.

closed 2021-07-07 by Mourp1979

glance-, 2021-07-16:
I'd suggest that we add some cmake limit to not accept Qt 6.