subsurface/subsurface #3275: Adding Bluetooth capability when compiling under and for macOS

Mourp1979, 2021-07-07:
How can I configure Subsurface/libdivecomputer to complie with Bluetooth capability?

I followed the install guide yet Subsurface crashes a few seconds after I enter the import dialog.

```
$ ./configure | grep -i blue
checking for BLUEZ... no
    Bluetooth : no
```

dirkhh, 2021-07-07:
we do not use the `libdivecomputer` bluetooth implementation, instead we use the Qt implementation.
After you switched from Qt6 (which is not supported) to Qt5, did you completely delete all build directories and start from scratch? Mixing Qt6 and Qt5 libraries will reliably lead to crashes.
So the crash may be an issue if you can reproduce it with a clean Qt5 build, but the report as stated above is invalid as this feature simply isn't used by Subsurface in the way you seem to expect it to be used.

closed 2021-07-07 by dirkhh

Mourp1979, 2021-07-07:
I did a `git clean -xfd`, hoped that was enough. To be sure I did it again, still crashes. What else could I delete?

```
Process:               Subsurface [80669]
Path:                  /Users/USER/*/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               5.0.2-11-gcfc87e9da3f5 (5.0.2-11-gcfc87e9da3f5)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Subsurface [80669]
User ID:               501

Date/Time:             2021-07-07 18:46:11.618 +0200
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)

Time Awake Since Boot: 69000 seconds
Time Since Wake:       18000 seconds

System Integrity Protection: enabled

Crashed Thread:        11  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace TCC, Code 0x0

[…]

Thread 11 Crashed:: Dispatch queue: com.apple.root.default-qos
0   libsystem_kernel.dylib        	0x00007fff203fc56e __abort_with_payload + 10
1   libsystem_kernel.dylib        	0x00007fff203fdfbd abort_with_payload_wrapper_internal + 80
2   libsystem_kernel.dylib        	0x00007fff203fdfef abort_with_payload + 9
3   com.apple.TCC                 	0x00007fff24c8c0ae __TCC_CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 163
4   com.apple.TCC                 	0x00007fff24c8c8df __TCCAccessRequest_block_invoke.130 + 498
5   com.apple.TCC                 	0x00007fff24c8a504 __tccd_send_message_block_invoke + 573
6   libxpc.dylib                  	0x00007fff20156a9c _xpc_connection_reply_callout + 36
7   libxpc.dylib                  	0x00007fff20156a24 _xpc_connection_call_reply_async + 69
8   libdispatch.dylib             	0x00007fff20265870 _dispatch_client_callout3 + 8
9   libdispatch.dylib             	0x00007fff2027e035 _dispatch_mach_msg_async_reply_invoke + 375
10  libdispatch.dylib             	0x00007fff2027551c _dispatch_kevent_worker_thread + 1390
11  libsystem_pthread.dylib       	0x00007fff2040c4a5 _pthread_wqthread + 386
12  libsystem_pthread.dylib       	0x00007fff2040b42f start_wqthread + 15
```

dirkhh, 2021-07-07:
depending on how you build, no this might not delete all the build artifacts
But looking at the crash - nothing in that call stack is Subsurface related. So not sure what to tell you.
I don't think you want the answer "it works here".
Have you tried the official binary? Does that crash as well?

Mourp1979, 2021-07-07:
I build with just the build script `./subsurface/scripts/build.sh`. I installed the dependencies with Homebrew.

The official binary 5.0.1 runs flawlessly. I also just updated to 5.0.2, which also runs without problems. My build doesn’t crash until I enter the “Import from dive computer” menu. Then, after about 5 seocnds, it crashes.

dirkhh, 2021-07-07:
Simple question 
(a) are you running `make install` or just `make`
(b) how are you starting Subsurface?

Mourp1979, 2021-07-07:
> (a) are you running `make install` or just `make`

Neither. Just running `./subsurface/scripts/build.sh` (which is probably running `make`).

> (b) how are you starting Subsurface?

I run Subsurface.app from the build folder (`open subsurface/build/Subsurface.app`)

dirkhh, 2021-07-07:
every time you build the binary, you need to run `make install`; the build script should do that for you, but if you run `make` again in the the build folder, things will crash at startup.

I don't think I have an easy way to debug what is failing for you - especially as I can't reproduce it. 
instead of trying to run open on the `Subsurface.app` folder, could you manually call the executable so you get to see any error output that it might provide before crashing?

So
```
cd subsurface/build
./Subsurface.app/Contents/MacOS/Subsurface -v
```

Also, just out of curiosity, is there a particular reason why you want to run a version built from source instead of the builds that we provide?

Mourp1979, 2021-07-07:
> instead of trying to run open on the `Subsurface.app` folder, could you manually call the executable so you get to see any error output that it might provide before crashing?

Running the executable `Subsurface.app/Contents/MacOS/Subsurface` doesn’t crash in the first place…

Terminal.app shows `starting BT/BLE discovery` as soon as I open the “Import from dive computer” menu. A few seconds later all the detected devices are listed. That must be the moment when it crashes when opened with `open subsurface/build/Subsurface.app`.

> Also, just out of curiosity, is there a particular reason why you want to run a version built from source instead of the builds that we provide?

For me the point of open source software is enabling users to build it from scratch, play around with the code and possibly contribute. That is enough for me to build from source. In addition though I want to see if I can add compatibility for the Aqualung i330R.

dirkhh, 2021-07-07:
> Running the executable `Subsurface.app/Contents/MacOS/Subsurface` doesn’t crash in the first place…
> 
> Terminal.app shows `starting BT/BLE discovery` as soon as I open the “Import from dive computer” menu. A few seconds later all the detected devices are listed. 

excellent
I wonder if calling it directly from within that folder leads to different libraries being loaded.
Can you check with `otool` to see if there are any libraries in the `Subsurface.app` folder that have dependencies to non-system libraries (i.e. things in `/opt` or `/usr/local` or in the Qt install path? All of the non-system libraries should be resolved relative to the `@rpath` or `@executable_path`?
One way to do this might be this:
```
for i in $(find Subsurface.app/Contents/ -name \*.dylib) ; do otool -L $i; done | grep -E -v '(@executable_path|@rpath)'
```
now eyeball the output to see if something looks wrong

> For me the point of open source software is enabling users to build it from scratch, play around with the code and possibly contribute. That is enough for me to build from source. In addition though I want to see if I can add compatibility for the Aqualung i330R.

Excellent reasons. Pardon my curiosity :)

Mourp1979, 2021-07-08:
> Can you check with `otool` to see if there are any libraries in the `Subsurface.app` folder that have dependencies to non-system libraries (i.e. things in `/opt` or `/usr/local` or in the Qt install [...]

Looking good. Some libraries are in `@loader_path`, `/System/Library/Frameworks` and `/usr/lib` (which should be fine, I guess?). Also excluding `@loader_path`, `/System/Library/Frameworks` and `/usr/lib` leaves only local libs:

```
Subsurface.app/Contents/PlugIns/geoservices/libqtgeoservices_googlemaps.dylib:
	libqtgeoservices_googlemaps.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/geoservices/libqtgeoservices_esri.dylib:
	libqtgeoservices_esri.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/geoservices/libqtgeoservices_mapboxgl.dylib:
	libqtgeoservices_mapboxgl.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/geoservices/libqtgeoservices_nokia.dylib:
	libqtgeoservices_nokia.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/geoservices/libqtgeoservices_itemsoverlay.dylib:
	libqtgeoservices_itemsoverlay.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/geoservices/libqtgeoservices_osm.dylib:
	libqtgeoservices_osm.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/geoservices/libqtgeoservices_mapbox.dylib:
	libqtgeoservices_mapbox.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/platforms/libqcocoa.dylib:
	libqcocoa.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/position/libqtposition_serialnmea.dylib:
	libqtposition_serialnmea.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/position/libqtposition_positionpoll.dylib:
	libqtposition_positionpoll.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/position/libqtposition_cl.dylib:
	libqtposition_cl.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/printsupport/libcocoaprintersupport.dylib:
	libcocoaprintersupport.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/styles/libqmacstyle.dylib:
	libqmacstyle.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/bearer/libqgenericbearer.dylib:
	libqgenericbearer.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/iconengines/libqsvgicon.dylib:
	libqsvgicon.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqgif.dylib:
	libqgif.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqwbmp.dylib:
	libqwbmp.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqwebp.dylib:
	libqwebp.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqico.dylib:
	libqico.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqmacheif.dylib:
	libqmacheif.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqjpeg.dylib:
	libqjpeg.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqtiff.dylib:
	libqtiff.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqsvg.dylib:
	libqsvg.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqicns.dylib:
	libqicns.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqtga.dylib:
	libqtga.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/PlugIns/imageformats/libqmacjp2.dylib:
	libqmacjp2.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/Resources/qml/QtQuick.2/libqtquick2plugin.dylib:
	libqtquick2plugin.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/Resources/qml/QtLocation/libdeclarative_location.dylib:
	libdeclarative_location.dylib (compatibility version 0.0.0, current version 0.0.0)
Subsurface.app/Contents/Resources/qml/QtPositioning/libdeclarative_positioning.dylib:
	libdeclarative_positioning.dylib (compatibility version 0.0.0, current version 0.0.0)
```

atdotde, 2021-07-08:
ah, you
open Subsurface.app/

I get the same crash here (with totally useless stack trace). But when running from the  command line (which is what I always do for self built Subsurface)
Subsurface.app/Contents/MacOS/Subsurface
which is preferable anyway since you get debugging output on the command line .

I just noticed something new in Big Sur when running like this: When starting the BT scan, a modal dialog opens asking me if it's ok if Terminal.app accesses bluetooth. Seems to be a new security feature and maybe it's that what causes the crash when opening Subsurface.app?

Mourp1979, 2021-07-08:
That’s it! The self built App is not asking for Bluetooth access but rather crashing instead.

When the executable (`Subsurface.app/Contents/MacOS/Subsurface`) is opened directly the parent process is Terminal.app (which in my case is granted Bluetooth access already, so it doesn’t crash).

After I manually allowed my self built Subsurface to use Bluetooth it didn’t crash anymore.

[image]

As this seems to only crash in debug builds, I guess it isn’t worth to fix it.

atdotde, 2021-07-08:
I am reading https://bugreports.qt.io/browse/QTBUG-90674 

You can add 
`<key>NSBluetoothPeripheralUsageDescription</key>
<string>A user application wants to access bluetooth.</string>`
to Subsurface.app/Contents/Info.plist and it will prevent the crash but it will then search forever for BT devices without finding anything. To make that work, you need to sign the app as well. Hmm.

Mourp1979, 2021-07-08:
Well, for Subsurface-mobile on iOS there is an entry for `NSBluetoothAlwaysUsageDescription` and `NSBluetoothPeripheralUsageDescription` (in `packaging/ios/Info.plist.in`):

```
<key>NSBluetoothAlwaysUsageDescription</key>
<string>Subsurface-mobile can download dive information directly from some Bluetooth LE enabled dive computers; for this it needs access to your Bluetooth peripherals.</string>
<key>NSBluetoothPeripheralUsageDescription</key>
<string>Subsurface-mobile can download dive information directly from some Bluetooth LE enabled dive computers; for this it needs access to your Bluetooth peripherals.</string>
```

The same should be done for `packaging/macosx/Info.plist.in`:

```
<key>NSBluetoothAlwaysUsageDescription</key>
<string>Subsurface can download dive information directly from some Bluetooth LE enabled dive computers; for this it needs access to your Bluetooth peripherals.</string>
<key>NSBluetoothPeripheralUsageDescription</key>
<string>Subsurface can download dive information directly from some Bluetooth LE enabled dive computers; for this it needs access to your Bluetooth peripherals.</string>
```

I guess with `NSBluetoothAlwaysUsageDescription` being set the infinite search would be fixed as well.

atdotde, 2021-07-08:
No, still scans forever. I have no experience with code signing. The simplest thing I tried
codesign -f --deep -s 'ME929L7V9A' Subsurface.app  
did not work (the google maps component causes an error).

dirkhh, 2021-07-08:
I think my signing script is part of the sources - you just need to adjust it to use your key...
https://github.com/subsurface/subsurface/blob/master/packaging/macosx/sign