subsurface/subsurface #3276: Dumpfile is not being written

Mourp1979, 2021-07-08:
I selected to save libdivecomputer logfile and dumpfile. When downloading from a currently unsupported dive computer I get a logfile stating `ERROR: Failed to receive the answer.` but no dumpfile at all. Is this supposed to happen? How else would I get the data so I or somebody else can add support for this computer?

atdotde, 2021-07-08:
Adding support for another dive computer is likely something for libdivecomputer  rather than Subsurface directly. @jefdriesen  is your man. Have a look at https://www.libdivecomputer.org

With google I found https://webcache.googleusercontent.com/search?q=cache:6osivPqG5lQJ:https://gitmemory.com/dannybarake+&cd=1&hl=de&ct=clnk&gl=de&client=safari which might be relevant to your case as well.

Mourp1979, 2021-07-08:
Thanks! This looks promising. Also found this memory dumping guide (https://www.libdivecomputer.org/contribute.html) which at first made me happy, quickly realising my endeavour adding support for the i330R might be pointless with:

- Not even the official software supporting the DC
- Transfer protocol part being windows only
- https://www.libdivecomputer.org/index.html not even listing Aqualung computers
- Download links for the test application being dead
- `/dev/ttyS0` or `/dev/tty.usbserial-xxx` not existing in my version of macOS (11.4)
- `dctool` reporting `Bad CPU type in executable` (for my 64 bit Intel CPU o_O)

Might give up on this one and let someone with more experience add the i330R.

jefdriesen, 2021-07-08:
Which dive computer did you select? For the i330R select another Aqualung model in subsurface. Most likely the download protocol hasn't changed and in that case downloading a memory dump should work just fine. You won't be able to download your dives, but that will be for the next step.

The dctool build on the website is a 32bit build, which doesn't work anymore on newer Mac versions. You can download a 64bit build directly from github actions:
https://github.com/libdivecomputer/libdivecomputer/suites/3170649892/artifacts/73027676

jefdriesen, 2021-07-08:
Forgot to mention that even if the download failed, the log may contain useful information. So also provide the logs of the failed attempts.

Mourp1979, 2021-07-08:
> Which dive computer did you select? For the i330R select another Aqualung model in subsurface. Most likely the download protocol hasn't changed and in that case downloading a memory dump should work just fine.

I tried the i300C (because it seems to be identical in terms of features), I also tried the i770R (because it has an “R” in its name). Same results for both (baudrate being 115200 for the i770R):

```
Error
Error opening the device LE:{[…]} Aqualung (i300C).
In most cases, in order to debug this issue, a libdivecomputer logfile will be useful.
You can create this logfile by selecting the corresponding checkbox in the download dialog.
```

Logfile:

```
[0.000000] INFO: Open: transport=32
[0.000765] INFO: Configure: baudrate=38400, databits=8, parity=0, stopbits=0, flowcontrol=0
[0.000768] INFO: Timeout: value=1000
[0.000768] INFO: DTR: value=1
[0.000769] INFO: RTS: value=0
[0.000769] INFO: Sleep: value=100
[0.102418] INFO: RTS: value=1
[0.102421] INFO: Sleep: value=100
[0.205410] INFO: Purge: direction=3
[0.205737] INFO: Write: size=5, data=CD40000184
[1.213074] INFO: Read: size=0, data=
[1.213127] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:680 (oceanic_atom2_packet)]
[1.213141] INFO: Sleep: value=100
[1.317160] INFO: Purge: direction=1
[1.317164] INFO: Sleep: value=1
[1.318723] INFO: Write: size=5, data=CD40000184
[2.330646] INFO: Read: size=0, data=
[2.330648] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:680 (oceanic_atom2_packet)]
[2.330650] INFO: Sleep: value=100
[2.435055] INFO: Purge: direction=1
[2.435062] INFO: Sleep: value=2
[2.437880] INFO: Write: size=5, data=CD40000184
[3.447141] INFO: Read: size=0, data=
[3.447144] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:680 (oceanic_atom2_packet)]
```

jefdriesen, 2021-07-08:
The error "Error opening the device LE:{[…]} Aqualung (i300C)." indicates that setting up the bluetooth connection already fails. I assume that also explains why in the libdivecomputer log there is no response from the dive computer. Can you try on another system? BLE often works more reliable on mobile (iOS, Android).

Mourp1979, 2021-07-08:
The iOS version‘s App log gives me a similar message `Error opening the device LE:{[…]} Aqualung (i300C)` and then something about sending it to developers via some about dialog I can’t find. Copy and paste is disabled in the log view. I could not find an option to enable libdivecomputer logfile in the iOS build. Does that mean that the communication protocol might have changed?

Mourp1979, 2021-07-08:
When I connect the computer via USB there appears a TTY named `/dev/tty.usbmodemFFFFFFFEFFFF1`. Using this as “Device or mount point” and unchecking “Choose Bluetooth download mode” and then hitting “Download” results in a very similar log file:

```
[0.000000] INFO: Open: name=/dev/tty.usbmodemFFFFFFFEFFFF1
[0.004054] INFO: Configure: baudrate=38400, databits=8, parity=0, stopbits=0, flowcontrol=0
[0.005153] INFO: Timeout: value=1000
[0.005156] INFO: DTR: value=1
[0.005184] INFO: RTS: value=0
[0.005397] INFO: Sleep: value=100
[0.108440] INFO: RTS: value=1
[0.108712] INFO: Sleep: value=100
[0.212877] INFO: Purge: direction=3
[0.213348] INFO: Write: size=1, data=84
[1.217868] INFO: Read: size=9, data=EE0008000030058CB8
[1.217901] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:680 (oceanic_atom2_packet)]
[1.217921] INFO: Sleep: value=100
[1.317954] INFO: Purge: direction=1
[1.318023] INFO: Sleep: value=1
[1.319440] INFO: Write: size=1, data=84
[2.319638] INFO: Read: size=9, data=EE0008000030058CB8
[2.319640] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:680 (oceanic_atom2_packet)]
[2.319642] INFO: Sleep: value=100
[2.420610] INFO: Purge: direction=1
[2.420675] INFO: Sleep: value=2
[2.423305] INFO: Write: size=1, data=84
[3.423400] INFO: Read: size=9, data=EE0008000030058CB8
[3.423402] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:680 (oceanic_atom2_packet)]
```

With the error dialog “Error Unable to open /dev/tty.usbmodemFFFFFFFEFFFF1 Aqualung (i300C)”.

This seems to eliminate a Bluetooth connection error. To me this more looks like the protocol has changed.

Mourp1979, 2021-07-09:
Is there anything else I could try before abandoning the little guy? Like…how would I find out about that protocol?

Mourp1979, 2021-07-09:
Using Bluetooth scanners I see one custom read service. Reading from it return nothing. Is that something to expect when there are no dives on the computer? Others returned 0x00 in this case.

Mourp1979, 2021-07-09:
I also found that there is some UUID for other Pelagic dive computers in qt-ble.cpp (cb3c4555-d670-4670-bc20-b61dbc851e9a). Don’t know much about Bluetooth.

Mourp1979, 2021-07-09:
Another interesting thing I found is this commit:

https://github.com/subsurface/libdc/commit/8b53c70

There seems to be something odd about the i770R, may be the same or a similar thing is odd about the i330R?

https://github.com/subsurface/libdc/commit/8b53c70#diff-289b7084a29b9b5bf08649bed7bac289698b9aa3c94526469d61ef943ba71d22

I tried to add I330R to `device->extra = model == PROPLUSX || model == I770R;` (and add the i330R to all the other places I could find, copying the i300C), didn’t help.

jefdriesen, 2021-07-09:
Do you see with the scanner the same bluetooth service uuid as the one in the code? Can you export the bluetooth services and characteristics, so we can have a look too?

The exceptions in the code only matter after the initial handshaking. We don't get that far yet.

The best way to check whether the protocol really changed is to download with the aqualung application and capture the bluetooth communication. But if the device isn't supported yet in their own app, that's not really an option. Can you check with aqualung? Maybe there is already a version that supports the i330R?

Note that I'm currently travelling, so I won't be able to respond as quickly during the next two weeks.

Mourp1979, 2021-07-11:
> Do you see with the scanner the same bluetooth service uuid as the one in the code?

It is a different one.

> Can you export the bluetooth services and characteristics, so we can have a look too?

I don’t see any export options in the Apps I used. There are only **2 services** anyways:

- **Device Information** (0x180A), with 5 characteristics:
    - Manufacturer Name String
    - Model Number String
    - Hardware Revision String
    - Firmware Revision String
    - PnP ID
- **CA7B0001-F785-4C38-B599-C7C5FBADB034**, with 2 characteristics:
    - CA7B0001-F785-4C38-B599-C7C5FBADB034 (**read**, notify)
        - Client Characteristics Configuration {length = 2, bytes = 0x0000}
    - CA7B0001-F785-4C38-B599-C7C5FBADB034 (write without response, **write**)

> The best way to check whether the protocol really changed is to download with the aqualung application and capture the bluetooth communication. But if the device isn't supported yet in their own app, that's not really an option.

The iPhone App already supports the i330R, it is the macOS App that doesn’t yet. Guess there is no (easy) way to capture the Bluetooth communication on an iPhone? Would consider jailbraking if it helps.

> Note that I'm currently travelling, so I won't be able to respond as quickly during the next two weeks.

Have fun and thanks for replying anyways! :)

jefdriesen, 2021-07-17:
Can you try starting subsurface from the command-line? It should print some debug info on stdout/stderr, including the discovered services, and the services it tries to connect with. That will already tell us whether it tries to use those unknown services or just refuses to continue.

Mourp1979, 2021-07-28:
Sorry for the late reply, did some traveling as well. I returned the computer, so can’t debug the connection any further. Hopefully the research done so far will help somebody with adding support for the i330R.

closed 2021-07-28 by dirkhh