subsurface/subsurface #3278: Renumber causing crash of 5.0.2 -

nachumk, 2021-07-13:
Using 5.0.2 on MacOS 11.4. Crash is reproducible every time. Lost tons of information as well as it was cloud synced and I had edited 36 dives before this happened - first time using the app...

I had 36 dives already in (imported from dive watch), updated all of those, and then began adding old ones before we got our computers. When I tried to renumber all of them starting from 1 the app crashed. No option for recovery either.

Renumbering all of the watch imported ones starting from 2 works fine. Renumbering the first dive to #3277 1 crashes the app.

```
Process:               Subsurface [3041]
Path:                  /Applications/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               5.0.2 (5.0.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Subsurface [3041]
User ID:               501

Date/Time:             2021-07-13 13:29:20.674 -0500
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        FEF00F8D-6E7B-81AB-AEC3-575C71AC0145

Sleep/Wake UUID:       34BE3E33-0E4A-4CA7-B469-8BFF7EF53BF5

Time Awake Since Boot: 14000 seconds
Time Since Wake:       2100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [3041]

VM Regions Near 0x10:
--> 
    __TEXT                      10544d000-105905000    [ 4832K] r-x/r-x SM=COW  /Applications/Subsurface.app/Contents/MacOS/Subsurface

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.subsurface-divelog        	0x0000000105682f2c get_plot_details_new + 236
1   org.subsurface-divelog        	0x00000001055713f3 ToolTipItem::refresh(dive const*, QPointF const&, bool) + 179
2   org.subsurface-divelog        	0x000000010557a807 ProfileWidget2::mouseMoveEvent(QMouseEvent*) + 279
3   org.qt-project.QtWidgets      	0x000000010a1b3e27 QWidget::event(QEvent*) + 487
4   org.qt-project.QtWidgets      	0x000000010a25c58d QFrame::event(QEvent*) + 45
5   org.qt-project.QtWidgets      	0x000000010a4d7b07 QGraphicsView::viewportEvent(QEvent*) + 1191
6   org.qt-project.QtCore         	0x000000010b040ceb QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) + 155
7   org.qt-project.QtWidgets      	0x000000010a1799d5 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 245
8   org.qt-project.QtWidgets      	0x000000010a17c755 QApplication::notify(QObject*, QEvent*) + 6965
9   org.qt-project.QtCore         	0x000000010b040a34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
10  org.qt-project.QtWidgets      	0x000000010a17a310 QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 896
11  org.qt-project.QtWidgets      	0x000000010a1d3022 QWidgetWindow::handleMouseEvent(QMouseEvent*) + 3266
12  org.qt-project.QtWidgets      	0x000000010a1d16e9 QWidgetWindow::event(QEvent*) + 233
13  org.qt-project.QtWidgets      	0x000000010a1799ea QApplicationPrivate::notify_helper(QObject*, QEvent*) + 266
14  org.qt-project.QtWidgets      	0x000000010a17ae11 QApplication::notify(QObject*, QEvent*) + 497
15  org.qt-project.QtCore         	0x000000010b040a34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
16  org.qt-project.QtGui          	0x000000010a7e6e2e QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 3534
17  org.qt-project.QtGui          	0x000000010a7cbd7b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 219
18  libqcocoa.dylib               	0x000000010d61a260 QCocoaEventDispatcherPrivate::processPostedEvents() + 320
19  libqcocoa.dylib               	0x000000010d61a9c8 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 40
20  com.apple.CoreFoundation      	0x00007fff206bd97c __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
21  com.apple.CoreFoundation      	0x00007fff206bd8e4 __CFRunLoopDoSource0 + 180
22  com.apple.CoreFoundation      	0x00007fff206bd664 __CFRunLoopDoSources0 + 242
23  com.apple.CoreFoundation      	0x00007fff206bc08c __CFRunLoopRun + 893
24  com.apple.CoreFoundation      	0x00007fff206bb64c CFRunLoopRunSpecific + 563
25  com.apple.HIToolbox           	0x00007fff288f9ab3 RunCurrentEventLoopInMode + 292
[... 560 lines cut ...]
Kernel Alloc Once                    8K        1 
MALLOC                           219.6M      110 
MALLOC guard page                   32K        7 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
OpenGL GLSL                        256K        3 
STACK GUARD                       56.0M       13 
Stack                             29.1M       13 
VM_ALLOCATE                        224K       38 
__DATA                            17.5M      404 
__DATA_CONST                      19.0M      210 
__DATA_DIRTY                       797K      118 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       526.8M       76 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2480K        2 
__TEXT                           328.1M      399 
__UNICODE                          588K        1 
mapped file                      554.2M       44 
shared memory                      760K       16 
===========                     =======  ======= 
TOTAL                              2.2G     1544 
TOTAL, minus reserved VM space     1.8G     1544 

```

atdotde, 2021-07-13:
Could you email me your logbook in the state just before renumbering? Without it, I could not reproduce the crash. In File-Export you would have the choice to anonymise your dive (removing all text entries). My email is helling@atdotde.de

nachumk, 2021-07-13:
Hi Robert,

Here you go:

BTW, I fixed the original numbers by exporting to ssrf and then manually editing. So to get it to crash, just select the bottom 4 in this logbook and renumber starting from 2. It crashes immediately.

Also - the app has become super-unstable. Just closes without any reason at random times (seems like it might be when switching back and forth).

Thanks,
Noah

nachumk, 2021-07-13:
For the second type of crash, here’s a crash report - might be the same issue occurring two different ways:

Process:               Subsurface [3409]
Path:                  /Applications/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               5.0.2 (5.0.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Subsurface [3409]
User ID:               501

Date/Time:             2021-07-13 14:01:13.747 -0500
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        FEF00F8D-6E7B-81AB-AEC3-575C71AC0145

Sleep/Wake UUID:       34BE3E33-0E4A-4CA7-B469-8BFF7EF53BF5

Time Awake Since Boot: 16000 seconds
Time Since Wake:       4000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [3409]

VM Regions Near 0x10:
--> 
    __TEXT                      10c828000-10cce0000    [ 4832K] r-x/r-x SM=COW  /Applications/Subsurface.app/Contents/MacOS/Subsurface

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.subsurface-divelog        	0x000000010ca5df2c get_plot_details_new + 236
1   org.subsurface-divelog        	0x000000010c94c3f3 ToolTipItem::refresh(dive const*, QPointF const&, bool) + 179
2   org.subsurface-divelog        	0x000000010c955807 ProfileWidget2::mouseMoveEvent(QMouseEvent*) + 279
3   org.qt-project.QtWidgets      	0x000000011130fe27 QWidget::event(QEvent*) + 487
4   org.qt-project.QtWidgets      	0x00000001113b858d QFrame::event(QEvent*) + 45
5   org.qt-project.QtWidgets      	0x0000000111633b07 QGraphicsView::viewportEvent(QEvent*) + 1191
6   org.qt-project.QtCore         	0x000000011219cceb QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) + 155
7   org.qt-project.QtWidgets      	0x00000001112d59d5 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 245
8   org.qt-project.QtWidgets      	0x00000001112d8755 QApplication::notify(QObject*, QEvent*) + 6965
9   org.qt-project.QtCore         	0x000000011219ca34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
10  org.qt-project.QtWidgets      	0x00000001112d6310 QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 896
11  org.qt-project.QtWidgets      	0x000000011132f022 QWidgetWindow::handleMouseEvent(QMouseEvent*) + 3266
12  org.qt-project.QtWidgets      	0x000000011132d6e9 QWidgetWindow::event(QEvent*) + 233
13  org.qt-project.QtWidgets      	0x00000001112d59ea QApplicationPrivate::notify_helper(QObject*, QEvent*) + 266
14  org.qt-project.QtWidgets      	0x00000001112d6e11 QApplication::notify(QObject*, QEvent*) + 497
15  org.qt-project.QtCore         	0x000000011219ca34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
16  org.qt-project.QtGui          	0x0000000111942e2e QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 3534
17  org.qt-project.QtGui          	0x0000000111927d7b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 219
18  libqcocoa.dylib               	0x0000000114a18260 QCocoaEventDispatcherPrivate::processPostedEvents() + 320
19  libqcocoa.dylib               	0x0000000114a189c8 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 40
20  com.apple.CoreFoundation      	0x00007fff206bd97c __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
21  com.apple.CoreFoundation      	0x00007fff206bd8e4 __CFRunLoopDoSource0 + 180
22  com.apple.CoreFoundation      	0x00007fff206bd664 __CFRunLoopDoSources0 + 242
23  com.apple.CoreFoundation      	0x00007fff206bc08c __CFRunLoopRun + 893
24  com.apple.CoreFoundation      	0x00007fff206bb64c CFRunLoopRunSpecific + 563
25  com.apple.HIToolbox           	0x00007fff288f9ab3 RunCurrentEventLoopInMode + 292
26  com.apple.HIToolbox           	0x00007fff288f96e6 ReceiveNextEventCommon + 284
27  com.apple.HIToolbox           	0x00007fff288f95b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
28  com.apple.AppKit              	0x00007fff22ebc6f2 _DPSNextEvent + 864
29  com.apple.AppKit              	0x00007fff22ebaec5 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
30  com.apple.AppKit              	0x00007fff22ead239 -[NSApplication run] + 586
[... 380 lines cut ...]
    0x7fff2befa000 -     0x7fff2bf01fff  com.apple.GPUWrangler (6.3.3 - 6.3.3) <768299B7-C4B5-307D-A413-52DA0C269A9D> /System/Library/PrivateFrameworks/GPUWrangler.framework/Versions/A/GPUWrangler
    0x7fff2bf02000 -     0x7fff2bf05fff  libRadiance.dylib (2130.5.4) <5C83C72F-9F7B-341D-ADD7-DA12A728C9EA> /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libRadiance.dylib
    0x7fff2bf06000 -     0x7fff2bf0bfff  com.apple.DSExternalDisplay (3.1 - 380) <AA11B104-262F-33B2-8564-EF47D24AC2B6> /System/Library/PrivateFrameworks/DSExternalDisplay.framework/Versions/A/DSExternalDisplay
[... cut ...]
    0x7fff2d17b000 -     0x7fff2d25afff  libSMC.dylib (20) <7B4581C7-3E3F-33F7-AC74-BFC57A2991C4> /usr/lib/libSMC.dylib
    0x7fff2d25b000 -     0x7fff2d2bafff  libcups.2.dylib (494.1) <C96214CD-19F2-334A-95A0-25BA714D984A> /usr/lib/libcups.2.dylib
    0x7fff2d2bb000 -     0x7fff2d2cafff  com.apple.LangAnalysis (1.7.0 - 254) <C53922F5-BD54-3594-9DCF-DF6D0379B40D> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/LangAnalysis.framework/Versions/A/LangAnalysis
    0x7fff2d2cb000 -     0x

atdotde, 2021-07-13:
I think you have to mail it directly as GitHub eats the attachment. 

The stack dump suggests it's crashing when composing the info box in the profile, in both cases (that's what get_plot_details_new does), maybe it's something there, maybe a strange event or something.

nachumk, 2021-07-13:
What’s your email? Or where should I mail it to?

atdotde, 2021-07-13:
helling@atdotde.de

atdotde, 2021-07-13:
Sorry to say: It does not crash for me.

Can you confirm it crashes for you with this text-stripped version? Otherwise it could be something in the text, maybe a language (character) thing?

What I tried: Load the logbook, select the four oldest dives (numbered 1--4), right click, select "renumber dive(s) from the context menu), set the counter to 2 in the modal dialog.

For me, it just renumbers the dives. 

Strange.

nachumk, 2021-07-13:
WIth the debug.ssrf, it crashes immediately for me when renumbering the first 4 starting from 2. What else can we check? Are you on the same MacOS version? I also have homebrew installed, perhaps something incompatible with QT libraries?

Noah

nachumk, 2021-07-13:
I also changed to imperial from metric, although just changed back and still crashes.

atdotde, 2021-07-13:
Now I managed to crash mine as well. I am now running the 5.0.2 version from the subsurface server on an old MBP with Big Sur.

atdotde, 2021-07-13:
I am a bit lost: I can reliably crash the official 5.0.1 and 5.0.2 binaries but those don't have debugging symbols. I cannot crash the binary I build here. Maybe @dirkhh has an idea?

atdotde, 2021-07-13:
Ah, I have an idea: It could be that I have already fixed this (in a commit after the 5.0.2 release): There was a problem with dives that did not have any depth samples (like your dives 1-4). Could you please try version 5.0.2-12 from the daily builds on 
https://subsurface-divelog.org/downloads/daily/ ?

nachumk, 2021-07-13:
That did fix it. Thank you!

Next time I’ll download the latest before filing - should’ve done that this time.

Noah

atdotde, 2021-07-13:
That wasn't obvious. I am glad we could solve this. Bugs that seem to evade the debugger can be such a pain.

closed 2021-07-13 by atdotde

dirkhh, 2021-07-13:
> The stack dump suggests it's crashing when composing the info box in the profile, in both cases (that's what get_plot_details_new does), maybe it's something there, maybe a strange event or something.

Both crashes seem the same -- I love reading optimized assembly code to figure out what's going on, but it appears that we are dereferencing the cylinder returned to us without checking that it's real. In the release build `plot_string` is inlined with `get_plot_details_new` and the crash happens here https://github.com/subsurface/subsurface/blob/235ee4803f9ffc3108e1e00b51afeeb86e8d95a6/core/profile.c#L1427