subsurface/subsurface #3281: Subsurface does not give the same results as DiveProMe+ (on Android). Is the latest code used ?

OlivierV78, 2021-07-16:
Subsurface does not give the same results as DiveProMe+ (on Android) with same Buhlmann ZHL-C
Where is the mistake ?
Alexey Vlasov, the developper of DiveProMe+, says Subsurface (and MultiDeco too) use an old code.  

### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

Subsurface does not give the same results as DiveProMe+ (on Android) and in a email discussion with the developper of DiveProMe+ (Alexey Vlasov) he said to me that perhaps Subsurface (and MultyDeco to) use old code from David Yount, Eric Maiken and Erik Baker. 

### Operating system:

Ubuntu 20.04

### Subsurface version:

5.0.2

### Steps to reproduce:

The example is a sample dive of 50 meters, 20 minutes and 80 % O2 decompression, Buhlmann ZHL-C GF 79/80 with 30 m/min down speed, 10 m/min up speed, las stop at 6 m. down to first deep 

Subsurface (and MultiDeco too) gives **43 minutes** runtime with **first stop at 12 meters**.
But DiveProMe+ gives **46 minutes** with fist stop at **15 meters**.

### Current behavior:

### Expected behavior:

### Additional information:

DiveProMe : https://play.google.com/store/apps/details?id=com.diveprome.avlasov.diveprome&hl=fr&gl=US

### Mentions:

atdotde, 2021-07-16:
Then Messieurs Yount, Maiken and Baker should fix the code in Subsurface. 

Seriously, I was going to reply that in https://thetheoreticaldiver.org/wordpress/index.php/2017/11/02/why-is-buhlmann-not-like-buhlmann/ you should not lose sleep because two different programs that both claim to compute "the Bühlmann" algorithm give somewhat different results.

But then I googled for DiveProMe+ and found this: https://www.scubaboard.com/community/threads/divepro-me-online-decompression-planner.546503/ 

Turns out, that program does not claim do compute a Bühlmann schedule. It does some machine learning voodoo that was trained on other implementations of Bühlmann. So it's a different model. Why do you expect it gives the same schedules?

Best
Robert

PS: Three minutes runtime is well within the model uncertainties, I don't think the two schedules have any noticeable difference in risk of DCS.

PPS:  I hope your gradient factors and choice of deco gas are for demonstration purposes and not intended to be dived.

OlivierV78, 2021-07-16:
Thank you for your answer.

I'm newby in Buhlmann. I used usually RGBM before and I wondered about this difference.
In DiveProMe I chose ZHL-C GF 79/80, so I thought this should be the used algorithm...
And for me the same algorithm with the same parameters should give the same results (Subsurface and Multideco give the same results with same first stop and only a very few difference).
But you give a explanation of the strange behaviour. Thank you. 

PS : In France we dive up to 60 meters with air and GF 80/80 (some people use GF 85/85)... (I not other people does not do so...). GF 80/80 gives approximately same deco time as Suunto RGBM. But this is not the subject.

OlivierV78, 2021-07-18:
Answer from Alexey Vlasov to my email : 
DiveProMe and DiveProMe+ is different projects. Probably it is my mistake. This projects nedd totaly different names. DiveProMe use mashine learning algorythms. **DiveproMe+ totaly different project. Use classic Bühlmann**. All is simple :)

closed 2021-07-28 by dirkhh