subsurface/subsurface #3295: Sync not working/settings missing

Mourp1979, 2021-07-28:
### Description:
I don’t want to use the cloud to sync and rather keep my data local. So, I was happy finding out that Subsurface supports local sync. I can’t find the settings for it though (or some kind of discovery of other local Subsurface instances). Is it supposed to be a process working out of the box (i.e. showing a message box if an instance is discovered)? I searched the documentation and couldn’t find any hints on how to set up or use local sync.

[image]

### Operating system:
iOS/macOS

### Subsurface version:
iOS: 3.2.1 (5.0.1)
macOS: 5.0.2

atdotde, 2021-07-28:
The only way the mobile app can sync your logbook is via the Subsurface cloud (which is git inside). In case there is no network connection, the data is saved to the device only (which is the "local sync") but you cannot get this data out. Only the desktop software is able to read and write regular files (in particular xml files) that you can then transport by whatever means you use to transfer files. But the mobile version cannot read or write those.

You have two options: You compile your own mobile version (it's open source software after all) where you make it communicate with another git server rather than the Subsurface cloud.

Or you use the Subsurface cloud. There is the standing promise that nobody will look at your log data unless you give explicit permission for debugging purposes. Obviously, this is only a promise. But on the other hand, if you don't trust us and @dirkhh in particular who runs the server, you should by the same token not trust the app since you cannot be sure that it does not leak your  data to places you don't intend. Said differently: by using the app you have already decided to trust us, since we control what the app does with your data at the end of the day.

Honestly, use the cloud service. This is the only way that with a limited amount of work for support we can ensure that the transfer works and you don't lose data.

closed 2021-07-28 by atdotde

Mourp1979, 2021-08-04:
> The only way the mobile app can sync your logbook is via the Subsurface cloud (which is git inside). In case there is no network connection, the data is saved to the device only (which is the "local sync") but you cannot get this data out.

Sorry for misunderstanding. Maybe a wording like “changes not synced” could avoid confusion.

> […] if you don't trust us and @dirkhh in particular who runs the server, you should by the same token not trust the app since you cannot be sure that it does not leak your data to places you don't [...]

It’s not about mistrusting any of the Subsurface developers. It is the black hat hackers I don’t trust. The less places I distribute my data to the less chance of somebody abusing it. Using the Cloud on top of the App in fact does increase risk (even if I host the cloud on my own server).

> You have two options: You compile your own mobile version (it's open source software after all) where you make it communicate with another git server rather than the Subsurface cloud.
> Or you use the Subsurface cloud.

The first option sounds good. I didn’t know that the Cloud actually is a plain Git server. I’d like to set up a local Git server (maybe on my phone) and use this in place of an external/remote one. I assume I would have to exchange some URL somewhere in the code? I checked and couldn’t find the line. I would appreciate if you could give me some hints.

dirkhh, 2021-08-04:
> Sorry for misunderstanding. Maybe a wording like “changes not synced” could avoid confusion.

But they are synced locally. Because of the storage mechanism used, every time you store things you actually sync them with what is in storage - the local storage is using the exact same technology as the remote storage. We could change it to something like "changes only synced locally" - but now we are going to start running out of space.

> It’s not about mistrusting any of the Subsurface developers. It is the black hat hackers I don’t trust. The less places I distribute my data to the less chance of somebody abusing it. Using the [...]

My recommendation would be that you use a paper logbook that you keep in a secure location that only you have access to.
It is extremely easy for the black hat hacker targeting your dive log to extract that from your phone.

Oh, you don't think they'd attack your phone? Then why do you think they'd attack my server?

Anyway, please don't use Subsurface.

> The first option sounds good. I didn’t know that the Cloud actually is a plain Git server. I’d like to set up a local Git server (maybe on my phone) and use this in place of an external/remote one. [...]

You already have a local git storage on your phone. That's the whole point.

There is, however, no point to this conversation.