subsurface/subsurface #3301: Right click on trip header causes a crash

mturkia, 2021-08-11:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Crash when right-clicking on a trip header. https://groups.google.com/g/subsurface-divelog/c/5uULC0msOtA/m/w3gtlm8GAgAJ

### Operating system:

Windows 10
Linux (Ubuntu 20.04)

### Subsurface version:

5.0.2

(Also crashed on self compiled version of bstoeger-membuffer, that I happened to have on disk)

### Steps to reproduce:

Right click on a collapsed trip header, might have to repeat this on next collapsed trip.

### Current behavior:

Crash

### Expected behavior:

Context menu pops up

### Additional information:

Thread 1 "subsurface" received signal SIGSEGV, Segmentation fault.
0x0000555555785c37 in DiveListView::contextMenuEvent (this=0x5555567af890, event=<optimized out>) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qlist.h:803
803	Q_OUTOFLINE_TEMPLATE void QList<T>::detach_helper()
(gdb) bt
#0  0x0000555555785c37 in DiveListView::contextMenuEvent(QContextMenuEvent*) (this=0x5555567af890, event=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qlist.h:803
#1  0x00007ffff72682b6 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007ffff7315d52 in QFrame::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007ffff749a482 in QAbstractItemView::viewportEvent(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff750862f in QTreeView::viewportEvent(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007ffff5b8364b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff7225a55 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff722fb93 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff5b8393a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff728494a in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff72871ec in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff7225a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff722f0f0 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff5b8393a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff671b7d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#15 0x00007ffff671d10b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#16 0x00007ffff66f735b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5from
#17 0x00007fffebee432e in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#18 0x00007ffff44fd17d in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007ffff44fd400 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007ffff44fd4a3 in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff5bdb565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff5b824db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff5b8a246 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00005555556fa604 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /home/mturkia/source/subsurface/subsurface-desktop-main.cpp:110

### Mentions:

@bstoeger

bstoeger, 2021-08-11:
It crashes in `if (expanded_nodes > 1 && d->divetrip)`. If we click on a trip header it seems logical that `d` is null, isn't it?

@dirkhh: You signed the commit e42fc1a1e9a13c77d3474dbcb26b68b8772b8c6d where this was introduced. I'm not sure what the point of this addition was.

Perhaps this should read as `if (expanded_nodes > 1 && (!d || d->divetrip))`?

bstoeger, 2021-08-11:
That said, my backtrace looks different, so perhaps this is a different crash condition?

mturkia, 2021-08-11:
> It crashes in `if (expanded_nodes > 1 && d->divetrip)`. If we click on a trip header it seems logical that `d` is null, isn't it?
> 
> @dirkhh: You signed the commit [e42fc1a](https://github.com/subsurface/subsurface/commit/e42fc1a1e9a13c77d3474dbcb26b68b8772b8c6d) where this was introduced. I'm not sure what the point of this addition was.
> 
> Perhaps this should read as `if (expanded_nodes > 1 && (!d || d->divetrip))`?

This seems to fix the crash for me.

jimbodude, 2021-08-12:
I assume this is known, but in case it is helpful - same behavior on Mac.

```
Process:               Subsurface [66417]
Path:                  /Applications/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               5.0.2 (5.0.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Subsurface [66417]
User ID:               502

Date/Time:             2021-08-08 21:48:27.936 -0400
OS Version:            macOS 11.5.1 (20G80)
Report Version:        12
Bridge OS Version:     5.5 (18P4759a)
Anonymous UUID:        5AE777DA-62D1-34AE-05DC-F6582FC7D1C9

Sleep/Wake UUID:       B3301A9D-84AC-4D89-A60E-E38C9EB5B0DB

Time Awake Since Boot: 380000 seconds
Time Since Wake:       1400 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [66417]

VM Regions Near 0:
--> 
    __TEXT                      10521f000-1056d7000    [ 4832K] r-x/r-x SM=COW  /Applications/Subsurface.app/Contents/MacOS/Subsurface

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.subsurface-divelog        	0x000000010527003d DiveListView::contextMenuEvent(QContextMenuEvent*) + 4333
1   org.qt-project.QtWidgets      	0x0000000109b46346 QWidget::event(QEvent*) + 1798
2   org.qt-project.QtWidgets      	0x0000000109bee58d QFrame::event(QEvent*) + 45
3   org.qt-project.QtWidgets      	0x0000000109d77ec9 QAbstractItemView::viewportEvent(QEvent*) + 1417
4   org.qt-project.QtWidgets      	0x0000000109de5225 QTreeView::viewportEvent(QEvent*) + 517
5   org.qt-project.QtCore         	0x000000010aa7eceb QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) + 155
6   org.qt-project.QtWidgets      	0x0000000109b0b9d5 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 245
7   org.qt-project.QtWidgets      	0x0000000109b0df57 QApplication::notify(QObject*, QEvent*) + 4919
8   org.qt-project.QtCore         	0x000000010aa7ea34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
9   org.qt-project.QtWidgets      	0x0000000109b6534a QWidgetWindow::handleMouseEvent(QMouseEvent*) + 4074
10  org.qt-project.QtWidgets      	0x0000000109b636e9 QWidgetWindow::event(QEvent*) + 233
11  org.qt-project.QtWidgets      	0x0000000109b0b9ea QApplicationPrivate::notify_helper(QObject*, QEvent*) + 266
12  org.qt-project.QtWidgets      	0x0000000109b0ce11 QApplication::notify(QObject*, QEvent*) + 497
13  org.qt-project.QtCore         	0x000000010aa7ea34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
14  org.qt-project.QtGui          	0x000000010a178e2e QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 3534
15  org.qt-project.QtGui          	0x000000010a15dd7b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 219
16  libqcocoa.dylib               	0x000000010d4c4260 QCocoaEventDispatcherPrivate::processPostedEvents() + 320
17  libqcocoa.dylib               	0x000000010d4c49c8 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 40
18  com.apple.CoreFoundation      	0x00007fff205b094c __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
19  com.apple.CoreFoundation      	0x00007fff205b08b4 __CFRunLoopDoSource0 + 180
20  com.apple.CoreFoundation      	0x00007fff205b0634 __CFRunLoopDoSources0 + 242
21  com.apple.CoreFoundation      	0x00007fff205af05c __CFRunLoopRun + 893
22  com.apple.CoreFoundation      	0x00007fff205ae61c CFRunLoopRunSpecific + 563
23  com.apple.HIToolbox           	0x00007fff287f3a83 RunCurrentEventLoopInMode + 292
24  com.apple.HIToolbox           	0x00007fff287f36b6 ReceiveNextEventCommon + 284
25  com.apple.HIToolbox           	0x00007fff287f3583 _BlockUntilNextEventMatchingListInModeWithFilter + 70
26  com.apple.AppKit              	0x00007fff22db5502 _DPSNextEvent + 864
27  com.apple.AppKit              	0x00007fff22db3cd5 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
28  com.apple.AppKit              	0x00007fff22da6049 -[NSApplication run] + 586
29  libqcocoa.dylib               	0x000000010d4c362f QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2495
[... 512 lines cut ...]
    0x7fff47a34000 -     0x7fff47a74fff  com.apple.osanalytics.OSAnalytics (1.0 - 1) <B1623916-5735-37BC-B424-0EFAD8662D74> /System/Library/PrivateFrameworks/OSAnalytics.framework/Versions/A/OSAnalytics
    0x7fff516a8000 -     0x7fff516d5fff  com.apple.HelpData (2.3 - 201) <A90FB226-6924-363C-BCF9-7E9D0EF685A0> /System/Library/PrivateFrameworks/HelpData.framework/Versions/A/HelpData
    0x7fff516d6000 -     0x7fff516f0fff  com.apple.shortcut (2.16 - 118) <B6CEE6FC-A5ED-3ECE-B1B0-5E1C05A2945B> /System/Library/PrivateFrameworks/Shortcut.framework/Versions/A/Shortcut
    0x7fff516f1000 -     0x7fff51790fff  com.apple.Symbolication (12.5 - 64544.70.1) <3A006E5D-FE72-3D34-B887-4EF7824B5A4B> /System/Library/PrivateFrameworks/Symbolication.framework/Versions/A/Symbolication
[... cut ...]
    0x7fff6e305000 -     0x7fff6e30cfff  com.apple.agl (3.3.4 - AGL-3.3.3) <C7C621A8-A60D-3486-A4E6-74A1D696A369> /System/Library/Frameworks/AGL.framework/Versions/A/AGL
    0x7fff6f584000 -     0x7fff6f58efff  com.apple.AccessibilityBundles (1.0 - 174) <06BB02D5-50F4-359B-AA41-42D08FE267CD> /System/Library/PrivateFrameworks/AccessibilityBundles.framework/Versions/A/AccessibilityBundles
    0x7fff77a37000 -     0x7fff77a3efff  libRosetta.dylib (203.58) <A7FFAFB5-B0AB-312E-8EB6-FAF56D76FB7D> /usr/lib/libRosetta.dylib
```

PR #3303 (2021-08-12): desktop: fix crash when right-clicking of trip headers
Commit e42fc1a1e9a13c77d3474dbcb26b68b8772b8c6d introduced a
crash condition. Apparently the code attempts to test whether
the clicked-on item is a top-level dive. The "Collapse others"
menu item should not be shown in that case. It does this by
testing "d->divetrip". However, "d" might quite logically be
null if clicking on an unexpanded trip header.

Therefore, check explicitly for the trip header case (which
should show the menu item) and for good measure prevent
the nullpointer access (that should be caught by testing
for trip, but who [... cut]
commit 67769235 (2021-08-12): desktop: fix crash when right-clicking of trip headers

Commit e42fc1a1e9a13c77d3474dbcb26b68b8772b8c6d introduced a
crash condition. Apparently the code attempts to test whether
the clicked-on item is a top-level dive. The "Collapse others"
menu item should not be shown in that case. It does this by
testing "d->divetrip". However, "d" might quite logically be
null if clicking on an unexpanded trip header.

Therefore, check explicitly for the trip header case (which
should show the menu item) and for good measure prevent
the nullpointer access (that should be caught by testing
for trip, but wh [... cut]
closed 2021-08-12 by bstoeger