subsurface/subsurface #3304: Can't upload OSTC parameters

dadefay, 2021-08-13:
### Describe the issue:

- [x ] Bug
- [x ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The "language" parameter is no more managed by hwOS (3.1.x and later), each language has its own hex file. When trying to upload parameters to the OSTC, language is not recognized, the computer exits Bluetooth mode and Subsurface locks itself : so it is impossible to change any OSTC parameter using Subsurface. By the way, computer model should be memorized by Subsurface, not only the manufacturer (I have to reselect "OSTC 2/3/Sport/cR" each time).
[subsurface.log](https://github.com/subsurface/subsurface/files/6981809/subsurface.log)

### Operating system:

Windows 10.

OSTC 2 hwOS 3.1.5.

### Subsurface version:

5.0.2.

Yes.

Official.

### Steps to reproduce:

1) Put the OSTC in Bluetooth Mode.
2) Select "Change settings on dive computer" in Subsurface.
3) Select "OSTC 2/3/Sport/cR" in the left panel.
4) Connect the OSTC (COM11 for my computer).
5) Once connected click on "Retrieve available details" : should be successful.
6) Don't change any parameter.
7) Click on "Save changes to device" : Subsurface displays "Error: Failed!" and the OSTC displays "Data Rx Timeout".

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dadefay, 2021-08-24:
Same behaviour with Subsurface 5.0.3.

glance-, 2022-01-06:
I just tested this with a OSTC 3 running 3.16, Subsurface 5.0.5-dev-latest, and It works as expected.

dadefay, 2022-03-26:
Same test with Subsurface 5.0.7 and OSTC 2 running hwOS 3.17, the same fault as described on August 13, 2021 occurs. Idem with  OSTC Plus.

I wonder how **glance** made it working...

glance-, 2022-03-26:
Can you connect to the dive computer, read the settings, save a backup of your settings to a xml file and then try to write them back to the computer, and post the xml backup and the logfile?

I think the actual issue doesn't have anything to do with your initial guess about language, but rather about that you're running the computer in CCR-mode: Auto SP which didn't exist when this code was written, thus things becomes pear shaped.

dadefay, 2022-03-27:
Thanks Anton. I did exactly what you asked, backup and log files attached (OSTC configured for CCR with auto setpoint).

Let me know if you need files with different configuration.

Kind regards.

[Backup_CCR_SP_Auto.zip](https://github.com/subsurface/subsurface/files/8357166/Backup_CCR_SP_Auto.zip)
[subsurface.log](https://github.com/subsurface/subsurface/files/8357164/subsurface.log).

glance-, 2022-03-27:
```
[8.684015] INFO: Write: size=1, data=72
[8.697320] INFO: Read: size=1, data=72
[8.697586] INFO: Write: size=1, data=1F
[8.729867] INFO: Read: size=1, data=02
[8.729903] INFO: Read: size=1, data=4D
```

```xml
<CcrMode>-1</CcrMode>
```
There's the bug. When we ask for CCR mode: 0x1F we get 02 as answer, and when this code was written there were no 02 value, so when we try to store 02 in a thing that can only be 00 or 01, we get a invald value -1. When we then try to write -1 back to the DC, it gets converted to 0xFF which causes the DC to go "WTF mate?" and just says 0xff back, which means ktnxbye.

One of the issues here was that you assumed this had something to do with the language parameter. I verified that this was not the case and closed the issue. If you would have phrased the issue differently I probably would have looked at the log file and concluded that this was not the case. You made a assumption, so I made a assumption that caused me to miss the real issue. Not any big issue but rather a reminder to all of us on how we ask questions and how we answer them.

I'll write up a patch.

dadefay, 2022-03-27:
Thanks Anton.

glance-, 2022-03-28:
Should be fixed as of merging https://github.com/subsurface/subsurface/pull/3428. You can build it yourself or test the CI builds.

RalphLembcke, 2022-03-29:
Hi,
i have analysed the subsurface log posted above. Here is the interesting section:

[61.013265] INFO: Write: size=1, data=77		CMD comm_write_option
[62.349378] INFO: Read: size=1, data=77    		ACKN
[62.349761] INFO: Write: size=5, data=10|0C|2B|01|48	Gas1|O2=12%,He=43%,type=first,changedepth=48m
[62.350670] INFO: Read: size=1, data=4D                 awaiting for next command

[62.351084] INFO: Write: size=1, data=77                CMD comm_write_option
[62.393114] INFO: Read: size=1, data=4D	         awaiting next command

In the 2nd block the 4D response after 77 indicates that the OSTC did not understood the command, else it would have answered 77 as can be seen in the 1st block. Actually the 4D at the end of the 1st block is not a "positive" response to the gas data, but instead the OSTC is already out of sync here. The reason is the following, taken from the OSTC comm protocol description (send / receive directions are from OSTC point of view):

[comm-docu.txt](https://github.com/subsurface/subsurface/files/8373752/comm-docu.txt)

	comm_write_option:					## write an option value ##

	send 0x77
	try to receive 1 byte option index, timeout 400 ms
	timeout              ->  goto comm_command_loop

	| Attention: do not send the option values too fast after the option index,
	|            as the OSTC does not know how many bytes will follow after the
	|            index before it has actually evaluated the index!
	|            This is a flaw in the protocol design...

	index = 0x10 - 0x19  ->  try to receive 4 byte in sequence O2%, He%, type, change depth
	                         timeout  ->  goto comm_command_loop
	                         else     ->  update option
	                                      goto comm_command_loop

Subsurface log says it sent the option value (0x10) and the 4 parameters (0C|2B|01|48) in one batch. That may have worked in the past with slower computers, slower BT, or whatever, but now that seems to be too fast so that the OSTC goes out of sync because the option values came too early after the option index. Try to split the 5 byte write operation into a 1 byte option index write and a 4 byte option value write with a little pause (1 ms) between. This should be done for all option indexes.

glance-, 2022-03-29:
Nice catch. I didn't even look in the second subsurface.log because I was pretty sure on my diagnosis after seeing the above cited values in the xml and previous log.

Both(!) comm-docu.txt in the [hwos_code](https://code.heinrichsweikamp.com/public/hwos_code) repo are newer than this code and such protocol details isn't handled in subsurface but rather in [libdivecomputer](https://github.com/subsurface/libdc/blob/Subsurface-NG/src/hw_ostc3.c#L975)

We should probably break this out to a separate issue, and probably in libdivecomputer.

I'll loop in @jefdriesen and see if he likes to pick up this one.

jefdriesen, 2022-03-30:
Strange that we never noticed this before. I also can't reproduce this with any of my ostc's. I assume this means the ostc uart doesn't have a (small) hardware buffer? I'm not even sure adding a small delay will fix the problem, because the bytes could still end up in the same USB or bluetooth packet. That's something that happens in the driver and/or the hardware, and thus outside our control.

RalphLembcke, 2022-03-30:
Yeah, that's why i wrote "with older / slower computers, ...". Anyhow, 
sending option index and option values in one batch is not good. There 
definitely a weakness in the OSTC protocol here, but we can't fix this 
without loosing compatibility with everything out there already.... ;-(

To manage the queuing on the TX side, there is a need to key the TX 
system with buffer commands alike 'flush out buffer now', then do a 
controlled pause, and then send the next batch of data.

The OSTC RX hardware only holds one byte, so the software needs to be 
really fast here. Indeed, the code is highly optimized here, but it 
seems that in some peer combinations even that doesn't help any more. So 
please be gentle with the tiny PIC µController when shooting it with 
serial data.

jefdriesen, 2022-03-31:
The attached [0001-Fix-changing-the-OSTC-settings.patch](https://github.com/subsurface/subsurface/files/8388972/0001-Fix-changing-the-OSTC-settings.patch.txt) should fix the problem. Since I can't reproduce the problem, someone else should try and confirm it actually works.

dirkhh, 2022-03-31:
@dadefay can you build from sources, or do you need me to create a binary for you in order to test Jef's patch?

dirkhh, 2022-03-31:
No problem. Scrolling back, you are on Windows 10. That makes building rather tricky, anyway.
I'll try to get you a test build today.

dirkhh, 2022-03-31:
Please try this one https://subsurface-divelog.org/downloads/test/subsurface-OSTC-test-5.0.8-2-gc4276d2d52dd.exe

dadefay, 2022-03-31:
I tried this with my OSTC 2 running hwOS 3.18 in CCR mode. The behaviour is much better, the communication doesn't fail systematically now. I can retrieve parameters, change some of them and send them back to the computer, it works... a number of times, after that I get Error: failed! on Subsurface and RX data timeout on the computer.

Hope this will help...

jefdriesen, 2022-04-01:
Can you try again, this time with the "Save libdivecomputer logfile" checkbox enabled. And then afterwards, attach the resulting log file to the ticket. Otherwise we have no idea why it failed.

dadefay, 2022-04-01:
Done, see attached file. It failed after I change a diluent then click on "Save changes to device", "Rx Data Timeout" on the computer.
[subsurface.log](https://github.com/subsurface/subsurface/files/8396489/subsurface.log)

RalphLembcke, 2022-04-01:
Hmmm, that's pretty much the same as in the first log posted above. I calculated the delta times, here's a section that looks good:

[14.264391] INFO: Write: size=1, data=77        -> comm_write_option
     33390
[14.297781] INFO: Read: size=1, data=77         -> ACKN
       231
[14.298012] INFO: Write: size=1, data=42        -> index
         5
[14.298017] INFO: Sleep: value=10               -> sleep
     10602
[14.308619] INFO: Write: size=1, data=00        -> value
     13029
[14.321648] INFO: Read: size=1, data=4D         -> ACKN, back in comm_write_option

And here's the section right before the communication goes out of sync:

[104.049702] INFO: Write: size=1, data=77       -> comm_write_option
** 1.382860 ***** ----> that is very very long......
[105.432562] INFO: Read: size=1, data=77        -> ACKN
        286
[105.432848] INFO: Write: size=1, data=10       -> index 0x10 (gas data)
          7
[105.432855] INFO: Sleep: value=10              -> sleep
      10946
[105.443801] INFO: Write: size=4, data=0C2B0148  ->value - 4 byte gas data
**       26 ***** -----> that's way too short...
[105.443827] INFO: Read: size=1, data=4D        -> ACKN, back in comm_write_option
        436
[105.444263] INFO: Write: size=1, data=77       -> comm_write_option
       9475
[105.453738] INFO: Read: size=1, data=4D        -> unexpected: ACKN, back in comm_write_option instead of 0x77 +++
[105.453765] ERROR: Unsupported command. [in src/hw_ostc3.c:312 (hw_ostc3_transfer)]
[110.653558] INFO: Write: size=1, data=FF       -> command to quit
[110.653584] INFO: Read: size=1, data=4D        -> unexpected: ACKN, back in comm_write_option instead of 0xFF
[110.653588] ERROR: Unsupported command. [in src/hw_ostc3.c:312 (hw_ostc3_transfer)]
[110.653594] ERROR: Failed to send the command. [in src/hw_ostc3.c:597 (hw_ostc3_device_close)]

So the 'insert a sleep' fix obviously does not cure the problem, still i recommend to keep it in the code to counteract the protocol weekness. The only idea i have in the moment is that there is a lot of buffering and batch-transmission going on, not only on the host side with its BT-driver, but also in the BT radio link itself, as well as inside the OSTC on the link between the BT chip and the µ-controller's UART. It would be interesing to know if that last gas data transmission delivered correct gas settings or not. Anyhow, we can't really control the latencies introduced by the bufferings if not adding long sleeps in all places, but that also doesn't seem to be needed as in most cases it works.

So maybe in case a 0x77 command gets answered by a 0x4D instead of the expected 0x77 we just give it a second try and only abort if 0x77 -> 0x4D happens a second (or 3rd?) time in sequence?!?!

dadefay, 2022-04-01:
I wonder if this could help analysis : the firmware update works, it didn't before (Ihad to use another software).

jefdriesen, 2022-04-01:
The first part of the log looks all good. After the first 14 seconds, there is a 90 seconds window with no communication at all. Next is a successful write followed by a failed one. Could it be that this 90 second window with no activity triggered a timeout somewhere, and that causes further communication to fail?

> So the 'insert a sleep' fix obviously does not cure the problem, still i recommend to keep it in the code to counteract the protocol weekness. The only idea i have in the moment is that there is a [...]

Well, it did work for all the writes in the first part of the log. As I mentioned in my previous comment, we have very little (or no) control over when the data is really send. It's certainly possible the send/write call simply adds the bytes to the output queue and returns immediately, while the actual transmission is done async in the background. That's why I said adding a small delay might not help.

> So maybe in case a 0x77 command gets answered by a 0x4D instead of the expected 0x77 we just give it a second try and only abort if 0x77 -> 0x4D happens a second (or 3rd?) time in sequence?!?!

That's not going to work, because we already use this sequence to detect unsupported commands.

jefdriesen, 2022-04-01:
> I wonder if this could help analysis : the firmware update works, with obvoiously a lot of data transmitted to the computer. It didn't before (I had to use another software).

No, that's completely unrelated. The fix you tried contained no change in the firmware update code. So if the firmware update failed before and works now, that could indicate the problem is located elsewhere.

RalphLembcke, 2022-04-02:
> The first part of the log looks all good. After the first 14 seconds, there is a 90 seconds window with no communication at all. Next is a successful write followed by a failed one. Could it be that [...]

The OSTC main command loop has a timeout period of 2 minutes, so either the subsurface code is slow here in switching between dive log download and parameter upload, or the program is holding the connection open while the user is slow in selecting different transfer types.

>>     So maybe in case a 0x77 command gets answered by a 0x4D instead of the expected 0x77 we just give it a second try and only abort if 0x77 -> 0x4D happens a second (or 3rd?) time in sequence?!?!
>
> That's not going to work, because we already use this sequence to detect unsupported commands.

Can't the code handle an unsupported command by just giving it a retry and then abort on the 2nd unexpected 0x4D ? This may help to get into sync again, and it is the last idea i have.

glance-, 2022-04-02:
In the configure dive computer dialog in subsurface, the connection is first opened, then settings and be read or written. Common pattern is that you read the settings, change something and then write them back to the device.

I've never seen it reported causing a issue, so I think it's fine for now.

jefdriesen, 2022-04-04:
@dadefay Can you repeat the experiment a few times, and attach the log for each attempt? I want to double check whether that error after 90 seconds idle was a coincidence or something systematic. So simply read/write the settings as before, wait some random time before making another change, and repeat that a few times.

dadefay, 2022-04-04:
Ok Jef, I do that tonight.

dadefay, 2022-04-04:
So, here's the results :
- Subsurface1 : connect, retrieve details (ok), no change, save changes to device (ok), select Advanced Settings tab, no change, save changes to device (ok), select Gas Settings tab, no change, save changes to device (ok), change one gas setting, save changes to device (failed, Data Rx Timeout).
- Subsurface2 : connect, retrieve details (ok), change salinity, save changes to device (ok), repeat this sequence several times (ok).
- Subsurface3 : connect, retrieve details (ok), select Advanced Settings tab, change GF High, quickly save changes to device (failed, Data Rx Timeout).
- Subsurface4 : same sequence as above several times (all ok), wait for a few tens of seconds (failed, Data Rx Timeout).
- Subsurface5 : connect, retrieve details (ok), select Advanced Settings tab, change GF High, wait for some seconds (much less than 90), save changes to device (failed, Data Rx Timeout).

I hope this will help you, it seems to me anything but simple !

Didier

[subsurface1.log](https://github.com/subsurface/subsurface/files/8412773/subsurface1.log)
[subsurface2.log](https://github.com/subsurface/subsurface/files/8412774/subsurface2.log)
[subsurface3.log](https://github.com/subsurface/subsurface/files/8412775/subsurface3.log)
[subsurface4.log](https://github.com/subsurface/subsurface/files/8412776/subsurface4.log)
[subsurface5.log](https://github.com/subsurface/subsurface/files/8412777/subsurface5.log)

jefdriesen, 2022-04-11:
- subsurface1: Log file is truncated. The part that's present is all good, including the writes.
- subsurface2: No info in the log beyond opening the serial port.
- subsurface3: One good write, followed by a failed one.
- subsurface4: Many successful writes, and then suddenly a failed one. The last successful write is exactly the same setting as in the previous log (100C2B0148, setting gas mix 1 to Tx12/43)
- subsurface5: Identical sequence as subsurface3.log

So in the last 3 logs, we always see the same pattern:
```
INFO: Write: size=1, data=77
INFO: Read: size=1, data=77
INFO: Write: size=1, data=10
INFO: Sleep: value=10
INFO: Write: size=4, data=0C2B0148
INFO: Read: size=1, data=4D
INFO: Write: size=1, data=77
INFO: Read: size=1, data=4D
ERROR: Unsupported command. [in src/hw_ostc3.c:312 (hw_ostc3_transfer)]
```
And it's always seems to be after the first multi byte setting where it goes wrong. In subsurface4 there were multiple single byte settings that were changed successfully. I wonder if that first multibyte setting is applied correctly or not. As Ralph already suspected, it probably already goes wrong during that last successful write.

To confirm that, can you try again, changing gas mix 1, every time to a slightly different gas mix (e.g. increment the O2 or He percentage by 1 or something like that), and then once it fails, check the setting on the ostc to see whether it applied the new value or still at the previous value. And also attach the logs.

dadefay, 2022-04-16:
Hi Jef,

Sorry for the delay, high pressure these days... 

First trial :connect, retrieve details,

dadefay, 2022-04-16:
Hi Jef,

Sorry for the delay, high pressure these days...

My trial :
- Connect.
- Retrieve details.
- Move to Gas Settings tab and slightly change a gas mix.
- Save changes to device : failed ! Data Rx Timeout on the OSTC.
- Disconnect / reconnect.
- Retrieve details : gas did not change.

Log file attached.

[subsurface.log](https://github.com/subsurface/subsurface/files/8500416/subsurface.log)

Regards.

Didier

jefdriesen, 2022-04-20:
The logs only shows the reading part. Anyway, based on the observation that the setting didn't change, the communications already goes wrong earlier, and we only notice something is wrong during the next command. I'll apply the patch because it appears to result in less errors. But I don't see where or why it still goes wrong now. @RalphLembcke Do you have any ideas? I don't see anything suspicious in the hwos firmware code.

DennisOTSC, 2022-11-09:
Moin,
is there a solution for the problem. it still comes the message when change. Save changes to device : failed ! Data Rx Timeout on the OSTC.