subsurface/subsurface #3305: Crash with ASAN - DiveEventItem isn't destroyed, slot gets called after change of dive list

dirkhh, 2021-08-14:
### Describe the issue:

- [x] Bug

### Issue long description:

So this is a place holder as I haven't fully debugged this, but I wanted to make sure I capture what I was able to figure out.
Scenario: I open Subsurface with a dive list that contains one dive with one event.
I then open cloud storage.
The initial dive list gets closed - but the `DiveEventItem` is never destroyed.
After we load the new dive list we select the last dive and call `select_single_dive()` which calls down into this glorious stack trace: 
```
    #0 0x55fb79cf4b84 in DiveEventItem::recalculatePos(int) /home/hohndel/src/subsurface/profile-widget/diveeventitem.cpp:258:116
    #1 0x55fb79cf7128 in DiveEventItem::DiveEventItem(dive const*, event*, gasmix, DivePlotDataModel*, DiveCartesianAxis*, DiveCartesianAxis*, int, double, QGraphicsItem*)::$_0::operator()() const /home/hohndel/src/subsurface/profile-widget/diveeventitem.cpp:35:19
    #2 0x55fb79cf7003 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, DiveEventItem::DiveEventItem(dive const*, event*, gasmix, DivePlotDataModel*, DiveCartesianAxis*, DiveCartesianAxis*, int, double, QGraphicsItem*)::$_0>::call(DiveEventItem::DiveEventItem(dive const*, event*, gasmix, DivePlotDataModel*, DiveCartesianAxis*, DiveCartesianAxis*, int, double, QGraphicsItem*)::$_0&, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:146:13
    #3 0x55fb79cf6f20 in void QtPrivate::Functor<DiveEventItem::DiveEventItem(dive const*, event*, gasmix, DivePlotDataModel*, DiveCartesianAxis*, DiveCartesianAxis*, int, double, QGraphicsItem*)::$_0, 0>::call<QtPrivate::List<>, void>(DiveEventItem::DiveEventItem(dive const*, event*, gasmix, DivePlotDataModel*, DiveCartesianAxis*, DiveCartesianAxis*, int, double, QGraphicsItem*)::$_0&, void*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:256:13
    #4 0x55fb79cf6ec0 in QtPrivate::QFunctorSlotObject<DiveEventItem::DiveEventItem(dive const*, event*, gasmix, DivePlotDataModel*, DiveCartesianAxis*, DiveCartesianAxis*, int, double, QGraphicsItem*)::$_0, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) /usr/include/qt/QtCore/qobjectdefs_impl.h:443:17
    #5 0x7f3f5ae8175a  (/usr/lib/libQt5Core.so.5+0x2f075a)
    #6 0x55fb79cd1771 in DiveCartesianAxis::sizeChanged() /home/hohndel/src/subsurface-clang-build/profile-widget/subsurface_profile_autogen/EWIEGA46WW/moc_divecartesianaxis.cpp:187:5
    #7 0x55fb79ce3863 in DiveCartesianAxis::animateChangeLine(QLineF const&) /home/hohndel/src/subsurface/profile-widget/divecartesianaxis.cpp:263:2
    #8 0x55fb79d4f34c in ProfileWidget2::updateAxes() /home/hohndel/src/subsurface/profile-widget/profilewidget2.cpp:813:17
    #9 0x55fb79d4b034 in ProfileWidget2::setProfileState() /home/hohndel/src/subsurface/profile-widget/profilewidget2.cpp:1113:2
    #10 0x55fb79d446ba in ProfileWidget2::plotDive(dive const*, int, bool, bool) /home/hohndel/src/subsurface/profile-widget/profilewidget2.cpp:547:3
    #11 0x55fb7988ff95 in MainWindow::plotCurrentDive() /home/hohndel/src/subsurface/desktop-widgets/mainwindow.cpp:431:12
    #12 0x55fb7988d594 in MainWindow::selectionChanged() /home/hohndel/src/subsurface/desktop-widgets/mainwindow.cpp:441:2
    #13 0x55fb798c9c76 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (MainWindow::*)()>::call(void (MainWindow::*)(), MainWindow*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:152:13
    #14 0x55fb798c99be in void QtPrivate::FunctionPointer<void (MainWindow::*)()>::call<QtPrivate::List<>, void>(void (MainWindow::*)(), MainWindow*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:185:13
    #15 0x55fb798c9566 in QtPrivate::QSlotObject<void (MainWindow::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) /usr/include/qt/QtCore/qobjectdefs_impl.h:418:17
    #16 0x7f3f5ae8175a  (/usr/lib/libQt5Core.so.5+0x2f075a)
    #17 0x55fb799bf481 in DiveListView::divesSelected() /home/hohndel/src/subsurface-clang-build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_divelistview.cpp:235:5
    #18 0x55fb79aae504 in DiveListView::selectionChangeDone() /home/hohndel/src/subsurface/desktop-widgets/divelistview.cpp:528:7
    #19 0x55fb79aaa247 in DiveListView::diveSelectionChanged(QVector<QModelIndex> const&) /home/hohndel/src/subsurface/desktop-widgets/divelistview.cpp:262:2
    #20 0x55fb79accdde in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QVector<QModelIndex> const&>, void, void (DiveListView::*)(QVector<QModelIndex> const&)>::call(void (DiveListView::*)(QVector<QModelIndex> const&), DiveListView*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:152:13
    #21 0x55fb79accafe in void QtPrivate::FunctionPointer<void (DiveListView::*)(QVector<QModelIndex> const&)>::call<QtPrivate::List<QVector<QModelIndex> const&>, void>(void (DiveListView::*)(QVector<QModelIndex> const&), DiveListView*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:185:13
    #22 0x55fb79acc716 in QtPrivate::QSlotObject<void (DiveListView::*)(QVector<QModelIndex> const&), QtPrivate::List<QVector<QModelIndex> const&>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) /usr/include/qt/QtCore/qobjectdefs_impl.h:418:17
    #23 0x7f3f5ae8175a  (/usr/lib/libQt5Core.so.5+0x2f075a)
    #24 0x55fb79df5d43 in MultiFilterSortModel::selectionChanged(QVector<QModelIndex> const&) /home/hohndel/src/subsurface-clang-build/qt-models/subsurface_models_desktop_autogen/EWIEGA46WW/moc_filtermodels.cpp:162:5
    #25 0x55fb79f24a7f in MultiFilterSortModel::selectionChangedSlot(QVector<QModelIndex> const&) /home/hohndel/src/subsurface/qt-models/filtermodels.cpp:46:7
    #26 0x55fb79f26d1e in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QVector<QModelIndex> const&>, void, void (MultiFilterSortModel::*)(QVector<QModelIndex> const&)>::call(void (MultiFilterSortModel::*)(QVector<QModelIndex> const&), MultiFilterSortModel*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:152:13
    #27 0x55fb79f26a3e in void QtPrivate::FunctionPointer<void (MultiFilterSortModel::*)(QVector<QModelIndex> const&)>::call<QtPrivate::List<QVector<QModelIndex> const&>, void>(void (MultiFilterSortModel::*)(QVector<QModelIndex> const&), MultiFilterSortModel*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:185:13
    #28 0x55fb79f26656 in QtPrivate::QSlotObject<void (MultiFilterSortModel::*)(QVector<QModelIndex> const&), QtPrivate::List<QVector<QModelIndex> const&>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) /usr/include/qt/QtCore/qobjectdefs_impl.h:418:17
    #29 0x7f3f5ae8175a  (/usr/lib/libQt5Core.so.5+0x2f075a)
    #30 0x55fb79df3923 in DiveTripModelBase::selectionChanged(QVector<QModelIndex> const&) /home/hohndel/src/subsurface-clang-build/qt-models/subsurface_models_desktop_autogen/EWIEGA46WW/moc_divetripmodel.cpp:158:5
    #31 0x55fb79e65fef in DiveTripModelTree::divesSelected(QVector<dive*> const&) /home/hohndel/src/subsurface/qt-models/divetripmodel.cpp:1419:7
    #32 0x55fb79ea389e in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QVector<dive*> const&>, void, void (DiveTripModelTree::*)(QVector<dive*> const&)>::call(void (DiveTripModelTree::*)(QVector<dive*> const&), DiveTripModelTree*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:152:13
    #33 0x55fb79ea35be in void QtPrivate::FunctionPointer<void (DiveTripModelTree::*)(QVector<dive*> const&)>::call<QtPrivate::List<QVector<dive*> const&>, void>(void (DiveTripModelTree::*)(QVector<dive*> const&), DiveTripModelTree*, void**) /usr/include/qt/QtCore/qobjectdefs_impl.h:185:13
    #34 0x55fb79ea31d6 in QtPrivate::QSlotObject<void (DiveTripModelTree::*)(QVector<dive*> const&), QtPrivate::List<QVector<dive*> const&>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) /usr/include/qt/QtCore/qobjectdefs_impl.h:418:17
    #35 0x7f3f5ae8175a  (/usr/lib/libQt5Core.so.5+0x2f075a)
    #36 0x55fb7a0617e6 in DiveListNotifier::divesSelected(QVector<dive*> const&) /home/hohndel/src/subsurface-clang-build/core/subsurface_corelib_autogen/M7EJYLGZXD/moc_divelistnotifier.cpp:695:5
    #37 0x55fb7a2d3dba in setSelection(std::vector<dive*, std::allocator<dive*> > const&, dive*) /home/hohndel/src/subsurface/core/selection.cpp:206:24
    #38 0x55fb7a2d486e in select_single_dive /home/hohndel/src/subsurface/core/selection.cpp:212:3
```
[... cut ...]
very recent development version (v5.0.2-39-g67769235e78e) compiled with ASAN enabled

### Steps to reproduce:

see above
### Current behavior:

we call `recalculatePos()` on a `DiveEventItem` from a dive that has already been deleted and freed

### Expected behavior:

we no longer try to reference freed objects

### Mentions:

@bstoeger you are my go to guy for life time of random C++ objects.
I know this will lead to another instance of you praising the brilliance of the Qt object model, but maybe you can still help me figure out why we don't destroy those objects...

bstoeger, 2021-08-14:
Replace `hideAll(eventItems)` by
```
        qDeleteAll(eventItems);
        eventItems.clear();
```
in `setEmptyState()`?

bstoeger, 2021-08-14:
If that doesn't help, also add it to `setProfileState()` *before* updating the axes.

Yes, this is a horrible mess!

dirkhh, 2021-08-15:
the first suggestion worked.
Of course, there are several other `hideAll()` calls right there - are those also suspect?
But I'll submit this as a fix for now

commit 5bcba774 (2021-08-15): desktop: avoid crash when changing dive logs

If the last displayed dive had events, those DiveEventItems had slots connected
that would update those icons if things changed. When closing the dive log and
switching to a different one, those slots were still called and would then access
freed memory (the event structure from that old dive that is long gone by then).
This code explicitly deletes those DiveEventItems which also removes those signal
slot connections.

Fixes #3305

Sugested-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
PR #3307 (2021-08-17): desktop: avoid crash when changing dive logs
If the last displayed dive had events, those DiveEventItems had slots connected
that would update those icons if things changed. When closing the dive log and
switching to a different one, those slots were still called and would then access
freed memory (the event structure from that old dive that is long gone by then).
This code explicitly deletes those DiveEventItems which also removes those signal
slot connections.

Fixes #3305

Sugested-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>

### Describe the pull request:

- [x] Bug fix

### Pull req [... cut]
commit f7e22271 (2021-08-17): desktop: avoid crash when changing dive logs

If the last displayed dive had events, those DiveEventItems had slots connected
that would update those icons if things changed. When closing the dive log and
switching to a different one, those slots were still called and would then access
freed memory (the event structure from that old dive that is long gone by then).
This code explicitly deletes those DiveEventItems which also removes those signal
slot connections.

Fixes #3305

Sugested-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
closed 2021-08-17 by dirkhh