subsurface/subsurface #3312: Corruption of Log File with Multiple Transmitters

jschripsema, 2021-08-31:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When saving a log with a dive that is using multiple transmitters, in 5.0.3, the log becomes unreadable and generates a "failed to parse" error.

### Operating system:

macOS 11.5.2 English

Shearwater Perdix AI with Shearwater Transmitters

### Subsurface version:

5.0.3

Does NOT happen with 5.0.2

Official release

### Steps to reproduce:

1) Open log that involves multiple transmitters.
2) Save log.
3) Attempt to open log.
4) Exporting log and then importing does not resolve the issue.

### Current behavior:

Log file fails to parse.

### Expected behavior:

Opening log displays dives and not an error message.

### Additional information:

Example log file.
```xml
<divelog program='subsurface' version='3'>
<settings>
<divecomputerid model='Shearwater Perdix AI' deviceid='c5305f40' serial='300f79b1' firmware='71'/>
</settings>
<divesites>
<site uuid='55043254' name='Xxxxxxx Xxxxx' gps='47.348611 -122.324444'>
  <geo cat='2' origin='2' value='Xxxxxx Xxxxxx'/>
</site>
</divesites>
<dives>
<dive number='236' rating='4' visibility='5' sac='11.729 l/min' otu='6' cns='4%' tags='drysuit, salt, shore' divesiteid='55043254' date='2017-11-11' time='12:03:59' duration='51:30 min'>
  <buddy>Xxxxx Xxxxxxxxxx</buddy>
  <notes>Xxxx xxxx Xxxxx xx Xxxxxxx, x xxxxx xxxx xx xxx xxxxx xxxxx xxx XX, x xxxx xxxx-xxx, xxx x xxxxx xxx xxxxxxxx! Xxxxx xxxx xxxx xxx Xxxxxx XX. Xxxxx xxx xxxxxxxx xxxx xx xx, xxx xxxx'x xxxx xxxx. Xxxxxxxx xxxxxxx'x xxxx xxxx xxxx xx xxx xxx…</notes>
  <suit>XxxXxxx Xx 5000</suit>
  <cylinder size='12.087 l' workpressure='237.318 bar' description='HP100' />
  <weightsystem weight='2.3 kg' description='Dolphin Tech Steel Backplate' />
  <weightsystem weight='12.7 kg' description='Weight Harness' />
  <weightsystem weight='1.4 kg' description='Ankle' />
  <divecomputer model='Shearwater Perdix AI' deviceid='c5305f40' diveid='0e936e9c'>
  <depth max='26.0 m' mean='14.525 m' />
  <temperature water='11.0 C' />
  <surface pressure='1.013 bar' />
  <extradata key='Serial' value='300f79b1' />
  <extradata key='FW Version' value='44' />
  <extradata key='Deco model' value='VPM-B/GFS +3 90%' />
  <extradata key='Battery type' value='1.2V NiMH' />
  <extradata key='Battery at end' value='1.3 V' />
[... 299 lines cut ...]
  <sample time='49:30 min' depth='6.8 m' />
  <sample time='49:40 min' depth='6.8 m' pressure0='45.781 bar' />
  <sample time='49:50 min' depth='6.7 m' pressure1='24.132 bar' />
  <sample time='50:00 min' depth='6.7 m' />
  <sample time='50:10 min' depth='6.7 m' pressure1='23.856 bar' />
  <sample time='50:20 min' depth='5.9 m' pressure0='44.54 bar' />
  <sample time='50:30 min' depth='4.6 m' pressure0='44.402 bar' />
  <sample time='50:40 min' depth='3.2 m' pressure0='44.126 bar' />
  <sample time='50:50 min' depth='3.5 m' />
  <sample time='51:00 min' depth='3.6 m' pressure0='43.437 bar' />
  <sample time='51:10 min' depth='2.8 m' />
  <sample time='51:20 min' depth='1.5 m' pressure0='43.299 bar' />
  <sample time='51:30 min' depth='0.0 m' pressure0='43.023 bar' />
  <sample time='51:40 min' depth='0.0 m' pressure0='40.541 bar' />
  <sample time='51:50 min' depth='0.0 m' pressure0='41.231 bar' />
  <sample time='52:00 min' depth='0.0 m' pressure1='18.754 bar' />
  <sample time='52:10 min' depth='0.0 m' />
  <sample time='52:20 min' depth='0.0 m' />
  </divecomputer>
</dive>
</dives>
</divelog>
```

### Mentions:

mturkia, 2021-08-31:
Subsurface 5.0.3 on my Ubuntu loads this sample log fine. If you start Subsurface from command line, do you get any error messages there? (I guess you should be able to start it from /Users/USER/Downloads/Subsurface.app/Contents/MacOS/Subsurface , but not sure as I am not familiar with MacOS)

dirkhh, 2021-08-31:
The log that's posted here is anonymized (which makes sense). But of course the question is... does the problem happen with the anonymized one? Or it is maybe something in your notes that ends up confusing the 5.0.3 build?

jschripsema, 2021-08-31:
It does fail with the anonymized version. 5.0.3 will LOAD this one just fine. This was created with 5.0.2. Save it, or export it with 5.0.3. It saves without error. Try opening the file you just saved, with either 5.0.2 or 5.0.3. That's when it generates the error.

jschripsema, 2021-08-31:
This is what it saves as, which is where you see the error:

```xml
<divelog program='subsurface' version='3'>
<settings>
</settings>
<divesites>
<site uuid='55043254' name='Xxxxxxx Xxxxx' gps='47.348611 -122.324444'>
  <geo cat='2' origin='2' value='Xxxxxx Xxxxxx'/>
</site>
</divesites>
<dives>
<dive number='236' rating='4' visibility='5' sac='11.729 l/min' otu='6' cns='4%' tags='drysuit, salt, shore' divesiteid='55043254' date='2017-11-11' time='12:03:59' duration='51:30 min'>
  <buddy>Xxxxx Xxxxxxxxxx</buddy>
  <notes>Xxxx xxxx Xxxxx xx Xxxxxxx, x xxxxx xxxx xx xxx xxxxx xxxxx xxx XX, x xxxx xxxx-xxx, xxx x xxxxx xxx xxxxxxxx! Xxxxx xxxx xxxx xxx Xxxxxx XX. Xxxxx xxx xxxxxxxx xxxx xx xx, xxx xxxx'x xxxx xxxx. Xxxxxxxx xxxxxxx'x xxxx xxxx xxxx xx xxx xxx…</notes>
  <suit>XxxXxxx Xx 5000</suit>
  <cylinder size='12.087 l' workpressure='237.318 bar' description='HP100' />
  <weightsystem weight='2.3 kg' description='Dolphin Tech Steel Backplate' />
  <weightsystem weight='12.7 kg' description='Weight Harness' />
  <weightsystem weight='1.4 kg' description='Ankle' />
  <divecomputer model='Shearwater Perdix AI' deviceid='e8bce4a4' diveid='0e936e9c'>
  <depth max='26.0 m' mean='14.525 m' />
  <temperature water='11.0 C' />
  <surface pressure='1.013 bar' />
  <extradata key='Serial' value='300f79b1' />
  <extradata key='FW Version' value='44' />
  <extradata key='Deco model' value='VPM-B/GFS +3 90%' />
  <extradata key='Battery type' value='1.2V NiMH' />
  <extradata key='Battery at end' value='1.3 V' />
  <extradata key='T1 battery' value='normal' />
  <extradata key='T2 battery' value='normal' />
  <event time='0:10 min' type='25' flags='1' name='gaschange' cylinder='0' />
  <sample time='0:10 min' depth='3.8 m' temp='12.0 C' pressure='170.99 bar' ndl='99:00 min' rbt='249:00 min' />
  <sample time='0:20 min' depth='6.2 m' pressure='169.059 bar' />
  <sample time='0:30 min' depth='7.9 m' pressure='168.508 bar' />
  <sample time='0:40 min' depth='8.4 m' pressure='167.818 bar' />
  <sample time='0:50 min' depth='8.5 m' pressure='167.543 bar' />
  <sample time='1:00 min' depth='8.6 m' pressure='167.68 bar' />
  <sample time='1:10 min' depth='8.6 m' pressure='166.853 bar' />
  <sample time='1:20 min' depth='9.2 m' />
  <sample time='1:30 min' depth='9.2 m' />
  <sample time='1:40 min' depth='10.1 m' pressure='165.75 bar' />
  <sample time='1:50 min' depth='10.2 m' pressure='165.06 bar' />
  <sample time='2:00 min' depth='10.3 m' pressure='164.785 bar' />
  <sample time='2:10 min' depth='10.5 m' pressure='164.509 bar' />
  <sample time='2:20 min' depth='10.6 m' pressure='164.095 bar' />
  <sample time='2:30 min' depth='11.0 m' pressure='164.233 bar' />
  <sample time='2:40 min' depth='11.4 m' pressure='162.992 bar' />
  <sample time='2:50 min' depth='11.5 m' pressure='163.13 bar' />
  <sample time='3:00 min' depth='12.3 m' pressure='162.165 bar' />
  <sample time='3:10 min' depth='13.6 m' pressure='161.475 bar' ndl='96:00 min' />
  <sample time='3:20 min' depth='15.2 m' temp='11.0 C' pressure='160.786 bar' ndl='69:00 min' />
  <sample time='3:30 min' depth='15.6 m' pressure='159.82 bar' ndl='57:00 min' />
  <sample time='3:40 min' depth='16.1 m' pressure='159.269 bar' ndl='55:00 min' />
  <sample time='3:50 min' depth='15.9 m' pressure='158.579 bar' ndl='52:00 min' />
  <sample time='4:00 min' depth='15.8 m' pressure='158.166 bar' ndl='54:00 min' />
  <sample time='4:10 min' depth='16.0 m' pressure='157.752 bar' />
  <sample time='4:20 min' depth='16.7 m' pressure='157.2 bar' ndl='49:00 min' />
  <sample time='4:30 min' depth='16.8 m' pressure='157.338 bar' ndl='45:00 min' />
  <sample time='4:40 min' depth='17.0 m' pressure='156.373 bar' ndl='42:00 min' />
  <sample time='4:50 min' depth='17.3 m' pressure='155.684 bar' ndl='41:00 min' />
  <sample time='5:00 min' depth='17.1 m' pressure='155.27 bar' ndl='42:00 min' />
  <sample time='5:10 min' depth='17.2 m' pressure='154.718 bar' ndl='41:00 min' />
  <sample time='5:20 min' depth='17.2 m' pressure='154.305 bar' ndl='43:00 min' />
  <sample time='5:30 min' depth='16.6 m' pressure='153.202 bar' />
  <sample time='5:40 min' depth='16.2 m' pressure='152.926 bar' ndl='47:00 min' />
  <sample time='5:50 min' depth='15.9 m' pressure='152.788 bar' ndl='52:00 min' />
  <sample time='6:00 min' depth='16.4 m' pressure='152.374 bar' ndl='50:00 min' />
  <sample time='6:10 min' depth='16.1 m' ndl='52:00 min' />
  <sample time='6:20 min' depth='16.3 m' pressure='150.857 bar' ndl='54:00 min' />
[... 256 lines cut ...]
  <sample time='49:10 min' depth='6.7 m' />
  <sample time='49:20 min' depth='6.9 m' pressure='45.919 bar' pressure='24.959 bar' sensor='-1' />
  <sample time='49:30 min' depth='6.8 m' />
  <sample time='49:40 min' depth='6.8 m' pressure='45.781 bar' sensor='0' />
  <sample time='49:50 min' depth='6.7 m' pressure='24.132 bar' sensor='-1' />
  <sample time='50:00 min' depth='6.7 m' />
  <sample time='50:10 min' depth='6.7 m' pressure='23.856 bar' />
  <sample time='50:20 min' depth='5.9 m' pressure='44.54 bar' sensor='0' />
  <sample time='50:30 min' depth='4.6 m' pressure='44.402 bar' />
  <sample time='50:40 min' depth='3.2 m' pressure='44.126 bar' />
  <sample time='50:50 min' depth='3.5 m' />
  <sample time='51:00 min' depth='3.6 m' pressure='43.437 bar' />
  <sample time='51:10 min' depth='2.8 m' />
  <sample time='51:20 min' depth='1.5 m' pressure='43.299 bar' />
  <sample time='51:30 min' depth='0.0 m' pressure='43.023 bar' />
  <sample time='51:40 min' depth='0.0 m' pressure='40.541 bar' />
  <sample time='51:50 min' depth='0.0 m' pressure='41.231 bar' />
  <sample time='52:00 min' depth='0.0 m' pressure='18.754 bar' sensor='-1' />
  <sample time='52:10 min' depth='0.0 m' />
  <sample time='52:20 min' depth='0.0 m' />
  </divecomputer>
</dive>
</dives>
</divelog>
```

dirkhh, 2021-08-31:
> This is what it saves as, which is where you see the error:
> 
> ```
>   <sample time='49:20 min' depth='6.9 m' pressure='45.919 bar' pressure='24.959 bar' sensor='-1' />
> ```

What the heck? Why are we writing an entry with two pressure values???

dirkhh, 2021-08-31:
So something makes us decide to write this out in the legacy format instead of the current format for pressure sensors.
Very odd. Still digging into this...

dirkhh, 2021-09-01:
This code wasn't touched in years, but its behavior here is... interesting. I have some more investigating to do, but at least I understand what's going on.

dirkhh, 2021-09-02:
A number of unfortunate things are happening here.
(1) your initial file makes no sense (yes, I know we wrote it, but...) you have two sensors, but only one tank defined. So you have sensor data for a cylinder that doesn't exist
(2) when we read the file back, we realize that that second sensor is bogus and mark that fact by saying this isn't really a sensor (we use the special value '-1' to mark that)
(3) before we write out the file again, we have to decide if for some reason this needs to be written in our current format or a legacy format that we used to use - and in that process we check specifically if there are multiple sensors - but allow for one of the sensors being a dedicated o2sensor (usually used in CCR contexts). Now while we scan the file, we use a special value to indicate that we haven't found an o2sensor yet. You guessed it, that's also '-1'.
So now we think that we should write this out in the old format, because we don't have two different pressure sensor (we think the second one is actually an o2sensor).
(4) when we write this out, the old format doesn't support writing two pressure sensor entries into the same sample - the test above was supposed to catch this case
(5) the resulting file is now invalid and gets rejected on the next load

@bstoeger @atdotde @torvalds - I'm trying to find the best way to fix this... there are a number of easy solutions for the "useful" case... but specifically once we have a file written out that has more sensors defined than cylinders, we run into problems. Because a sensor id of -1 simply isn't read back in correctly (`pressure-1` isn't parsed). So should we add a fake cylinder? Should we allow a sensor value for a cylinder that doesn't exist (as we clearly do when downloading from the Shearwater dive computer)? I think it's that second option that's least invasive and most useful...

Thoughts?

dirkhh, 2021-09-02:
That's great information - I was wondering why you didn't have that second cylinder.
But even with that information... we still should save the bogus data. So there is work needed to fix the problem that you've shown us.

bstoeger, 2021-09-02:
> This code wasn't touched in years, but its behavior here is... interesting. I have some more investigating to do, but at least I understand what's going on.

Yes it has - I "fixed" an out-of-bound access when merging this kind of dives by making the sensor -1, which I thought means no sensor. Apparently not. I don't understand that sensor thing, sorry.

In any case the parsing shouldn't fail.

bstoeger, 2021-09-02:
> before we write out the file again, we have to decide if for some reason this needs to be written in our current format or a legacy format that we used to use

Does that still make sense? Could we simply remove writing of the legacy format?

bstoeger, 2021-09-02:
> Should we allow a sensor value for a cylinder that doesn't exist (as we clearly do when downloading from the Shearwater dive computer)?

Probably, yes. Because what happens if the user erroneously deletes a cylinder and readds it later? Currently, all the samples of that cylinder are lost.

Another option would be to disallow/warn of deletion of cylinders that have samples referring to them: "This cylinder has pressure readings, which will be deleted. Are you sure?".

For now, I would suggest to revert all the sensor-id sanitizing and throw these samples away when merging dives.

bstoeger, 2021-09-02:
Ok, I now believe the correct thing to do is to delete sensor readings if a cylinder is deleted (but warn the user!) but also relabel sensor readings!

I will do that, but give me a day or two. Too many other things bothering me at the moment.

dirkhh, 2021-09-02:
> Ok, I now believe the correct thing to do is to delete sensor readings if a cylinder is deleted (but warn the user!) but also relabel sensor readings!

As I am thinking through the option, I believe that would address the issue here nicely.

mikeller, 2026-10-17:
Hi @jschripsema,

Thanks for the detailed anonymised log and the patience working through this. The root cause (a sensor-index sentinel value colliding with pressure-format detection logic, corrupting the file when writing more pressure sensors than cylinders) was fixed across three PRs: #3316 "core: properly clear pressure data", #3521 "Sensor index fixup", and more recently #4523 "Fix the Handling of Tank Pressure Data for Certain CCR Dives", which removed the last related legacy-format special case.

Closing this as fixed. Please reopen if you still see a log file fail to reparse after saving a dive with multiple pressure transmitters on a current build.

closed 2026-10-17 by mikeller