subsurface/subsurface #3317: Suunto Eon Steel (with buhlmann upgrade) unable to open in subsurface

simenon, 2021-09-04:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I think the issue arose after the buhlmann upgrade of the eon steel. Since then i am unable to connect ot the EON Steel. I have updated to the lates 5.0.3 version of subsurface. I tried also with the Eon Steel Black but that doesn't either connect.

[image]

### Operating system:

Windows 10

Suunto Eon Steel (with buhlmann upgrade)

### Subsurface version:

5.0.3

release

### Steps to reproduce:

Connect Eon Steel to USB and try to import

### Current behavior:

Can't find Eon Steel

### Expected behavior:

Should find Eon Steel and import

### Additional information:

### Mentions:

dirkhh, 2021-09-04:
So this device shows the new `USB ID VID 1493 / PID 0036` which is supported in Subsurface 5.0.3
Please try to download again, but in the download dialog click on "Save libdivecomputer logfile". Send that logfile and also the two Subsurface logfiles that you can find (after you exit Subsurface) in `C:\Users\Home\AppData\Roaming\Subsurface` : `subsurface_out.log` and `subsurface_err.log`

simenon, 2021-09-05:
[subsurface.log](https://github.com/subsurface/subsurface/files/7111256/subsurface.log)
[subsurface_err.log](https://github.com/subsurface/subsurface/files/7111257/subsurface_err.log)
[subsurface_out.log](https://github.com/subsurface/subsurface/files/7111260/subsurface_out.log)

dirkhh, 2021-09-06:
Errors aren't insightful at all. All it tells me is that it can't connect to the HID device. Duh, yeah, that was obvious.
We need more interesting error messages :)

Joking aside... let me dig into this some more and try to figure out what's going on. I don't have an EON Steel Black, but Linus does. Maybe I can borrow his and try it with my Mac and figure out what's up.

simenon, 2021-10-15:
I recently had a windows update failure resorting me to fully reinstall windows, and guess what, the problem is gone. Subsurface sees the EON Steel Black and can download dives through USB. I guess it was a windows/driver issue