subsurface/subsurface #3320: Planned Dives do not render properly when viewed on mobile

torresjoj, 2021-09-07:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I noticed dives created with the planner do not render properly when viewed on mobile. HTML tags are visible and the ascent/descent/etc arrows, etc are not rendered.

### Operating system:
Pixel 4a running android 11
Issue also present when running mobile on desktop locally

### Subsurface version:

Plan created on 5.01
Viewed on mobile 3.3.0(5.0.3)

### Steps to reproduce:

Create plan on desktop
Load plan on mobile and view

### Current behavior:
Renders html as plain text

### Expected behavior:
Renders html properly

### Additional information:
Example:
[image]

### Mentions:

torresjoj, 2021-09-07:
I also noticed that if you edit a planned dive in the mobile app, you lose all newlines. There is some interesting formatting logic around notes along with some juggling between html and plain text in both the mobile code and the generic formatting code.

Before creating a second, related issue (and investing more time understanding and adding yet more special cases to the formatting logic if I fix it), I wanted to ask overall what is the feeling towards rich text for notes? If the idea is to keep notes more simple, and planned dives "need" more sophisticated formatting, should planned dives use a new, different field (e.g. "Dive Plan") without the formatting dance and full rich text support?

Otherwise, I found removing all the formatting logic all together for the notes field seemed to work fine with some superficial testing. Creating and viewing notes across platforms, dive planner, etc all worked fine (though old existing notes would need backwards compatible changes).

dirkhh, 2021-09-07:
I think you are making a good point. Plans and Notes could be handed differently.
Rich text support is really annoying and frankly overkill for regular notes. And 95+% of our users never use the dive planner :)

Usually @atdotde is the one who cares the most about this topic, so let's see what he thinks.

dirkhh, 2021-09-07:
Your PR closed this issue - but admittedly it doesn't really 'fix' it, right?
You change the code that displays the notes, but not the way they are edited. So if a user edits a plan on mobile, the formatting is still lost. But that's where it gets tricky, because while it appears to be safe to always display notes as rich text, always editing them as rich text doesn't work (as that now inserts formatting data). I think it would be better to refuse to edit rich text notes on mobile...

closed 2021-09-07 by dirkhh

torresjoj, 2021-09-07:
I'm not sure the best way to handle/track the issue.

On one hand, the I tried to be specific when creating this issue to just refer to the "viewing" of rich text, so it would seem to "close" this specific issue. 

On the other hand, we still loose some things like bold/coloration on plans due to the formatting and stuff and there is the broader issue of editing rich text as you mentioned.

Personally, I think I'd be comfortable calling this "closed" in that, as you said, 95% of users don't even use plans, and I can't really imagine I'd ever edit one on mobile. I just like to reference it when writing plans/tables in my wet notes for deco and the current (fixed) presentation is totally adequate for that. I can create another issue that raises the broader "editing" of plans and reference this discussion.

To your point about refusing to edit rich text, I wonder if the behavior would be more explicit if we removed the RichText flag in the qml for the editting version of the page. If it still behaves the same for plain text, and just lets you edit the html for plans, that might be a "cleaner" way to handling editing given the current mix. I can check that out in the next day or two.

PR #3321 (2021-09-07): Add rich text support for viewing mobile notes

### Describe the pull request:

- [X] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
I noticed dives created with the planner do not render the notes field properly when viewed on mobile. 
The dive planner saves notes with some html formatting, which won't render properly without rich text support.

### Changes made:
Added support for rich text on the DiveDetailsView notes field

### Related issues:

Fixes #332 [... cut]