subsurface/subsurface #3322: XML file is not loading Start Date or Start Time as well as End Date or End Time - frustrating bug

Love2Dive, 2021-09-07:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

Love2Dive, 2021-09-07:
<?xml version="1.0" encoding="UTF-8"?>

-<dive version="3" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema">

-<diveLog>

<number>429</number>

<gfMin>40</gfMin>

<gfMax>85</gfMax>

<surfaceMins>28940</surfaceMins>

<imperialUnits>true</imperialUnits>

<startBatteryVoltage>1.30000007</startBatteryVoltage>

<startCns>0</startCns>

<startDate>9/5/2021 6:00:10 PM</startDate>

<startO2SensorStatus>false</startO2SensorStatus>

<startLowSetPoint>0</startLowSetPoint>

<startHighSetPoint>1</startHighSetPoint>

<computerFirmware>81</computerFirmware>

<maxDepth>13.5</maxDepth>

<maxTime>2202</maxTime>

<errorHistory/>

<endBatteryVoltage>1.2</endBatteryVoltage>

<endCns>0</endCns>

<endDate>9/5/2021 6:36:47 PM</endDate>

Using XML and importing into Subsurface the Start Date and time are being lost as well as I believe the Stop Date and time.

mikeller, 2026-10-17:
Hi @Love2Dive,

Thanks for reporting the missing start/end date and time on XML import. To look into this we'd need the actual XML file that's failing to load those fields correctly, without seeing the file's exact structure we can't tell what Subsurface's parser is tripping on.

If we don't hear back with a sample file within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.