subsurface/subsurface #3325: Subsurface always transfers all dives (starting from 5.0.3)

BastiSch, 2021-09-17:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Starting with version 5.0.3 subsurface takes ~4 minutes when importing from my Aladin Sport Matrix to show the message that there are no new dives to download or to download one new dive. The log also shows, that all dives are transferred.
Version 5.0.2 did take seconds before showing that there are no new dives (log shows only some bytes transferred) or only some seconds to transfer one new dive.

### Operating system:
Debian testing (bookworm) amd64
The same thing happens on Android

### Subsurface version:
Happening on version 5.0.3 AppImage and 5.0.3-1~bpo12+1 debian repository and 5.0.3 Android
Not happening on 5.0.2 and 4.9.10 AppImages and 5.0.2 Android

### Steps to reproduce:
Start subsurface, Select Import from Divecomputer, Select Download

### Current behavior:
Progress bar is slowly increasing for 4 minutes before showing no new dives.

### Expected behavior:
Taking only seconds to show no new dives.

### Additional information:

[subsurface-5.0.2.log](https://github.com/subsurface/subsurface/files/7186963/subsurface-5.0.2.log)
[subsurface-5.0.3.log](https://github.com/subsurface/subsurface/files/7186971/subsurface-5.0.3.log)

### Mentions:

dirkhh, 2021-09-17:
While I try to parse those logs to figure out what's different, can you do me a favor and look on your dive computer and tell me if there's a serial number (either printed on the device, or available through some info menu in the device UI)?

dirkhh, 2021-09-17:
In the 5.0.3 case we appear to have no fingerprint for this device. We made some changes to how we handle the device id and I wonder if this has an unintended side effect here. Can you send us the result of `ls -l ~/.subsurface/fingerprints` ?
And for each file in that directory, could you run `od -h fullpathtofile` ?

Do you have more than one dive computer that you download from?

BastiSch, 2021-09-17:
The serial of the dive computer is according to Subsurface 5.0.2 is 63068970, which is mentioned in the log as well. I wouldn't know how to get the serial number from the dive computer itself.
In my logbook, I just noticed, dives downloaded with earlier versions show the dive computer as "Scubapro Aladin Sport Matrix (ec4a1642)" and dives downloaded with 5.0.3 show only "Scubapro Aladin Sport Matrix".

ls -l ~/.subsurface/fingerprints:
-rw-r--r-- 1 bastian domain users 8 Feb  8  2019 9a054d81
-rw-r--r-- 1 bastian domain users 8 Sep 17 18:06 ec4a1642
-rw-r--r-- 1 bastian domain users 8 Mär 24  2019 f01a850

od -h for all three files:
9a054d81:
0000000 29e6 47d0 e808 afb5
0000010

ec4a1642:
0000000 6526 5196 45f7 3aaa
0000010

f01a850:
0000000 77bd 4851 7192 52ba
0000010

I currently have one dive computer. I had one 2 years ago prior to this, also Aladin Sport Matrix. And another Aladin 2G as a backup, which I probably did not download from for over a year.

BastiSch, 2021-09-20:
I did some debugging of the issue myself. And I believe the following is happening.
I pinpointed the long downloading to happen, because my last dive is missing a deviceid in the git cloudstorage Divecomputer file. So the dive in the fingerprint is not found. I checked, why this is the case and noticed, that all of my dives starting from August 14th this year are missing a deviceid in the respective Divecomputer file (July 31st is still fine).

My usual workflow is downloading the dive with my mobile and adding some details at my PC. This used to work fine before August.
I did my workflow in single steps, checking, when the deviceid gets lost:
Download on Android version 5.0.3, deviceid correct
Changing data of dive on Android version 5.0.3 (pressure, weight), deviceid correct
Changing data of dive on PC (changing just divesite), deviceid is gone
Version on the PC was debug build of current source, as I used it to get to this point. I will try to reproduce this with 5.0.3 tomorrow.
If I have enough time, I will also try to reproduce using only the PC.

Before trying the different versions as described in the initial post, I used just 4.9.10 AppImage on the PC. But the date between July 31st and August 14th is corresponding to updating the Android App to 5.0.3, if I remember correctly.

dirkhh, 2021-09-20:
We understand why this is happening and have a solution for this. 
My expectation is that the next version of Subsurface and Subsurface-mobile will address the issue.

dirkhh, 2021-09-20:
see #3326

BastiSch, 2021-09-21:
I can confirm that it is working on my PC after #3326. Thank you!
Just to answer my own questions: Loading a dive from git will no longer load the deviceid (in 5.0.3) and thus any change causing it to be saved will have the deviceid removed from the file. After #3326 it is not even stored anymore.

So I understand that the deviceid is removed and why.
I have seen commits to use the serial number from devdata for other divecomputers recently (Mares Icon and Suunto Vyper). Would this also be an idea for the Aladin divecomputers?
All my Aladin dive computers (2x Scubapro Aladin Sport Matrix and Scubapro Aladin 2G) reliably report an individual (not changing) serial number in devdata but it is not part of the dive header, so it is not stored with the dive or shown anywhere.
For the Alading 2G, I can confirm that the same serial number from the devdata is also shown on the device UI. For the Aladin Sport Matrix, there is no longer a possibility to retrieve the serial number using the device itself (but the number is in a simlar range).