subsurface/subsurface #3330: Subsurface not responding

SousaMartins, 2021-10-04:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface not responding in one account, but working fine on another account on the same computer. The day before was working in all accounts

### Operating system:
Windows 10 Pro 21H1
Desktop computer

### Subsurface version:
5.0.3
No
Official release

### Steps to reproduce:

### Current behavior:
Subsurface not responding

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2021-10-04:
"not responding" is such a strange description of a but.
And of course it's Windows, so who knows...

Can you tell us a little more? Is it not opening a window? Or if it is, what does it show?
If you start it manually from a cmd.exe, are there any messages that would help us understand where it is hanging?

SousaMartins, 2021-10-04:
Yes it opens a window and it is completely white

[image]

I started it from the desktop shortcut

I uninstall and install the application twice on both accounts and deleted all files from "AppData\Local\Subsurface" and "AppData\Roaming\Subsurface

[image]

When run it from the cmd window I get the following message:
QObject::connect(QQuickWindow, QDeclarativeGeoMap_QML_6): invalid nullptr parameter

dirkhh, 2021-10-04:
hmm, interesting. Not something that I have seen recently.
If you open a `cmd.exe` and cd to `C:\Program Files\Subsurface` and run subsurface.exe from the cmd.exe - is there anything interesting in the output?

SousaMartins, 2021-10-04:
[image]

Running the subsurface.exe from a cmd window opens the same white window saying (Not Responding).

dirkhh, 2021-10-04:
if you now hit Ctrl-C in that cmd.exe and then grab the two log files in `C:\Users\Home\AppData\Roaming\Subsurface` : `subsurface_out.log` and `subsurface_err.log` that hopefully will tell us what's happening

SousaMartins, 2021-10-04:
The two files are empty.

dirkhh, 2021-10-04:
UGH.
That sucks. So we don't even get to write the initial messages out.

No idea what's going on... and almost impossible to debug from here :(

SousaMartins, 2021-10-05:
How can I debug the app?
I do not have the code.

dirkhh, 2021-10-05:
debugging on Windows is beyond painful as we have no native tool chain that supports Windows and running things under the debugger is near impossible.
Frustrating as this answer may be - I don't have a better one.

Something sends Subsurface into an infinite loop or some other issue that causes it to freeze... a silly (but often useful) thing to try would be to use one of the latest test builds and see if the problem persists - because if it is caused by an access to uninitialized memory, then using a different build is often a useful temporary workaround.
A recent test build would be here: https://subsurface-divelog.org/downloads/test/subsurface-5.0.3-42-g41258647d232.exe

SousaMartins, 2021-10-05:
I solved the problem restoring the system to a previous point before the problem raised.

I believed it was a windows "thing".

Thanks for the support

closed 2021-10-05 by SousaMartins