subsurface/subsurface #3346: Bluetooth detection Perdix and macos Monterey issue

Calumnist, 2021-11-06:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I'm not able to detect my Shearwater Perdix AI dive computer since my mac os Monterey update (Mac book Pro 2020)

### Operating system:
Mac Os Monterey
Shearwater Perdix AI

### Subsurface version:
Subsurface 4.8 
I don't know if it's the same on a previous version of SubSurface
Official release

### Steps to reproduce:
nominal scenario to detect dive computers by BT LE

### Current behavior:
The list of device is empty and it indicated that an error ocured when trying to detect device

### Expected behavior:
Detect my dive computer to download dives

thx ;-)

dirkhh, 2021-11-07:
I don't have a Mac running Monterey - my guess is that Apple once again is breaking things for app developers, but frankly, I have no idea.
A quick Google search seems to indicate that a log of people who've done upgrades have lost access to BLE scanning. There are a bunch of rather, err, brutal suggestions how to fix that - I am not going to reproduce them here, use at your own risk... but it seems pretty clear from what I can find that this is a Monterey issue, not a Subsurface issue.

People keep asking me why I'm still on macOS 10.15 with all of the machines that I actually use. Well... it works and I haven't seen a single new feature in 11.x that would make me want to upgrade... but so many people had so many things go wrong during the upgrades, apps stop working, hardware stops being supported... yeah, no thank you.

Calumnist, 2021-11-07:
Hi Dirk,
Yes, I saw there is some issues with Monterey and BT LE devices. So I've just try to connect to my Perdix AI with another tools like BlueSee (I can see the Perdix into the BT connectable devices) or Shearwater App and I succedded to get my dives... That's the reason I thought it could be a lack/issue/missing technical feature of SubSurface app...
regard

jschripsema, 2021-11-12:
Computer: Perdix AI (but I think the failure is likely all bluetooth computers and not dependent on device).

Just wanted to add a +1 as well as a "not all of us upgraded or have an option of downgrading". I've been using an ancient MacBook Pro 2013 for far longer than I probably should. Upgraded to the new MacBook Pro, with a preinstalled/minimum OS of 12.x. I tried using the iOS App to download my dives as a workaround and that was… not a very good user experience. It almost got it after a few tries, but ultimately failed to save. Tried exporting from Shearwater Cloud, which technically worked, but stripped off a ton of data (not least of all the date and cylinder readings). I finally just grabbed my old machine so I could import my dive.

I have zero problems with Bluetooth on this machine other than Subsurface. Connecting to Apple Keyboard/Trackpad/Mouse all works. Connecting to 3rd party speaker systems works. Connecting to my dive computer via Shearwater Cloud works, I was even able to update the firmware and boot image. Pairing to the dive computer using BlueSee works as expected.

Of note, I saw that Zwift community also had issues with Bluetooth when going to Monterey, but that they had resolved them in the last few days. Also, in the Security and Privacy "Allow the apps below to use Bluetooth" I was never prompted for Subsurface. I adding it manually, but that didn't actually resolve anything. I think something is fundamentally failing to engage with Bluetooth, or the Bluetooth app based security mechanism.

dirkhh, 2021-11-12:
I hear all of you.
And I am happy to accept patches.
I don't have a macOS 12.x computer. I have no interest in adding one at this stage (or upgrading mine).
So unless someone else figures out how to fix this, things will stay exactly as they are.

olstutz, 2021-11-25:
Just another +1 from my side. I am running the Monterey also, 12.0.1 to be specific. Perdix AI and Subsurfacee 5.05.

I too am unable to connect the Perdix anymore. I did update to the latest Subsurface in hope that the issue might have been fixed already, but no joy. 

How can we help get this resolved?

olstutz, 2021-11-25:
One addition to getting around this for now which I only just thought of is to use the mobile app. I already have it and am also on the latest iOS and it managed to download without issues. You can just sync to your cloud and then edit them on your computer. That's if you use the cloud option. If not, then just start to get it all synced up.

mikeller, 2026-10-17:
Hi @Calumnist,

This should now be fixed: macOS requires an app to declare NSBluetoothAlwaysUsageDescription in its Info.plist since Big Sur/Monterey tightened Bluetooth access permissions, or it silently gets no BLE access at all (exactly the empty-scan-list symptom you and several others saw after upgrading to Monterey). I've checked cmake/Modules/MacOSXBundleInfo.plist.in directly in current source and that entitlement is now declared.

Closing this as fixed, though we don't have an explicit confirmation from an affected Monterey user that it resolves the empty BLE scan. Please reopen if your Perdix AI (or other BLE dive computer) still doesn't show up in a scan on a current macOS build.

closed 2026-10-17 by mikeller