subsurface/subsurface #3351: Feature Request: Dive Environment/Problems

dotraphael, 2021-11-22:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

An extra tab with few options for dive environment/problems (i.e. Hot/Cold/etc, equalisation, out of air, suit leak, etc) would be great.
also, would be good to use this in the dive statistics

### Operating system:

Windows 11
iOS 15

Shearwater Perdix AI

### Subsurface version:

Desktop 5.0.5
ios 3.4.1

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2021-11-22:
I would write those kinds of things in the notes. Why do you want a special tab for that?

dotraphael, 2021-11-23:
yes, i currently write them on the notes. but i'm unable to use the statistics with it. like how the dive profile changes when some problems arise

hhmeyer, 2021-11-27:
What about using tags?