subsurface/subsurface #3359: Shifted dive time preview not correctly displayed

jpagett, 2021-12-05:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When right-clicking dive and choosing "Shift dive times", the "Current time:" field always displays "0:0", and the "Shifted time:" field displays a time relative to midnight on 12/31/1969 (shifted by the increment specified by the user).

Example:
If I put in "1:00" into the box to choose how much I'd like to shift my dive time, then select "Earlier", the date/time displayed in the "Shifted time:" field is "Wed Dec 31, 1969 11:00 PM".

### Operating system:

Windows 10, version 21H2. Custom PC, can specify more detail if desired, but doesn't seem relevant here.

### Subsurface version:

Subsurface version 5.04. Have not checked other versions, though I recall this happening before the most recent update as well.

### Steps to reproduce:

1) Right click a dive.
2) Click "Shift Dive Times"
3) Input a time in the input field
4) Observe the error

### Current behavior:

The fields "Current time" and "Shifted time" are not correctly populated

### Expected behavior:

The fields "Current time" and "Shifted time" should display a time relative to the selected dive's actual dive time.

### Additional information:

### Mentions:

dirkhh, 2022-01-05:
Sorry for taking so long to comment on this. I'm still trying to figure out what's going on here.
In my quick tests this seemed to work correctly (but then, I never use Windows).
I'll revive my Windows system and see if maybe I can reproduce it there.

mikeller, 2026-10-17:
Hi @jpagett,

This should be fixed: PR #3573 "desktop: pass dive list to ShiftTimesDialog" replaced the fragile dive lookup that was causing the "Current time"/"Shifted time" fields to show a 1969/1970 date instead of your dive's actual date.

Closing this as fixed, though we don't have an explicit confirmation from you that this resolved it on your end. Please reopen if the Shift Dive Times dialog still shows the wrong date on a current build.

closed 2026-10-17 by mikeller