subsurface/subsurface #3360: Ascent rate

PlongeurSDF, 2021-12-05:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

PlongeurSDF, 2021-12-05:
is it possible to implement an ascent rate curve like DiversDiary 

[image]

atdotde, 2021-12-05:
Do you realise this information is in the color of the depth profile line? It is described in the user manual here: https://subsurface-divelog.org/documentation/subsurface-4-user-manual/#S_DiveProfile

Please also note that even in the plot you show, there are only a few possible values: This comes about as there is a depth sample only every few seconds (the exact number depending on the dive computer) and often enough (I don't know about Mares) also the possible depths come in quite large steps (for example for my Suunto computers, depths are always integer multiples of 30cm/1ft).

From the plot I guess there is one depth sample every 10 seconds. Even if you ascent with the largest velocity you encounter which with 15m/min is quite fast, this is 2.5m / 10sec, so there aren't too many possible depth differences in a 10 sec interval.

I will therefore close this request but feel free to reopen it if you want to add something.

closed 2021-12-05 by atdotde

PlongeurSDF, 2021-12-05:
Yes I know,

Sorry for my bad English.
I’m a diver instructor in France.
A very important exercice for level 2, 3, 4 and Instructor, is « assisted ascent for a diver in distress ».
But we have a target for the ascent rate. And the curve is very good information for the interns.

In the past, I used DiversDiary for Marès. But the software no longer exist.

On the forum French plongeur.com <http://plongeur.com/>, we have talking about Subsurface and a new development.

We thought it would have been a good idea.

Best regards,

Jean-Christophe DEGRAS
+33 6 72 47 50 00
***@***.***

atdotde, 2021-12-05:
Hmm, I don't think the information from the dive computer is useful for this when computed between sample points. If you want more detailed information than the "the line is at most yellow, so we are fine and under 10m/min" you could use the numerical vertical velocity that is shown in information box. But I think it is better to use the ruler (see https://subsurface-divelog.org/documentation/subsurface-4-user-manual/#_the_profile_toolbar ) for a longer time interval like 10m of ascent or so as that has much fewer noise.

PlongeurSDF, 2021-12-05:
For an example, 

Here is two examples of an analysis I did for students 
Thanks,

atdotde, 2021-12-05:
[image]
I think you wanted to attach an image but it did not appear. So here is one from my log. You can see that during the period measured by the ruler, I ascended on average by 3.72 m/min.

PlongeurSDF, 2021-12-05:
Thanks,

You can see my exemple with this link 
https://drive.google.com/drive/folders/1Nu-kfoVzEMXZSwTmR2QDV-drk8RijXKG?usp=sharing <https://drive.google.com/drive/folders/1Nu-kfoVzEMXZSwTmR2QDV-drk8RijXKG?usp=sharing>

Best regards