subsurface/subsurface #3363: Windows 10: Surface not responding

SousaMartins, 2021-12-23:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
On lunch the app it starts and immediately hangs saying Surface (not responding)

### Operating system:
Windows 10 version 21H2 (64bits) US-EN
Personal Computer

### Subsurface version:
5.0.5
Yes 4.9.* version
Official release

### Steps to reproduce:
Running the application

### Current behavior:
Not responding

### Expected behavior:
Application running

### Additional information:
I am unable to debug since I do not have the source files
I already try to uninstall and install again the application
A week ago the app was running find
I already try to run as administrator
I was unable to attach the dump file "surface.dmp" generated by windows
The problem seems to be an incompatibility issue
I already had this problem once and if I remember correctly I solve it restoring a recovery point. This time it did not worked
If you think Windows 11 will solve the problem I will try to install it
With version 4.9.* It starts to connect to the cloud before became irresponsive

I know it will difficult to solve this problem but ... 

### Mentions:

[Output.txt](https://github.com/subsurface/subsurface/files/7770755/Output.txt)

SousaMartins, 2021-12-23:
More information about the hardware and OS
Processor: AMD Ryzen 9 5950X 16-Core Processor 3.40 GHz

Edition	Windows 10 Pro
Version	21H2
Installed on	‎2/‎6/‎2021
OS build	19044.1415
Experience	Windows Feature Experience Pack 120.2212.3920.0

dirkhh, 2021-12-23:
The silly answer is of course "well, it works here". 
I have no idea if upgrading to Win11 will make any difference (but I doubt it).

One thing to try is to run it from the command line and see if that shows us anything useful.
Also, don't use any of your existing setting in case something there is the cause of all this (that's what the `--user` option does)

So open a `cmd.exe`
Navigate to where the app is installed (usually `C:\Program Files\Subsurface`)
`subsurface.exe -v --user=test`

SousaMartins, 2021-12-23:
It worked

I was able to run the app with the command line
and the verbose was:

_**C:\Program Files\Subsurface>
Subsurface v5.0.5,
built with libdivecomputer v0.8.0-devel-Subsurface-NG (b6df353752b659654919eec3e98ced02158456d1)
built with Qt Version 5.15.1, runtime from Qt Version 5.15.1
built with libgit2 1.0.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 4.6."
using custom config for "Subsurface-test"
QObject::connect(QQuickWindow, QDeclarativeGeoMap_QML_5): invalid nullptr parameter
loading dive data from ("C:\\Users\\Martins\\AppData\\Roaming\\Subsurface\\Jose Martins.xml")

File locations:

Cloud email:
Unable to get local git directory
Cloud URL: No valid cloud credentials set.

Image filename table: C:\Users\Martins\AppData\Roaming\Subsurface/hashes
Local picture directory: C:\Users\Martins\AppData\Roaming\Subsurface/picturedata/**_

and when I open the cloud storage I get the following msgs:

**_
returning cloud URL "https://ssrf-cloud-eu.subsurface-divelog.org/git/josecdsmartins@gmail.com[josecdsmartins@gmail.com]"
Opening cloud storage from: "https://ssrf-cloud-eu.subsurface-divelog.org/git/josecdsmartins@gmail.com[josecdsmartins@gmail.com]"
Cloud storage: checking connection to cloud server https://ssrf-cloud-eu.subsurface-divelog.org/git/josecdsmartins@gmail.com
Checking cloud connection...
Cloud storage: successfully checked connection to cloud server
git storage: Synchronising data file
INFO: git storage: update local repo

INFO: git storage: sync with remote https://ssrf-cloud-eu.subsurface-divelog.org/git/josecdsmartins@gmail.com[josecdsmartins@gmail.com]

git storage: Sync with cloud storage
INFO: git storage: delete proxy setting

Cloud storage: checking connection to cloud server https://ssrf-cloud-eu.subsurface-divelog.org/git/josecdsmartins@gmail.com
Checking cloud connection...
Cloud storage: successfully checked connection to cloud server
INFO: git storage: fetch remote https://ssrf-cloud-eu.subsurface-divelog.org/git/josecdsmartins@gmail.com

git storage: Successful cloud connection, fetch remote
INFO: git storage: certificate callback for host ssrf-cloud-eu.subsurface-divelog.org with validity 1

INFO: git storage: certificate callback for host ssrf-cloud-eu.subsurface-divelog.org with validity 1

INFO: git storage: check remote status

INFO: git storage: try to update

git storage: Done syncing with cloud storage
git storage: Load dives from local cache
git storage: Successfully opened dive data
Set the current dive site: 3226160447
setupPixmap fontPrintScale 1 metrics 24 16 sz_bigger 40
Profile calculation for dive  1120 took 11 ms  -- calculated ceiling preference is false
Set the current dive site: 463918168
setupPixmap fontPrintScale 1 metrics 24 16 sz_bigger 40
Profile calculation for dive  1119 took 14 ms  -- calculated ceiling preference is false
Set the current dive site: 463918168
setupPixmap fontPrintScale 1 metrics 24 16 sz_bigger 40
Profile calculation for dive  1118 took 14 ms  -- calculated ceiling preference is false
Set the current dive site: 3673759214
setupPixmap fontPrintScale 1 metrics 24 16 sz_bigger 40
Profile calculation for dive  1116 took 10 ms  -- calculated ceiling preference is false
Set the current dive site: 3226160447
setupPixmap fontPrintScale 1 metrics 24 16 sz_bigger 40
Profile calculation for dive  1120 took 13 ms  -- calculated ceiling preference is false
Set the current dive site: 1806042461
setupPixmap fontPrintScale 1 metrics 24 16 sz_bigger 40
Profile calculation for dive  1105 took 9 ms  -- calculated ceiling preference is false
Set the current dive site: 3226160447
setupPixmap fontPrintScale 1 metrics 24 16 sz_bigger 40
Profile calculation for dive  1120 took 12 ms  -- calculated ceiling preference is false_**

Trying to open the app in the usual way however  still does not work

How can I solve the problem?

dirkhh, 2021-12-24:
Well - I'm guessing that there's something in the settings that causes everything to throw up.
The most brutal way to deal with that would be to rename
`Computer\HKEY_CURRENT_USER\SOFTWARE\Subsurface\Subsurface` to something different, like `Computer\HKEY_CURRENT_USER\SOFTWARE\Subsurface\Subsurface-broken` - you can do that with `regedit` in Windows.

Now you can start Subsurface normally, connect to your cloud storage and you should be able to work again.

SousaMartins, 2021-12-25:
It worked.
Thank you.

SousaMartins, 2023-01-08:
I belive the problem is the NahimicOSD.dll.

closed 2023-01-08 by SousaMartins