subsurface/subsurface #3365: switching windows using alt-tab moves cursor in notes field of a dive

tdyas, 2022-01-01:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
With the notes field of a dive focused and cursor positioned within field, switching to another window via Alt-Tab and then back to Subsurface with Alt-Tab moves the cursor to the start of the notes field. I expected the cursor to remain at the same position with the Notes field.

### Operating system:
Ubuntu Linux 20.04.3 LTS
KDE Plasma Version 5.18.5
KDE Frameworks Version 5.68.0
Qt Version 5.12.8

### Subsurface version:
5.0.5

### Steps to reproduce:
Click on a dive.
Click in the Notes field, for example at the end.
Press Alt-Tab to switch to another window.
Press Alt-Tab to switch back to Subsurface.
The cursor will have moved position.

### Current behavior:
The cursor moves position.

### Expected behavior:
The cursor does not move position.

dirkhh, 2022-01-05:
This may be a Ubuntu bug.
This happens neither on macOS nor on openSUSE.
In both cases the cursor stays exactly where it was before you switch away...
I don't have a Ubuntu 20.04 VM with Plasma at this point, so I can't reproduce the behavior

bstoeger, 2022-01-12:
Can confirm on Kubuntu.
Could be caused by firing the undo command when losing focus. We might improve that by "accumulating" undo and not playing tricks with focus...?

bstoeger, 2022-01-12:
Yes, that's it:
If I remove the `editingFinished()` call in `TextEdit::focusOutEvent()`, this problem disappears.

I'll try to fix this. It will take some time though, since it needs some rework of the undo machinery and is not a top priority for now.

bstoeger, 2022-01-13:
Please check #3379.

commit ea81d8fb (2022-02-14): desktop: don't update notes field when executing command

User report: when switching focus between windows, the
cursor position gets lost. This is due to a note-edit
command being fired, which then overwrites the notes tab.

To prevent this, don't update the notes field when placing
a command. Moreover, generally don't update the dive
selection when placing a command as that also rewrites all
the values.

Should this be extended to other fields?

Fixes #3365
commit 889827aa (2022-02-15): desktop: don't update notes field when executing command

User report: when switching focus between windows, the
cursor position gets lost. This is due to a note-edit
command being fired, which then overwrites the notes tab.

To prevent this, don't update the notes field when placing
a command. Moreover, generally don't update the dive
selection when placing a command as that also rewrites all
the values.

Should this be extended to other fields?

Fixes #3365
closed 2022-02-15 by dirkhh