subsurface/subsurface #3367: GoogleMaps Qt plugin: direct maps data download violation of Google's terms of service ?

cgilles, 2022-01-01:
Hi all and happy new year,

This is a question about the use of GoogleMaps in Subsurface. Sounds like a QtPositioning plugin is used for this task:

https://github.com/subsurface/googlemaps

After a quick look in source code, I see that it loads tiles directly from the Google server,
bypassing the API. This sound like a violation of Google's Terms of Service:

https://developers.google.com/maps/faq#tos_tiles

Please, let's me hear if I'm wrong with this report.

Thanks in advance

Gilles Caulier

dirkhh, 2022-01-05:
Interesting observation
(a) I'm not sure what 'quick look' you did at the the sources, but `QGeoCodingManagerEngineGooglemaps` most definitely uses the API to access the Google backend
(b) while we have our own fork (because of some past build problems with that project), questions about the project should be directed to its maintainers... https://github.com/vladest/googlemaps
(c) when first using the Google map data (actually using a different map mechanism back then) I had a chance to discuss the usage of Google satellite imagery in Subsurface with the Google open source team and they were completely unconcerned with our use of their imagery - so I'm curious what your interest in this topic is

cgilles, 2022-01-06:
Hi, and thanks for your feedback.

Well, I'm working in digiKam team, a photo management program based on Qt. We have a long time bug report about the GoogleMaps support. Typically, GoogleMaps (and also OSM), can be used in digiKam to show the geolocation of picture.

GoogleMaps was one of the geolocation service supported in digiKam, 10 years ago as i remember. And the users feedback have been very good.

But one year ago, since Google has changed the policy to use the web service, the GoogleMaps view become unsuitable, showing an overlay "for developer purpose only".

https://i.imgur.com/ug9Kh9g.png

To resume, because it's a very long story, if you want to use the service you need to pay a Google Cloud account. This is unacceptable for an open source application as digiKam.

The bug file is this one : https://bugs.kde.org/show_bug.cgi?id=433732

Read well the comments in this report, all is well explained. And of course we tried to talk with Google team, but we receive no response.

This is why when i tested Subsurface on my computer, i was surprised to see a geolocation view based on GoogleMaps working like a charm. So i would to know how you deal with the service technically, without to 
violate the Google term of service.

Note: if you is interested by the code used in digiKam, it's hosted here: 

https://invent.kde.org/graphics/digikam/-/tree/master/core/utilities/geolocation/geoiface/backends

Best regards

Gilles Caulier

mikeller, 2026-10-17:
Hi @cgilles,

Thanks for raising this carefully rather than assuming the worst. As explained above: our GoogleMaps plugin (QGeoCodingManagerEngineGooglemaps) genuinely goes through Google's API rather than scraping tiles directly, it's a fork with its own maintainers for plugin-specific questions, and we'd previously discussed Subsurface's use of Google satellite imagery directly with Google's own open source team, who were completely unconcerned about it.

Closing this as addressed via that explanation. We'd revisit this if Google ever took concrete enforcement action against Subsurface specifically, but we're not aware of that having happened.

closed 2026-10-17 by mikeller