subsurface/subsurface #3371: wrong depths starting from 5.0.3

kruegerha, 2022-01-07:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I'm using a Scubapro Tec 3G divecomputer, read out via IRDA.
After downloading dives, the depths in Subsurface are approx. 2% larger than shown on the dive computer.
This wrong(?) behaviour starts from Subsurface v5.0.3  (libdivecomputer at b6fd35...)
In 5.0.2 (libdivecomputer at 9ab0800...) seems to be correct, and shows the same values as on the divecomputer.

Not sure if this could be a misinterpretation of freshwater/saltwater? The divecomputer is correctly set to freshwater, this also show up correctly on the dives information tab.

### Operating system:
Linux-4.14.261-gentoo

Scubapro Tec 3G

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

kruegerha, 2022-01-09:
I tracked this down to commit cfc9ddc in libdivecomputer...
The "bug" actually is inside my divecomputer, which obviously uses 10 as an approximation of the standard gravity, otherwise the values wouldn't be different when libdivecomputer uses the correct standard gravity (after the mentioned commit).
Is that common in all devices? 

After using subsurface across several versions, it will be impossible to find out how the stored depths were derived.
For most people this will not make a difference, some may wonder why the maximum depths between the dive computer and subsurface differs.

A more data oriented approach would be to store the raw sensor pressure values and apply conversions and corrections later. This would also allow for later changes/calibration of the water density... etc.

dirkhh, 2022-01-09:
I think the reasonable suggestions would be to bring this up with the libdivecomputer maintainer @jefdriesen 

My initial response would be that I'd prefer the data that libdivecomputer returns to be aligned with what the dive computer actually shows on screen - but in my experience Jef is very careful in what he does and usually thinks these things through, so I'm wondering if there is a reason why he made that change that isn't obvious to me right now. Making the change to the uwatec backend in that commit seems... surprising to me.

jefdriesen, 2022-01-10:
This has already been reverted in commit a8bcfb998b9c400e42b98db437cbf1aba6c13bfc.

Anyway, in general libdivecomputer prefers to use the scientific correct conversion formulas, and thus also the non-approximated gravity factor (9.80665 m/s²). This allows an application to convert the depth back into an absolute pressure (taking into account atmospheric pressure and water density) and use that value to perform decompression calculations. But that's of course only possible if the same gravity constant is used in all backends. Otherwise the application needs to know which value was used, and unlike the atm pressure and water density, libdivecomputer doesn't tell you that.

But as Dirk mentioned, users expect to get the same depth values as shown by their computer. A difference (even if technically correct) is indeed confusing for users. That's why I already reverted the change. But that fix is probably not yet in subsurface?

Note that reporting raw pressure values and leaving the conversion to the application isn't possible because most dive computers simply don't store that information. They store the already converted depth, and often the exact conversion factors (atm pressure, water density and gravity) aren't available or known.

dirkhh, 2022-01-17:
ok, I'll include this in 5.0.6 which I will release in the "relatively near future" :)

closed 2022-01-17 by dirkhh