subsurface/subsurface #3375: Unable to edit tanks and gas mixes in the planner.

glance-, 2022-01-10:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Unable to edit tanks and gas mixes in the planner.

When I hit the + sign and a new gas row is added, I can't edit that line. It's just "blank". Subsurface handles it as a air cylinder with your default cylinder size, and gas changes chan be added to air.

Not even the trash can to delete that row 

### Operating system:
Linux, Qt 5.15.2, Ubuntu 20.04

### Subsurface version:
Current master, v5.0.5-173-gf6a1340a2976.

### Steps to reproduce:
Start subsurface, enter the planner, hit the + cylinder button.

### Current behavior:
I can't edit the cylinder row.

### Expected behavior:
I would like to be able to edit the cylinder row to change gasmix and tank size.

### Additional information:
The depth table can be edited, so something's just wrong with the gas table. The initial gas is there, but I just can't edit new lines.

glance-, 2022-01-10:
Works as expected in v5.0.5

glance-, 2022-01-10:
Bisected this one down to a40b40ae7a3a6d17b942b62b265d27a3623da4b9

glance-, 2022-01-10:
@bstoeger Do you have any top of mind ideas on how to fix this?

bstoeger, 2022-01-11:
I'll look at it by the end of the week.

PR #3377 (2022-01-17): planner: properly keep track of cylinder count in model
In a40b40a the cylinder-hiding functionality was changed,
which made it necessary to keep track of the number of
cylinders. Ironically, that code was removed previously,
as it was redundant. The count was not readded to the
functions called by the planner, making editing of cylinders
in the planner impossible.

I wonder more and more if the models for planner and the
equipment tab should be changed. They are too different.

Fixes #3375

Reported-by: Anton Lundin <glance@acc.umu.se>

### Describe the pull request:

- [x] Bug fix
- [ ] Func [... cut]
commit 1af4032d (2022-01-17): planner: properly keep track of cylinder count in model

In a40b40a the cylinder-hiding functionality was changed,
which made it necessary to keep track of the number of
cylinders. Ironically, that code was removed previously,
as it was redundant. The count was not readded to the
functions called by the planner, making editing of cylinders
in the planner impossible.

I wonder more and more if the models for planner and the
equipment tab should be changed. They are too different.

Fixes #3375

Reported-by: Anton Lundin <glance@acc.umu.se>
closed 2022-01-17 by dirkhh