subsurface/subsurface #3403: Unable to select second tank for first segment in Dive points table

jirkapok, 2022-02-23:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
A user wants to plan shallow dive with two different tanks. He wants to start with stage cylinder and keep back gas (first cylinder) for later use.

### Operating system:
* Windows 10 (2020 H2 update)

### Subsurface version:
Discovered in 5.0.5, still present in 5.0.6, 64bit desktop installer

### Steps to reproduce:
1. Create or edit dive in the planner
2. Add two cylinders (content and size irrelevant)
3. Add the following waypoints: 
- 0>20meters, 2 minutes
- 20>20meters, 10 minutes
- 20>20meters, 50 minutes
4. For the first waypoint use cylinder dropdown to select the second cylinder

### Current behavior:
Even you select the second tank from dropdown still first appears, but the second cylinder is assigned to the second segment instead of the first one. This issue appears only for the first waypoint. The user is able to select another tank for the rest of the user-defined waypoints.

### Expected behavior:
The user is able to select any cylinder for any waypoint using the Cylinder dropdown.

### Additional information:
None

### Mentions:
No

dirkhh, 2022-02-23:
The dive always starts on the first tank. So simply make the gas you want to start with your first tank.
When you select a gas at a way point, what you are effectively doing is "switch to this gas here", so the gas is used starting at that waypoint.
So I'd say that Subsurface works as designed.

jirkapok, 2022-03-03:
I see, if I switch to another cylinder on the first segment, it automatically reorders the cylinders, so the newly selected is now the first one. I missed that.
Ok, sorry for disturbing, closing as by design.

closed 2022-03-03 by jirkapok