subsurface/subsurface #3404: Picture of dive profile is incorrect

satoschu, 2022-02-24:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The pictures of the dive profiles are shown incorrect.

### Operating system:

Android 10 in a Samsung A7

### Subsurface version:

3.4.3 (5.0.6.14)

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

satoschu, 2022-02-24:
[image]

dirkhh, 2022-02-24:
You are using the beta app. This is a known issue that is currently under investigation.

mikeller, 2026-10-17:
Hi @satoschu,

Thanks for reporting the incorrect dive profile picture on mobile. This report is a bit light on specifics for us to reproduce, could you give us the exact steps (which dive, what the picture shows vs. what it should show) and a screenshot of the incorrect profile?

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.

satoschu, 2026-10-23:
Hi mikeller,
It depends on the Dive Profile If the Pictures is viewed correct or Not.

[image]
[image]

mikeller, 2026-10-23:
Thanks, the new screenshots show that the problem is data-dependent rather than affecting every dive profile.

Could you please check whether this still happens with the current Subsurface-mobile release, and provide:

- the exact Subsurface-mobile version;
- the Android version and device model;
- the affected dive exported as a `.ssrf` file, with any personal details removed if needed;
- a screenshot of the full profile for one affected dive and one unaffected dive.

The `.ssrf` sample is important because the profile rendering depends on the recorded samples and events. With one affected sample we can try to reproduce this and determine whether the problem is in the imported dive data or the mobile profile display.