subsurface/subsurface #3434: Subsurface is unable to find my Peregrine when scaning for BLE devices

danielRicardo, 2022-04-04:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When scanning for Bluetooth device as described in the manual, Subsurface is unable to find my Shearwater Peregrine computer.

### Operating system:
Windows 10 Pro - Build 19044.1586

### Subsurface version:
Version 5.0.8 - official windows build

### Steps to reproduce:

1. Start Bluetooth on the Peregrine
1. Import -> Import from Dive Computer
2. Vendor = Shearwater, Dive Computer = Peregrine
3. Choose Bluetooth download mode
4. Scan (both using the Auto option, as well as force LE)

### Current behavior:

The Peregrine does not appear on the list of discovered devices

### Expected behavior:

The Peregrine should appear on the list of discovered devices

danielRicardo, 2022-04-05:
Unfortunately I was unable to do so. Even tried the method mentioned in #2956, but to no avail.
I no it's not a problem with Subsurface, but if anyone has a suggestion - I'm all ears.

dirkhh, 2022-04-05:
The Shearwater BLE stack behaves in interesting ways. One thing that I've noticed is that sometimes when the Shearwater device "sees" what it considers its peer (so for example the phone that you last used to download from it), it will stop advertising... I usually suggest people use phone to download from BLE dive computers (usually more reliable). And with a phone you can "walk away" from all your other computers (I literally something end up going out into my back yard for this).

MdeLv, 2025-06-21:
In brief, how can Subsurface/Win11 communicate with a Peregrine ?
Who did it with reproducibility ?

What would be the best bench test to see exactly how BLE communication is going one and which component is not doing its job ?

mikeller, 2026-10-17:
Hi @danielRicardo,

Thanks for reporting that Subsurface can't find your Peregrine during a BLE scan on Windows. This looks like it may be the same cross-device "failed to enable notifications"/BLE-discovery class we've seen with several other dive computers, but we'd need a log capture to confirm that for your specific case, enabling verbose logging before scanning and sharing the output would help.

If we don't get a log within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.