subsurface/subsurface #3437: Failure retrieving dives from cloud storage

kittykat, 2022-04-11:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I have been using cloud storage to access my dives from two devices (Linux and Android) for a while. When I last started up the app on Linux, I couldn't connect to cloud storage. I can see the right dives listed on Android and at https://ssrf-cloud-eu.subsurface-divelog.org/user/kittykat3756@gmail.com/dives.html, including the latest ones.

On Linux, this is the error I got:
```
Cloud storage: checking connection to cloud server https://ssrf-cloud-eu.subsurface-divelog.org/git/kittykat3756@gmail.com
Checking cloud connection...
Cloud storage: successfully checked connection to cloud server
INFO: git storage: delete proxy setting

INFO: git storage: certificate callback for host ssrf-cloud-eu.subsurface-divelog.org with validity 1

git storage: Transfer from storage (0/60)
git storage: Transfer from storage (1/60)
git storage: Transfer from storage (2/60)
git storage: Transfer from storage (3/60)
git storage: Transfer from storage (4/60)
git storage: Transfer from storage (5/60)
git storage: Transfer from storage (6/60)
git storage: Transfer from storage (7/60)
git storage: Transfer from storage (8/60)
git storage: Transfer from storage (9/60)
git storage: Transfer from storage (10/60)
git storage: Transfer from storage (11/60)
git storage: Transfer from storage (12/60)
git storage: Transfer from storage (13/60)
git storage: Transfer from storage (14/60)
git storage: Transfer from storage (15/60)
git storage: Transfer from storage (16/60)
git storage: Transfer from storage (17/60)
git storage: Transfer from storage (18/60)
git storage: Transfer from storage (19/60)
git storage: Transfer from storage (20/60)
git storage: Transfer from storage (21/60)
git storage: Transfer from storage (22/60)
git storage: Transfer from storage (23/60)
git storage: Transfer from storage (24/60)
git storage: Transfer from storage (25/60)
git storage: Transfer from storage (26/60)
git storage: Transfer from storage (27/60)
git storage: Transfer from storage (28/60)
git storage: Transfer from storage (29/60)
git storage: Transfer from storage (30/60)
git storage: Transfer from storage (31/60)
git storage: Transfer from storage (32/60)
git storage: Transfer from storage (33/60)
git storage: Transfer from storage (34/60)
git storage: Transfer from storage (35/60)
git storage: Transfer from storage (36/60)
git storage: Transfer from storage (37/60)
git storage: Transfer from storage (38/60)
git storage: Transfer from storage (39/60)
git storage: Transfer from storage (40/60)
git storage: Transfer from storage (41/60)
git storage: Transfer from storage (42/60)
git storage: Transfer from storage (43/60)
git storage: Transfer from storage (44/60)
git storage: Transfer from storage (45/60)
git storage: Transfer from storage (46/60)
git storage: Transfer from storage (47/60)
git storage: Transfer from storage (48/60)
git storage: Transfer from storage (49/60)
git storage: Transfer from storage (50/60)
git storage: Transfer from storage (51/60)
git storage: Transfer from storage (52/60)
git storage: Transfer from storage (53/60)
git storage: Transfer from storage (54/60)
git storage: Transfer from storage (55/60)
git storage: Transfer from storage (56/60)
git storage: Transfer from storage (57/60)
git storage: Transfer from storage (58/60)
git storage: Transfer from storage (59/60)
git storage: Transfer from storage (60/60)
git storage: Transfer from storage (61/80)
git storage: Transfer from storage (62/80)
git storage: Transfer from storage (63/80)
git storage: Transfer from storage (64/80)
git storage: Transfer from storage (65/80)
git storage: Transfer from storage (66/80)
git storage: Transfer from storage (67/80)
git storage: Transfer from storage (68/80)
git storage: Transfer from storage (69/80)
git storage: Transfer from storage (70/80)
git storage: Transfer from storage (71/80)
git storage: Transfer from storage (72/80)
git storage: Transfer from storage (73/80)
git storage: Transfer from storage (74/80)
git storage: Transfer from storage (75/80)
git storage: Transfer from storage (76/80)
git storage: Transfer from storage (77/80)
git storage: Transfer from storage (78/80)
git storage: Transfer from storage (79/80)
git storage: Transfer from storage (80/80)
INFO: git storage: clone of https://ssrf-cloud-eu.subsurface-divelog.org/git/kittykat3756@gmail.com failed
INFO: git storage: loc https://ssrf-cloud-eu.subsurface-divelog.org/git/kittykat3756@gmail.com wasn't found or is not a directory
```

### Operating system:
Arch Linux

### Subsurface version:
5.0.8 installed as a Flatpak from Flathub

I haven't tested the AUR packages as they are failing to build for me at the moment (probably missing dependencies).

### Steps to reproduce:

1. Start Subsurface
2. File -> Open Cloud Storage

### Current behavior:
3. "Error connecting to Subsurface cloud storage"

### Expected behavior:

Dives are openned

### Additional information:

### Mentions:

jb2cool, 2022-04-25:
Please could you try the stand-alone AppImage and see if that works any better for you.

dirkhh, 2022-04-26:
The Flatpak isn't maintained by the developers and receives no testing from us. Jason is right about the AppImage (definitely gets a lot more testing) or the Snap (also gets some fairly frequent use).
And building the AUR shouldn't be hard... it's unfortunately unmaintained, but building Subsurface from source on a Linux system is really rather straight forward...

kittykat, 2022-04-29:
I'll have a go at building from source and AppImage - I used to build from source but had problems with it a few months ago

kittykat, 2022-12-13:
Forgot to update that the AppImage is working fine :+1:

mikeller, 2026-10-17:
Hi @kittykat,

Thanks for confirming this. At the time, the root cause was the community Flatpak package, which wasn't maintained by the project and received no testing from us, switching to the AppImage worked around that. Since then, Subsurface has started maintaining its own official, verified Flatpak (https://flathub.org/apps/org.subsurface_divelog.Subsurface), so that's now a solid option too, alongside the AppImage and Snap.

Closing this as resolved since switching to the AppImage fixed it for you at the time. Please reopen if you still see cloud sync timeouts on a current official AppImage, Snap, or Flatpak build.

closed 2026-10-17 by mikeller