subsurface/subsurface #3438: Subsurface does not transfer diving data from Seac Screen

Micio2001, 2022-04-12:
Hi everyone.
My name is Alessandro, I am new to this Forum and I apologize for my bad English.
A friend of mine bought a new dive computer (Seac Screen) and we can't transfer dives from Screen to Subsurface.
In Subsurface, for Seac computers, transfers from Seac Guru / Jack are possible but not from Screen.
The error message is "error importing dive data".
The Screen-pc connection works well because we have used SeacSync and all dives have been successfully transferred.
What can we do?
Thank you, see you soon.

[image]

dirkhh, 2022-04-15:
I don't think any of the developers have seen this dive computer, yet. It often takes a while before new dive computers become supported.

Micio2001, 2022-04-16:
Hi Dirk and thanks for your reply.

Yes, it can be as you say.

But I'm worried because in the Subsurface Google Group the problem has been open since January 2021!
The conversation at this link: https://groups.google.com/g/subsurface-divelog/c/LCzfRCcAfYQ

At the moment I think to try the solution suggested by Kisdaróczi Péter and James W.
But I have to wait for the SeacSync files from my friend Greta.
She's the one with the problem.
Greta is a member of Diver Safety Guardian (Dan) and it is important for her to send to Dan her dives.

Do you think Subsurface can be updated and when?
If necessary or useful I can help the development team... if I know how.
Screen is not the only SEAC computer not managed by Subsurface, even diving with Action has the same problem.

Can you please help me handle this problem?

Greetings from Turin and I look forward to hearing from you.

dirkhh, 2022-04-16:
You are fundamentally misunderstanding how open source works.

If SeacSync would like to see their devices supported  in Subsurface, they can provide us with the necessary API and file format documentation, just like many other dive computer vendors do. Without that, this will require someone with sufficient reverse engineering expertise, the time and **interest** to do the hard work of reverse engineering.

If a user needs Subsurface to support their dive computer, they need to buy a dive computer that is supported. This will (over time) create the economic pressure on the dive computer vendors to work with the Subsurface developers. Unfortunately, a lot of users simply assume that "someone else" will do the hard work, for free, on a time line that suits the user.

closed 2022-04-16 by dirkhh