subsurface/subsurface #3450: XML does not update from site it overwrites online data, please add a warning

pyhoff, 2022-05-04:
You have no directional indication or a warning, if your XML file is behind from Online it will overwrite online, all the data is lost, no warning. Really disappointing. I spent hours updating my dives with GPS coordinates, locations and other stuff while on a boring car ride, come home and my computer version was behind, so naturally let me sync; well all my updates are now lost. Not even a warning. 
If someone replies with RTFM, GFY you should at least warn and give an opportunity to cancel.

### Describe the issue:

- [  ] Bug
- [ x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Data is lost if local copy is behind online, if you sync it does not try to diff local copy overwrites online, say goodbye to your data.

### Operating system:

IOS and Windows

Centauri

Not sure not going to try and sync to lose all my data again.

### Subsurface version:

5.0

N/A

Oficial release

### Steps to reproduce:

1) Update data on phone, sync with online
2) Open on Windows computer, 
3) Click on sync online, kiss your data goodbye.

### Current behavior:

### Expected behavior:

Not to loose my data

### Additional information:

### Mentions:

dirkhh, 2022-05-04:
I'm curious about the exact time line of what you did.
So you entered all that data on the phone and synced on the phone.
THEN you got to the Windows computer, did something, synced, and then everything was gone?

Because if that's the timeline, then your data aren't gone at all.
Send me an email FROM the account that you use for your login and I can check if they can be restored.

mikeller, 2026-10-17:
Hi @pyhoff,

Apologies this one was left hanging, dirkhh had offered to check/restore your data privately via email at the time, and we don't have a record of whether that happened or whether your GPS/location edits were recovered. Could you let us know the current status: did the private follow-up resolve the data-recovery side, and are you still seeing cloud sync silently overwrite newer data with an older local copy with no warning?

Separately, cloud-sync overwrite safety has had real attention since your 2022 report (a related overwrite-confirmation fix shipped more recently), so it's worth checking whether current Subsurface now warns you before a sync like this happens.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed/resolved, but please reply or reopen whenever convenient.