subsurface/subsurface #3453: All cloud items in File menu are greyed out, even with correct credentials

MiM-TMassey, 2022-05-09:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I am trying to upload my recent dives to the cloud.  I don't keep the password stored in my preferences, so I went to File/Preferences/Cloud:  the correct email was there, so I type in my password and clicked Save.  (It's actually possible that I might have clicked Apply first, then Save.)

I then click File, but all three cloud options are greyed out.

In trying to make this work, I updated to the latest SS (5.0.8 x64).  I also tried to reset all settings:  I exited SS, and then deleted the Subsurface folders in %APPDATA% and %LOCALAPPDATA% and the HKCU\Software\Subsurface registry key.  I then started SS, opened the log file, went into File/Preferences/Cloud and entered the correct email and password and clicked save.  The cloud options are still greyed out.

I have successfully opened the cloud account on https://cloud.subsurface-divelog.org/ without issue:  my previously-uploaded dives are there.

### Operating system:
Windows 10 Pro 21H2, fully up to date, English.
No other devices are involved in this process.

### Subsurface version:
Subsurface 5.0.8 x64
The problem started with 5.0.3, and I upgraded as part of debugging.
The installers are downloaded from the SS website.

### Steps to reproduce:
File/Preferences/Cloud:  confirm that the correct email and password are present.  Click Save
File menu:  the cloud items are greyed out.

### Current behavior:
The cloud items are greyed out in the File menu.

### Expected behavior:
The cloud items are not greyed out in the File menu.

### Additional information:
I have successfully opened the cloud account on https://cloud.subsurface-divelog.org/ without issue:  my previously-uploaded dives are there.

### Mentions:

mikeller, 2026-10-17:
Hi @MiM-TMassey,

Thanks for the thorough troubleshooting you already did on this (ruling out stale settings and bad credentials via a full reset and reinstall). We haven't been able to take it further without a maintainer diagnosis of the cloud-enablement logic itself, specifically, we'd need a verbose log of a credential-save attempt, so we can see what Subsurface thinks is missing when it decides to leave those menu items greyed out.

If we don't hear back with that within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient, you clearly did good diagnostic work here and we'd like to follow it up properly.

mikeller, 2026-10-19:
Hi @MiM-TMassey,

Following up on this from our side: I tried to reproduce it directly. Starting from a state with no valid cloud credentials (cloud menu items greyed out), I went into File/Preferences/Cloud, entered valid credentials, and clicked Save straight away.

Back in the File menu, the cloud items were still greyed out immediately afterwards, but checking again about 20 seconds later they had become active.

So in my testing this wasn't a case of the items staying permanently disabled, it was a short delay while Subsurface verifies the credentials in the background before enabling them.

Could you confirm whether that matches what you're seeing, i.e. do the items become active if you wait 20-30 seconds after saving, or do they stay greyed out indefinitely even after waiting? That will tell us whether there's still a genuine bug here, or whether this is just the (admittedly unclear) delay while verification completes.