subsurface/subsurface #3465: Segfault when opening cloud storage

micah, 2022-06-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When I attempt to synchronize the cloud storage, Subsurface dumps core.

### Operating system:
Debian Bullseye

### Subsurface version:
Snap v5.0.8 Rev 4807

The same issue does not happen with the Debian package.

### Steps to reproduce:
1) launch subsurface
2) open File->Preferences ->Cloud-> Input cloud credentials
3) File->Open Cloud Storage

### Current behavior:
Subsurface opens the model dialog saying, "Syncing cloud storage" and then it crashes with this in the console:
```
Completed connection with cloud storage backend, response "[OK]"
INFO: git storage: unable to open local cache at /home/micah/snap/subsurface/4807/.subsurface/cloudstorage/5e114fe3f88c0691: failed open - '/etc/gitconfig' is locked: Permission denied
INFO: git storage: problems with local cache, moved to /home/micah/snap/subsurface/4807/.subsurface/cloudstorage/5e114fe3f88c0691.1
INFO: git storage: clone of https://ssrf-cloud-us.subsurface-divelog.org/git/micah@xxx failed
INFO: git storage: error message was failed open - '/etc/gitconfig' is locked: Permission denied

Segmentation fault (core dumped)
```

### Expected behavior:
It should sync.

atdotde, 2022-06-12:
The actual problem appears to be `error message was failed open - '/etc/gitconfig' is locked: Permission denied`. Could you check if there is anything that looks like a lock file in /etc and who owns it? Could it also be that your user does not have permission to read that file? Maybe you have done something as root previously that you should not have done as root and that left files around that your normal user cannot access anymore.

micah, 2022-06-12:
On 2022-06-12 06:18:07, Robert C. Helling wrote:
> The actual problem appears to be `error message was failed open -
> '/etc/gitconfig' is locked: Permission denied`. Could you check if
> there is anything that looks like a lock file in /etc and who [...]

`-rw-r--r-- 1 root root 126 Aug 14  2021 /etc/gitconfig`

definitely have the ability to read it as my user. 

I dont see any obvious lockfile in /etc

dirkhh, 2022-06-12:
Well, the thing that I am triggering off is in the first two of these lines:
```
INFO: git storage: unable to open local cache at /home/micah/snap/subsurface/4807/.subsurface/cloudstorage/5e114fe3f88c0691: failed open - '/etc/gitconfig' is locked: Permission denied
INFO: git storage: problems with local cache, moved to /home/micah/snap/subsurface/4807/.subsurface/cloudstorage/5e114fe3f88c0691.1
INFO: git storage: clone of https://ssrf-cloud-us.subsurface-divelog.org/git/micah@xxx failed
INFO: git storage: error message was failed open - '/etc/gitconfig' is locked: Permission denied
```
You are running a snap instead of a locally built binary (or a native Ubuntu .deb which often (but not always) can be installed on Debian). My guess is that the magic sand-boxing that Snap does on your machine doesn't allow the Subsurface Snap to access that file.

This is yet another twist of my long running frustration with "Desktop Linux". It's fundamentally broken. Having 270 different distros is broken. Having cross-distro container package formats like Snap and Appimage And Flatpak that aren't really working across distros is fundamentally broken.

I have no idea how to debug this, I don't claim to understand how Snap works and how it does its magic on a Debian box.

A simple click on your GitHub user account, @micah , seems to indicate that you are reasonably comfortable with development tools. So most likely the quickest path to success is to build Subsurface from source. Everything you need should be explained in the INSTALL file (and if you stumble across something that doesn't work, **THAT** I am usually easily able to help with).

closed 2022-10-03 by dirkhh

micah, 2022-10-11:
On 2022-06-12 12:14:00, Dirk Hohndel wrote:
> Well, the thing that I am triggering off is in the first two of these lines:
> ```
> INFO: git storage: unable to open local cache at [...]

Likely snap has some kind of containment that is making it so it cannot
access /etc/gitconfig outside of that containment.

> This is yet another twist of my long running frustration with "Desktop Linux". It's fundamentally broken. Having 270 different distros is broken. Having cross-distro container package formats like Snap and Appimage And Flatpak that aren't really working across distros is fundamentally broken.

I agree. Its practically impossible for software development to package
things for every distro that exists, not to mention mac and windows and
android and ios! I think it is totally reasonable for the project to
only support the distribution mechanism they prefer, and let the
community contribute more.

I do appreciate the subsurface project making the effort to make things
available via different distribution mechanisms, and I understand it is
hard and sometimes frustrating. Because of that, I consider it
best-effort and when things don't work (such as in my bug report), I
don't expect it to be a priority to be fixed, instead I just open the
bug report to get community support, someone may come around with the
answer if the problem is known!

> I have no idea how to debug this, I don't claim to understand how Snap
> works and how it does its magic on a Debian box.

Sorry, I don't know either.

> A simple click on your GitHub user account, @micah , seems to indicate
> that you are reasonably comfortable with development tools. So most
> likely the quickest path to success is to build Subsurface [...]

as I indicated in the bug report, the debian package does not have this
issue. I knew that if this didn't work, I could use the package, or the
flatpak, or build it from source, so thanks again for making options
available, but don't let the options burn you out :D

Saviq, 2024-08-19:
Shame I wasn't aware of this before (I bootstrapped and maintain the snap). While I agree with some sentiments above, we are where we are…

Surely lack of access to `/etc/gitconfig` shouldn't be fatal to Subsurface?

I tried a few things and none of them worked off the bat, the most promising was [`GIT_CONFIG_NOSYSTEM`](https://git-scm.com/book/en/v2/Git-Internals-Environment-Variables], but that doesn't seem to be respected by whatever Subsurface uses to access the git repos.

I'll see what I can do.

Sorry for necromancing an ancient issue.

Saviq, 2024-08-19:
The simplest workaround is… :facepalm: removing the `/etc/gitconfig` file.

Saviq, 2024-08-19:
And a workaround within the snap:

https://github.com/subsurface/subsurface/pull/4292

PR #4292 (2024-08-21): snap: use a local `/etc/gitconfig`
Fixes: #3465

### Describe the pull request:
- [x] Bug fix

### Pull request long description:
Use a local version of `/etc/gitconfig` in the snap, to avoid `libgit2` barfing at the system one.

### Changes made:
Lay a local `$SNAP_DATA/gitconfig` at `/etc/gitconfig`.

### Related issues:
#3465
commit 8b435c9d (2024-08-21): snap: use a local `/etc/gitconfig`

Fixes: #3465