subsurface/subsurface #3466: Merging dives with slight pressure increase in the middle

briangweber, 2022-06-12:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I did a dive last night where we swam for ~15 minutes, buddy had an equipment issue so we went up and fixed that and continued on the dive. We were up long enough that my DC registered them as 2 separate dives. As usual, I downloaded the dives and merged them, but this time the SAC showed 0.08 and the ending pressure was the pressure when I surfaced the first time. The issue appears to be that my tank pressure increased a few PSI "between dives" since the surface water was slightly warmer.

I was able to overcome the issue by editing the cylinder end pressure to be the pressure after we surfaced the second time.

### Operating system:
Windows 10

### Subsurface version:
Running: 5.0.8 official
Also happened on 5.0.7 official

### Steps to reproduce:
Using the provided log file, merge the two dives and notice the ending pressure indicated.

### Current behavior:
Uses the pressure from the end of the first dive.

### Expected behavior:
Uses the pressure from the end of the second dive

### Additional information:
Attached log (change file extension to .ssrf)
[buggy merge.txt](https://github.com/subsurface/subsurface/files/8886099/buggy.merge.txt)

### Mentions:

dirkhh, 2022-06-12:
So. Hmm.
I tried your test file. And this is the merged dive. Admittedly, with the latest test build, not with the 5.0.7 (or 5.0.8) release builds, but nothing that has changed since then should affect this merge. So I'm a bit confused...
[image]

So for you, when you use the attached file, you get a merged dive that has only that first part of the pressure line?
Can you attach a screenshot of how the merged dive looks?

briangweber, 2022-06-13:
That's what I see on the graph, but if you look at the ending pressure on
the equipment tab, it matches the ending pressure of the first dive rather
than the second dive.

[image: image.png]

This means that things like gas consumed, SAC rate, etc. are not computed
correctly.
[image: image.png]

briangweber, 2022-06-13:
That's what I see on the graph, but if you look at the ending pressure on
the equipment tab, it matches the ending pressure of the first dive rather
than the second dive.

[image]

This means that things like gas consumed, SAC rate, etc. are not computed
correctly.
[image]

dirkhh, 2022-06-13:
Save the file as is.
Open it again.
It will be fixed.
But yes, valid bug, I need to look into this to figure out why it doesn't get updated right away...