subsurface/subsurface #3469: Feature Request: New dive graph toggle - Show 24hr/local time

JasonGuthrie, 2022-06-16:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

A (hopefully) small feature request. 
If possible, it would be very useful for marine species recording if Subsurface had a feature where the user could view the 24hr time directly on the dive graph (as part of the pop out bubble when mousing over the profile). This would be in addition to (or an alternative to) the bottom timer time value.

The reason for this request is that I attempt to create accurate records of the species I see on my dives (for use in citizen science), with the aid of underwater camera. In order to achieve accurate depth readings of the animals I photograph, I sync the clocks on my camera and dive computer so I can note the exact depth and temperature that a particular animal is found in. 

Although I can add the instantaneous bottom time to the dive start time to determine the depth at which a photo was taken, it would be much easier if I could have a 24hr value directly on the dive graph (which would also allow for better synchronisation of dive and photos).

Thank you for all the hard work that has gone into Subsurface, it is a wonderful tool.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2022-06-16:
This is one of those requests that make perfect sense. But that I also struggle with a little bit. Because it's just one more thing. Just one more setting. Just one more option that very very few of our users will ever use.

We have something really similar already. If you add your pictures to the dive they will show up on the profile at the correct time (since your clocks are synced). And that allows you to then go to File->Export->"Image depths" which will generate a file that tells you the depth you were at when you took each of these pictures.

I know that this is not exactly what you are asking for... but I'd argue that it might ultimately be maybe even more useful

JasonGuthrie, 2022-06-16:
Thank you for your reply @dirkhh. I wholly appreciate that this request was quite niche in application and I do not wish to take up any of your time when it is put to better use elsewhere.

To be honest, I have not experimented with importing photos into Subsurface as I tend to work exclusively in RAW formats. I'll give it a whirl tonight and I suspect that are probably right that this would be a better way to process my datasets.

dirkhh, 2022-06-16:
I don't use it a lot, but I think for the purpose that you describe, having a profile with those pictures visible on the profile would actually be useful...
Please give it a try and let us know. If this is completely the wrong workflow for you, we can obviously look at other options, including the one you proposed. I just prefer to err towards not adding "one off" things to Subsurface... we have way too many of them (including this depth export feature I mentioned 🤷🏼‍♂️)

rafsalvioni, 2023-04-10:
@JasonGuthrie I did a small script that using a date, can get the dive data (depth, temp) and GPS coordinates (from dive or GPX) and makes a CSV. If it util for you...

https://rafsalvioni.github.io/subsurface-tools/date-fusion.html

commit 4df44bc6 (2022-06-30): profile: add full datetime to profile's tooltip-item

User-request, mostly trivial to implement.

Fixes #3469.
commit 4a235715 (2022-06-30): profile: add full datetime to profile's tooltip-item

User-request, mostly trivial to implement.

Fixes #3469.
commit f13aa8ee (2022-07-08): profile: add full datetime to profile's tooltip-item

User-request, mostly trivial to implement.

Fixes #3469.
commit 318da260 (2022-07-31): profile: add local time to profile tooltip

This is a user request: add local time of dive to the tooltip.

Showing a localized time was surprisingly complex. First I had
to add a function that decomposes a timestamp into h/m/s. Then
the QLocale time-string has to be stripped of the timezone.

Fixes #3469.