subsurface/subsurface #3476: Dive computer reported ceiling with the Scubapro G2

huguesbrun68, 2022-07-12:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ x] New feature request
- [ ] Discussion request

### Issue long description:

The ceiling from Scubapro G2 are not imported/view 

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2022-07-12:
Most dive computers don't support downloading of the reported ceiling. I don't know if the G2 makes those data available.
@jefdriesen do you happen to know?

jefdriesen, 2022-07-13:
As far as I know, it's not available for the Uwatec/Scubapro computers. It's of course always possible it was added in a recent firmware version. There is a MISC sample that contains some unknown data, that could contain decompression info. Does the Uwatec/Scubapro software show the ceiling? If not, it's very likely not available.

mikeller, 2026-10-17:
Hi @huguesbrun68,

Thanks for reporting the dive-computer-reported ceiling issue with your Scubapro G2. Could you confirm the exact scenario where this comes up (which dive, what you expect to see vs. what's shown), and do you know whether the G2 actually transmits this ceiling data in the first place? We're not certain the device sends what would be needed to display it, so that's the key thing to confirm.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.