subsurface/subsurface #3477: Automated merging after editing notes on multiple devices loses tank information

mwerle, 2022-07-29:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I recently performed the following actions:

1. Imported dives (from multiple dive computers, one of which is AI) on my phone, then sync'd to cloud.
2. Opened Subsurface on my table and did some initial editing. Sync'd some of it to cloud; some not (lost internet connectivity on way home).
3. Opened Subsurface on my PC (Windows, v5.08 of Subsurface), sync'd, then proceeded to edit the dives, and sync'd to cloud.
4. Sync'd the tablet data as I had written some notes which I did not want to re-edit.
5. Lost tank, AI, exposure protection, and weight data for several dives.

ANALYSIS:
The "Notes" data was incorrectly merged, resulting in an additional quote character to be in the data.

EG: 

```
duration 49:00 min
tags "boat"
divesiteid c5299b50
suit "5mm long + 3mm hooded vest"
notes "First boat dive, (blah blah blah
    blah blah blah.
    blah blah blah "
 
    blah blah blah.
    blah blah blah "
 cylinder vol=10.0l description="10.0ℓ" o2=32.0% start=190.295bar end=52.676bar depth=39.845m
weightsystem weight=2.0kg description="weight"
```

Of course, this then stuffs up the parsing, which makes Subsurface think there is no tank.  But instead of reading the dive-computer data from the separate file and re-creating the tank, it _deletes_ the AI data from the dive computer file!

### Operating system:

Phone and tablet:
Created by Subsurface-mobile:3.4.3(5.0.6.83):Android  (9.0):arm64:en-GB

PC:
Created by Subsurface:5.0.8.0:Windows 10 Version 2009:x86_64:en-GB

Shearwater Perdix AI; Shearwater Peregrine

The issue arose due to a merge created by editing on multiple devices in parallel.

### Subsurface version:
Created by Subsurface-mobile:3.4.3(5.0.6.83):Android  (9.0):arm64:en-GB
Created by Subsurface:5.0.8.0:Windows 10 Version 2009:x86_64:en-GB

Official releases of Subsurface; did not try to reproduce the issue artificially as yet (sorry, would debug and attach a patch, but very limited time at present).

### Steps to reproduce:

Please see the "Issue Long Description".

I think it should be sufficient to:

1. Have a dive with AI data
2. Edit the dive's notes on Device A
3. Edit the dive's notes on Device B
4. Sync Device B
5. Sync Device A

### Current behavior:

Due to the spurious quote character added to the "Note" field, the parsing in the client breaks and removes the tank information.

Instead of warning the user or recreating the tank from the Divecomputer file, it silently discards the AI data.

### Expected behavior:

Merging notes during sync should not duplicate the quote character.  This would resolve the issue.

Also, a more robust implemtation that, if a tank is missing from the metadata, Subsurface would warn the user or create the tank again instead of deleting the AI information from the Divecomputer file, would be a much safer way to handle such data corruption.

A program should NEVER automatically delete information, only create it.

### Additional information:

You have my permission to view my cloud sync data. Dives 539, 540, and 541 are examples of this issue. (As you can see, it took me a little while to work out exactly what was going on, as the client wasn't displaying the duplicate notes or providing any sort of error or warning messages..)

### Mentions:

dirkhh, 2022-07-29:
Thank you for that detailed report and analysis.
Sadly, yes, there are corner cases in which our merge algorithm does the wrong thing and information does get deleted (please note that it generally isn't lost - it's just not easy for a non-expert to restore it... so I guess it's only "functionally lost"... which I'm sure is a huge relief to the average diver `</sarcasm>`)

Based on your report, I'll take another stab at tackling this thorny issue. I'm not promising a solution, but I promise I'll look into it.

torvalds, 2022-07-29:
@mwerle - can you make the git repo of your dives available to me somehow? I'd love to see the merge and have a real-life test-case that I could actually play with.

Now, part of the design with the git save format using a line-based file format was very much "we can just merge using git, ignore the conflict markers, and try to parse things one line at a time". So in a sense, this part is working by design.

What sounds like an outright bug is that 

> But instead of reading the dive-computer data from the separate file and re-creating the tank,
> it deletes the AI data from the dive computer file!

and that I think is a major misfeature.

But it's also true that the whole "line-based format" falls down a bit on the few multi-line things we have, notably comments. They are *supposed* to be something subsurface also can deal with because the multi-line format is special (ie spaces at the beginning of lines), so mixing a note line continuation with a non-note line *should* just work still, but it clearly doesn't really work.

Which is why I'd like to get that real-life example instead of trying to make up one of my own.

torvalds, 2022-07-31:
Ok, I hope that pull request I just opened fixes this.

@dirkhh - I'm not entirely convinced this should go into some imminent 5.0.9. It looks good to me, but it's kind of core code and should probably see more testing before put into a release.

That said, I did make it do that

        SSRF_INFO("git storage: Ignoring line ...);

thing for when the new code triggers. So you'll *see* if it triggers. Which it shouldn't in normal circumstances, but ...

(Of course, even if this works 100% correctly, you'll only see that git storage note when loading from git state that contains that merged data. If you then update and write the dive out again, the messed-up conflict data will be gone and you won't see that message any more)

torvalds, 2022-07-31:
@mwerle note that for your repository in particular, that new code will not fix your problem, because you already edited the dives in question and thus the issue went away.

So the problem is visible in the state of one of the borked merges, but then when you fixed it up by hand it's now gone and running a new version of subsurface with my suggested fix won't actually do anything.

dirkhh, 2022-07-31:
I'll play with it, but I agree that adding this to 5.0.9 would be unwise. I hope to do that tomorrow and then never this afterwards

mwerle, 2022-07-31:
Morning guys!  Thanks so much for looking into this so quickly.

Ah, makes a lot more sense to me now how the merge could fail in this way if you're "simply" doing a git merge instead of having a context-aware merge built into Subsurface.

@torvalds  I know based on the mailing list that Dirk can access users' online data after they give permission. If you need me to provide you with additional information to access it, please let me know. But based on your later comments it appears you may have managed to access it already?

@dirkhh Yes, I'm aware the data isn't "lost" per se, but for all intents and purposes for a normal person it is.  My point was that even if a tank is deleted (whether due to a bug such as this, or whether a user actually deletes a tank using the UI) the associated Air-Integration data in the dive computer file should not be deleted.

dirkhh, 2022-07-31:
Oh yes, I totally got what you said (see the sarcasm tag). And yes, I interpreted your permission to access the data to extend to Linus.

Note, the fix is NOT in Subsurface 5.0.9 - we'll make another version fairly soon that will include this change.

mwerle, 2022-08-01:
> But it's also true that the whole "line-based format" falls down a bit on the few multi-line things we have, notably comments. They are _supposed_ to be something subsurface also can deal with [...]

Thinking about this further, would it make sense to just remove the quotes enclosing the "notes" field entirely?  Since, by definition, it's a string already, and continuation lines are "special" by being indented, there's no need to additionally wrap the note in quotation marks. 

That should resolve this issue as well.  Of course, there's the problem of backward-compatibility if the file format is modified.

dirkhh, 2022-08-01:
I just realize that the commit didn't include a `Fixes:` line...
But the change was just merged into master.

closed 2022-08-01 by dirkhh