subsurface/subsurface #3483: zoomed dive profile does not pan

jeichhor, 2022-08-09:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When I try to pan the zoomed in dive profile, the profile gets lighter after touchbegin, but does not move.

### Operating system:
LineageOS 19.1

Google Pixel 3a

### Subsurface version:

3.4.6(5.0.9)

### Steps to reproduce:

open dive details
zoom into dive profile
try to pan dive profile

### Current behavior:

dive profile lights up as it touchbegin fires, but does not move

### Expected behavior:

I can inspect dive profile in higher resolution

### Additional information:

[2022-08-09 subsurface.log](https://github.com/subsurface/subsurface/files/9293248/2022-08-09.subsurface.log)

### Mentions:

dirkhh, 2022-08-10:
The user experience for the zoom is absolutely ridiculous and horrible, but at least for me it works.
I don't understand what you are saying with "dive profile lights up as it touchbegin fires".
The way zoom works is with the scroll wheel (or two finger slide down on a touchpad) and then moving the mouse itself pans things around. Terrible, but it's what we have until someone implements something better.

jeichhor, 2022-08-10:
Thanks @dirkhh for looking into this - I've noticed the behaviour in the mobile app on android. 

For the science I've plugged keyboard+mouse into the phone:
Zooming using mousewheel works, but can't drag the zoomed dive profile into any direction. The dive profile gets lighter on mousedown, but nothing happens on mousedrag.

dirkhh, 2022-08-10:
My mistake - I completely missed the fact that you were talking about Android (Lineage OS made me think of desktop... yeah, I know...)
And you are correct, this seems broken. I can zoom, with the touch screen, but I can't pan. I need to add this to my two mile-long list of things to look into. Thanks for the report.

tggm, 2022-08-12:
The biggest issue for me is, when zoomed in it's very difficult to see the end of the profile as we need to drag the mouse pointer almost beyond the main window.

Perhaps add some padding to the end of profile chart area? Of course the ideal solution would be to disconnect the Zoom from the Pan actions.

bstoeger, 2022-08-28:
> The biggest issue for me is, when zoomed in it's very difficult to see the end of the profile as we need to drag the mouse pointer almost beyond the main window.

I think you are talking about desktop, not mobile, which is what this bug report is about.

Yes, the panning on desktop is terrible and has to be redone.