subsurface/subsurface #3484: Discussion Request: Native iOS UI

Hustenbonbon, 2022-08-10:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

Even if people made a lot of effort in improving the UI for mobile, the App feels not like a modern iOS app because of its buttons, dropdowns, and other UI elements. 

As an iOS developer who plans to use subsurface to log my dives, I would love to help to create a native iOS UI that uses all of the logic layer of subsurface under the hood. 
Of course there would also be the possibility of using another cross-platform UI framework more optimized for mobile like Flutter or React Native (which would also work on Android). If wished, I can go into detail on why I think in this case a native app is a much better idea, and also, I would be interested in doing the main initial work *if* it is a native iOS app.

Advantages I see:
- With few lines of code a modern/easy-to-use UI
- Less breaking UI code on new iOS versions
- No problems with getting accepted into the App Store

Disadvantages:
- Instead of QT, native iOS SwiftUI code needs to be maintained
- no shared codebase with Android UI anymore

My main question here is if this would be welcome and supported by the maintainers before I spend a couple of weeks working on the PR.

### Operating system:
iOS

### Additional information:

### Mentions:

dirkhh, 2022-08-10:
This is such a difficult question.
I have no experience with Swift and the iOS ecosystem. And no bandwidth to learn. Which means unless we find someone who is willing to LONG TERM be the maintainer of that UI, I think this is not a good investment of time.
Additionally, a lot of the infrastructure that we use to communicate with dive computers, communicate with the backend services, parse and validate data, etc, are written in a combination of C, C++, and Qt. I have no idea how those would interact with a Swift UI.

So in principal? Sure, I'd love to see a better UI.
Practically I am very concerned about the amount of effort this would take, about the challenges of integrating this with our existing code base, about the challenges to avoid massive amounts of code duplication, and fundamentally, about the long term maintainability of what would be created.

BTW: the app store acceptance has NOTHING to do with the QML UI. The problem is that we allow account creation in the app, but don't have a feature to delete accounts in the app. Today this is a manual process. So I first need to implement a way to automate account deletion in the backend (about 75% there), and then need to add the feature to trigger that in the mobile UI (about 50% there). My guess is that this will take me another week or so, plus whatever the random review delay from the app store might add...

Hustenbonbon, 2022-08-16:
Hey, 
thanks for the quick and differentiated response! Your arguments make a lot of sense.
When I find the time, I will try around playing with the QML UI or trying to integrate native UI on iOS at some places, and if I find anything interesting, I'll comment it here or will create a PR if there is something of value.

In general, I'm amazed that an open source app like this exists with all the dive computer integrations and data processing and visualization.

My comment about app store acceptance was not about a current issue (I wasn't aware of it), but rather that in my past experience, apple sometimes had a very strict position on how an app ui has to look and feel. If that already isn't an issue here, even better.

bstoeger, 2022-08-28:
> Of course there would also be the possibility of using another cross-platform UI framework more optimized for mobile like Flutter or React Native (which would also work on Android).

Frankly, this would be my preferred option, since I despise QML. However, note that many of the core structures rely heavily on the Qt infrastructure (e.g. QString, QVector, signal/slots, translation, etc.), so this would be quite a difficult task. I would be more than happy to cleanly separate the core and UI layers and remove the Qt dependencies from the core. Perhaps I could provide support with that, though my time is currently very limited. I certainly will not be able to help with native iOS though. Apple UIs do not work at all with me, sorry.

bstoeger, 2022-08-28:
Also note that the statistics widget is implemented in `QtQuick` and the plan is to turn the profile into a `QtQuick` widget. Not sure how you plan to convert those.

dirkhh, 2022-08-28:
As I have said many times - I hate QML, I hate the current UI.
But the choice of QML was driven by the fact that we wanted to share as much as possible (and in the end, not quite as much as we wanted) between desktop and mobile.
We had a number of people suggest that we should build a different UI. But we have no developers to do so. I certainly don't have the time and don't have the skill.
I obviously can't stop anyone from working on this - but that's a massive amount of work and I'm not sure what the long term trajectory would be.

Hustenbonbon, 2022-08-29:
Hi @bstoeger, separating the core and UI layer stricter would certainly help with another UI, but I couldn't promise that it leads to any results on my side.
I spent a day trying to make the project build on my app, but didn't get the build system running (I'm not the most experienced with CMake, and I couldn't find the problem in the bash script, but thats not a topic for this thread).

The advantage that I would see with native iOS is that you can use C and C++ dependencies due to Objective-C-bridging headers. Certainly, it would be better in perspective of amount of overall code to at least have a shared UI for iOS and Android, but my personal experience as a mobile developer is that this often is a false assumption that you actually have less code with cross-platform apps (I haven't done a real flutter app yet though). 

With React Native, we certainly would end up needing a lot of React Native UI code, iOS-platform code, Android-platform code, iOS- C++ adapter code and Android C++ adapter code. This is why I personally wouldn't advise for this way. Flutter looks similar for me at first glance, but I can't tell it from experience.
From my experience, cross-platform apps require a lot more active maintenance too, as their framework interfaces and underlying native interfaces change more often.

I'll keep you updated if I make any progress in trying to use the core in an iOS app. But my first tries were not very promising so far 😂 

@dirkhh I'm totally fine with your statement, and I will keep in mind to not expect someone else to keep maintaining some  new code I write in a different framework/programming language. If I would find the time actually creating a native iOS version (not that likely at the moment to be frank), I would maintain it as a fork for as long as I seem it worth.

dirkhh, 2022-08-29:
@Hustenbonbon the iOS app is built with qmake using Subsurface-mobile.pro
take a look at packaging/ios for the scripts that I use to build things here.
If you run into problems with that, please let me know and I'll try to help

jfietz, 2024-02-08:
@bstoeger Did you ever get around to the project for better separation of core and UI, or do you have a description to explain what ideas you had in mind?

I've always loved subsurface but end up not using it due to its mobile app (with my alternative being no logs). Last time I looked into the UI layer, the task was too daunting. Reading the comments here, it seems like I may be able to contribute to at least enable a new UI  down the line. I'm sure there are a few more in my shoes,  so some ideas and guidance may attract some people?