subsurface/subsurface #3485: Regain control over your data: Export to / Import from file

sjvudp, 2022-08-13:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

I stored my dives locally on my phone (not liking clouds). When I got a new phone I wanted to transfer by data.

After having copied all the "suspect" files to the new device, it still did not find any dives!

So I realized that I have no idea where the data is stored actually!

I see that you can upload your data to some sites, but you cannot save your data in a file!

I think the app should be able to save (archive) the data to a file (maybe some ZIP), and allow restoring the data on a different device.
That would give the user real freedom of choice where to save the data.

### Operating system:

Old phone: Android 8
New phone: Android 12

Mobile

### Subsurface version:

3.4.6 (5.0.9)

official release

### Steps to reproduce:

### Current behavior:

No dives found

### Expected behavior:

### Additional information:

### Mentions:

SirHardware, 2023-06-19:
I wanted to use my own Nextcloud to sync between Desktop and Mobile. 
Sadly on mobile there is no option to specify the path to the file like on Desktop.
I would love to see this option since it would increase the flexibility.

mikeller, 2026-10-18:
Hi @sjvudp,

We're separately checking whether the mobile XML export/import work coming out of the ongoing Qt6/Kirigami modernisation effort (GH#4594) already covers what you're after here, a contributor there mentioned XML export/import working as part of that broader rework. Before we start anything new specifically for this, can you confirm: are you still hitting this gap on a current build, and would a straightforward export-to-file / import-from-file menu action (independent of cloud sync) solve it for you?

This is a real concern (losing all local dive data on device migration is serious), so we want to make sure we're not duplicating work already landing elsewhere.

sjvudp, 2026-10-28:
Honestly I wasn't using the app (desktop or mobile) recently, but I refuse to use cloud storage, preferring to have my data locally.
So an export/import of data for backup, synchronization, recovery and troubleshooting would be a nice feature.

glance-, 2026-10-28:
On android, the data is stored in `/data/data/org.subsurfacedivelog.mobile/files/cloudstorage/localrepo`

On a previous rooted phone I used to manually sync data on and off between my phone and laptop with just running:
`/data/data/org.subsurfacedivelog.mobile/files/cloudstorage/localrepo # /data/data/com.termux/files/usr/bin/git daemon --verbose --reuseaddr  --export-all --enable=receive-pack`

Subsurface uses libgit2 to sync to and from "cloud storage" , so nextcloud won't work. It's due to the union merge capability of git and clever data format design the synchronization of data can be made.

The app used to have support for pushing/pulling your dive data to and from a regular git server over ssh. It's still party there, but I don't think it's built into the "regular" builds any more. It's never been a documented feature, and it's way to nish for the masses but if you really care for your data and staying of "the could" feel free to build your own subsurface apk and use it.

glance-, 2026-10-28:
You can export your dives from the app to a subsurface xml file via Menu -> Dive management -> Export. That xml file can then be synced via whatever way you'd like to either another phone or the desktop app import / open it there.

mikeller, 2026-10-28:
Thanks both for confirming, and for the detail on the local git-based sync, @glance-.

Good timing on this one: subsurface/subsurface#5063 was just merged (26 Oct 2026) and adds a proper "Save to file" option for the Subsurface XML export on Android and iOS. It uses each platform's native save dialog (Storage Access Framework on Android, the Files picker on iOS), so you can now save the exported .ssrf file straight to a folder, Google Drive, a Nextcloud client, etc, without needing a cloud account or emailing it to yourself first. That should land in the nightly builds shortly, if it isn't already there.

That covers getting your data out. The piece that's still missing, and the one that actually caused your original problem, is the reverse: there is no "Load from file" / import action anywhere in Subsurface-mobile. So even with a saved .ssrf file in hand, there's currently no in-app way to load it back onto another phone. The only supported paths onto a device remain Subsurface Cloud sync or a fresh dive-computer download. The manual git-based sync @glance- described is real, but it's an undocumented, advanced option that needs a rooted device and Termux, not something we'd point most users at.

@sjvudp, we're now tracking a "Load from file" action for Subsurface-mobile as the counterpart to the export work in #5063, so that local backup and restore will work without a cloud account. I can't give you a timeline, but it's noted for planning.

In the meantime, if you want to move your current data off the old phone: use the new "Save to file" export once it's available in a nightly build (or the email-based export in the meantime), transfer the file to the new device however suits you, and if you have access to a desktop/laptop, open it there with File -> Import. A Subsurface Cloud account would then let the phone and desktop stay in sync automatically going forward, but that remains entirely optional.

mikeller, 2026-10-30:
Hi all,

Following up on the "Load from file" counterpart discussed above: a pull request is now open to add it: #5088 (mobile: add "Load from file" dive import for Android, iOS and desktop). It's currently a draft PR while it goes through review, so please don't expect it in a release build yet.

Once CI finishes building test artifacts for that PR, anyone who'd like to try it out on Android is welcome to do so, it's an unmerged test build, not a release. Please report your results directly on the PR thread so they're easy to track there.

Thanks again for the report and for the detail everyone added in this thread.

closed 2026-11-01 by mikeller