subsurface/subsurface #3489: Official subsurface builds for jammy (ubuntu 22.04) don't support irda computers

theundefined, 2022-08-15:
### Describe the issue:

- [ ] Bug
- [ X] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Current build for ubuntu 22.04 don't support older irda connected computers. Main reason is lack of irda support in current kernel. But - there is a project to install irda support into modern kernel (https://github.com/cschramm/irda), what after adding to dkms gives support for irda. It would be great if subsurface could also support irda computers.

### Operating system:

Ubuntu 22.04

### Subsurface version:

5.0.9

### Steps to reproduce:

Install subsurface and check if import from uwatec 2G is available. Checked both on snap and launchpad version.

### Current behavior:

uwatec 2G is not on the list of available computers to import.

### Expected behavior:

Uwatec 2G is on the list of available computers to import

### Additional information:

I've created dummy docker file to build subsurface with older linux-libc-dev library. After build in that way - installed deb subsurface works fine with my old computer. Docker and script for build I've put on https://github.com/theundefined/mydockers/tree/master/subsurface
I know that it's a dirty hack, but it's much simpler than create virtualmachine ;)

### Mentions:

dirkhh, 2022-08-15:
I believe it is reasonable to assume that an app built for a specific OS is targeting the kernel / libraries that are shipped with that OS -- which is what Subsurface does.

Given how easy it is to build Subsurface from source, I think that someone who is willing to go through the steps to install a specific kernel and a different dev environment would be capable of building their own Subsurface binary.

closed 2022-08-15 by dirkhh

theundefined, 2022-08-15:
I disagree. It's true for builds included with distribution - in that case subsurface can be castrated from features. But launchpad and snaps are independed from ubuntu distribution - why not support in that case full functionality?